gtxaspec/wz_mini_hacks #598: test & transition to kernel 4.4.94

gtxaspec, 2023-08-25:
test kernel build

sideup66, 2023-08-25:
Whats the goal of moving kernel version? It might be a tight squeeze to fit kernel 4 on some of these cameras. For instance the v3 only has ~100 MB of ram onboard. Its perfectly fine memory wise with kernel 3, not to mention 4 may open the door to security concerns as itll have more features then 3....does the newest wyze firmware already offer kernel 4 on stock?

Not complaining...just curious :)

gtxaspec, 2023-08-26:
no one offers kernel 4 on a commercial product that i've seen so far.  for us, better usb support (floodlight etc) and a few other efficiency tweaks.

gtxaspec, 2023-09-01:
```
root@openipc-t31:/sys/class/gpio# dmesg
[    0.000000] Linux version 4.4.94 (devel@mips-development) (gcc version 4.7.2 (Ingenic r2.3.3 2016.12) ) #7 PREEMPT Thu Aug 31 22:52:29 PDT 2023
[    0.000000] CPU0 RESET ERROR PC:80234EC0
[    0.000000] [<80234ec0>] __delay+0x0/0x10
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00d00100 (Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] MIPS: machine is generic,personalcam
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00696000 @ 00010000 (usable)
[    0.000000]  memory: 0003a000 @ 006a6000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 06300000 @ 00000000 (usable)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x00000000062fffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x00000000062fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000062fffff]
[    0.000000] On node 0 totalpages: 25344
[    0.000000] free_area_init_node: node 0, pgdat 806a30e0, node_mem_map 8100f000
[    0.000000]   Normal zone: 198 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 25344 pages, LIFO batch:7
[    0.000000] Primary instruction cache 32kB, VIPT, 8-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] Unified secondary cache 128kB 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 25146
[    0.000000] Kernel command line: mem=99M rmem=29M@0x6300000 console=ttyS1,115200n8 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),64k(env),3072k(kernel),10240k(rootfs),-(rootfs_data) nogmac
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Memory: 92972K/101376K available (5050K kernel code, 229K rwdata, 1456K rodata, 232K init, 134K bss, 8404K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:200
[    0.000000] ###### ingenic_num_chips = 2   ###  num_chips = 2  
[    0.000000] t31 Clock Power Management Unit init!
[    0.000000] =========== t31 clocks: =============
[    0.000000] 	apll     = 1392000000 , mpll     = 1200000000
[    0.000000] 	cpu_clk  = 1392000000 , l2c_clk  = 696000000
[    0.000000] 	ahb0_clk = 200000000 , ahb2_clk = 200000000
[    0.000000] 	apb_clk  = 100000000 , ext_clk  = 24000000, ddr_clk  = 600000000
[    0.000000] 
[    0.000000] === enter clocksource_probe  ===
[    0.000000] =====>   
[    0.000000] [ingenic_ost_init][267]====== ingenic_ost_init ====== 
[    0.000000] clocksource: ingenic_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1623fa770, max_idle_ns: 1763180808952 ns
[    0.000004] sched_clock: 64 bits at 1500kHz, resolution 666ns, wraps every 4398046510951ns
[    0.008511] ERROR epc 0x80234ec0
[    0.011784] Console: colour dummy device 80x25
[    0.016216] Calibrating delay loop... step 0 
[    0.108435] 1391.00 BogoMIPS (lpj=6955008)
[    0.112503] pid_max: default: 32768 minimum: 301
[    0.117251] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.123901] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.132347] devtmpfs: initialized
[    0.139366] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.149455] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.155665] pinctrl core: initialized pinctrl subsystem
[    0.161900] NET: Registered protocol family 16
[    0.170376] ingenic pinctrl 10010000.pinctrl: 3 gpio chip add success, pins 96
[    0.178126] ingenic pinctrl 10010000.pinctrl: ingenic pinctrl probe success
[    0.190117] dmmu_init 807 PTRS_PER_PTE = 1024
[    0.252094] ingenic-dma 13420000.dma: INGENIC SoC DMA initialized
[... 67 lines cut ...]
[    1.213099] ledtrig-cpu: registered to indicate activity on CPUs
[    1.220167] usbcore: registered new interface driver usbhid
[    1.225915] usbhid: USB HID core driver
[    1.230437] Netfilter messages via NETLINK v0.30.
[    1.235398] ip_set: protocol 6
[    1.238887] ip_tables: (C) 2000-2006 Netfilter Core Team
[    1.244523] NET: Registered protocol family 17
[    1.253628] ingenic,mmc 13450000.msc: no support for card's volts
[    1.260897] input: gpio_keys as /devices/platform/gpio_keys/input/input0
[    1.268035] mmc0: error -22 whilst initialising SDIO card
[    1.274085] hctosys: unable to open rtc device (rtc0)
[    1.281068] ingenic,mmc 13450000.msc: no support for card's volts
[    1.288204] ALSA device list:
[    1.291263]   No soundcards found.
[    1.295174] mmc0: error -22 whilst initialising MMC card
[    1.303451] VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
[    1.311143] Freeing unused kernel memory: 232K
[    1.342221] ingenic,mmc 13450000.msc: no support for card's volts
[    1.354759] mmc0: error -22 whilst initialising SDIO card
[    1.364430] ingenic,mmc 13450000.msc: no support for card's volts
[    1.370799] mmc0: error -22 whilst initialising MMC card
[    1.382228] ingenic,mmc 13460000.msc: card claims to support voltages below defined range
[    1.396745] mmc1: new SDIO card at address 0001
[    3.032303] random: nonblocking pool is initialized
```

gtxaspec, 2023-09-01:
```
root@t31:/sys/class/gpio# dmesg
[    0.000000] Linux version 4.4.94 (devel@mips-development) (gcc version 4.7.2 (Ingenic r2.3.3 2016.12) ) #7 PREEMPT Thu Aug 31 22:52:29 PDT 2023
[    0.000000] CPU0 RESET ERROR PC:80234EC0
[    0.000000] [<80234ec0>] __delay+0x0/0x10
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00d00100 (Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] MIPS: machine is generic,personalcam
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 00696000 @ 00010000 (usable)
[    0.000000]  memory: 0003a000 @ 006a6000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 06300000 @ 00000000 (usable)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x00000000062fffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x00000000062fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000062fffff]
[    0.000000] On node 0 totalpages: 25344
[    0.000000] free_area_init_node: node 0, pgdat 806a30e0, node_mem_map 8100f000
[    0.000000]   Normal zone: 198 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 25344 pages, LIFO batch:7
[    0.000000] Primary instruction cache 32kB, VIPT, 8-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] Unified secondary cache 128kB 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 25146
[    0.000000] Kernel command line: mem=99M rmem=29M@0x6300000 console=ttyS1,115200n8 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),64k(env),3072k(kernel),10240k(rootfs),-(rootfs_data) nogmac
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Memory: 92972K/101376K available (5050K kernel code, 229K rwdata, 1456K rodata, 232K init, 134K bss, 8404K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:200
[    0.000000] ###### ingenic_num_chips = 2   ###  num_chips = 2  
[    0.000000] t31 Clock Power Management Unit init!
[    0.000000] =========== t31 clocks: =============
[    0.000000] 	apll     = 1392000000 , mpll     = 1200000000
[    0.000000] 	cpu_clk  = 1392000000 , l2c_clk  = 696000000
[    0.000000] 	ahb0_clk = 200000000 , ahb2_clk = 200000000
[    0.000000] 	apb_clk  = 100000000 , ext_clk  = 24000000, ddr_clk  = 600000000
[    0.000000] 
[    0.000000] === enter clocksource_probe  ===
[    0.000000] =====>   
[    0.000000] [ingenic_ost_init][267]====== ingenic_ost_init ====== 
[    0.000000] clocksource: ingenic_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1623fa770, max_idle_ns: 1763180808952 ns
[    0.000004] sched_clock: 64 bits at 1500kHz, resolution 666ns, wraps every 4398046510951ns
[    0.008511] ERROR epc 0x80234ec0
[    0.011784] Console: colour dummy device 80x25
[    0.016216] Calibrating delay loop... step 0 
[    0.108435] 1391.00 BogoMIPS (lpj=6955008)
[    0.112503] pid_max: default: 32768 minimum: 301
[    0.117251] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.123901] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.132347] devtmpfs: initialized
[    0.139366] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.149455] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.155665] pinctrl core: initialized pinctrl subsystem
[    0.161900] NET: Registered protocol family 16
[    0.170376] ingenic pinctrl 10010000.pinctrl: 3 gpio chip add success, pins 96
[    0.178126] ingenic pinctrl 10010000.pinctrl: ingenic pinctrl probe success
[    0.190117] dmmu_init 807 PTRS_PER_PTE = 1024
[    0.252094] ingenic-dma 13420000.dma: INGENIC SoC DMA initialized
[... 67 lines cut ...]
[    1.213099] ledtrig-cpu: registered to indicate activity on CPUs
[    1.220167] usbcore: registered new interface driver usbhid
[    1.225915] usbhid: USB HID core driver
[    1.230437] Netfilter messages via NETLINK v0.30.
[    1.235398] ip_set: protocol 6
[    1.238887] ip_tables: (C) 2000-2006 Netfilter Core Team
[    1.244523] NET: Registered protocol family 17
[    1.253628] ingenic,mmc 13450000.msc: no support for card's volts
[    1.260897] input: gpio_keys as /devices/platform/gpio_keys/input/input0
[    1.268035] mmc0: error -22 whilst initialising SDIO card
[    1.274085] hctosys: unable to open rtc device (rtc0)
[    1.281068] ingenic,mmc 13450000.msc: no support for card's volts
[    1.288204] ALSA device list:
[    1.291263]   No soundcards found.
[    1.295174] mmc0: error -22 whilst initialising MMC card
[    1.303451] VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
[    1.311143] Freeing unused kernel memory: 232K
[    1.342221] ingenic,mmc 13450000.msc: no support for card's volts
[    1.354759] mmc0: error -22 whilst initialising SDIO card
[    1.364430] ingenic,mmc 13450000.msc: no support for card's volts
[    1.370799] mmc0: error -22 whilst initialising MMC card
[    1.382228] ingenic,mmc 13460000.msc: card claims to support voltages below defined range
[    1.396745] mmc1: new SDIO card at address 0001
[    3.032303] random: nonblocking pool is initialized
```

gtxaspec, 2023-10-10:
Skip to kernel 6.4, faster.

archandanime, 2023-10-11:
Great work!!!

closed 2024-11-05 by gtxaspec