gtxaspec/wz_mini_hacks #611: v3 not booting up only red light flashing

dopeytree, 2023-09-13:
Trying this out but when I put the sd card in I only get a red flashing led light.
Something seems to be stopping it booting.. any ideas?

How long should it take to boot?

virmaior, 2023-09-14:
it should boot rather quickly.

Off the top of my head,

1. has the V3 been previously setup using the Wyze app?
2. Does the camera function normally without an SD card inserted?
3. What version of wyze firmware are you using?
4. is the SD card properly formatted as FAT32 (not ExFat)?
5. Does the root directory contain the `factory_...` file and the `wz_mini` directory?
6. Did you edit the wz_mini.conf file using a windows editor (often corrupts the formatting especially for people who don't much about that area)?
7. Which version of wz_mini_hacks did you use?

dopeytree, 2023-09-14:
OK I tried again this time using the unraid server (linux) to edit the wz_min.conf instead of text editor on mac and now it boots up.

The web server didn't seem to load but RTSP is working.

robughblah, 2023-09-14:
Ed,i recommend the BBEdit text editor for mac at https://www.barebones.com/products/bbedit/index.htmlyou can save text files with either Windows, Linux or Mac carriage return line feeds. And it doesn’t suck. Rob

biggt, 2023-09-14:
I am also having this issue with the V3 camera. 
I formatted the 128GB uSD card to FAT32 on a linux machine. 
After downloading the wz_mini_hacks zip to a linux machine, unzipping, copying the contents of SDROOT to the sd card, I modified the config file to what I want. 

After inserting the the sd card and plugging in the cam it sits at a red flashing light and does not say anything. 
I have also held the setup button and plugged in the cam and it still sits at a red flashing light. 

This is happening with a brand new camera out of the box (haven't even setup with Wyze app yet, so assuming something close to the 4.36.10.4054 firmware. 
The other camera is a v3 as well that has already been setup with the wyze app that is on 4.36.10.4054 firmware. 

Please advise on how to achieve this.

virmaior, 2023-09-14:
@biggt using wz_mini_hacks without first using wyze's app is non-trivial.
also in general 4.36.10.4054 is above the numbers most people use for wyze. would suggest downgrading the firmware and then using wyze;s app to setup the camera first.
if you're up for it, https://leo.leung.xyz/wiki/Wyze_Cam provides exceptional information on how to do setup a wyze camera without the app.

dopeytree, 2023-09-14:
Ok so you need to open the firmware.zip then you will find the file.bin that goes on the SD card... lol

biggt, 2023-09-14:
> @biggt using wz_mini_hacks without first using wyze's app is non-trivial. also in general 4.36.10.4054 is above the numbers most people use for wyze. would suggest downgrading the firmware and then [...]

Thanks for the direction. 
I have attempted to roll back the firmware just as the leo site and wyze sites both say, using the firmware that everyone most recommends. 
Even after putting demo_wc3.bin and even renaming to demo.bin, if I plug in without holding setup, it gets stuck at a red flashing light. 
If I hold setup and plug in, it will go to setup prompt, and eventually goes to red flashing light. 
At no point do I get a purple light.

dopeytree, 2023-09-14:
Are you definitely extracting the file first? I say this because the instructions say to rename the file but that is wrong. 

Download.
Exract the file.
Then rename the extracted file to demo_wc3.bin
Then hold down setup button while pluging the power in.
Light goes purple.
Then after a few mins light goes red.
Firmware has been downgraded.
Turn off.

No you can start copying the WZ hacks onto the SD card.

dopeytree, 2023-09-14:
Ok it appears they have installed something that makes it update the firmware when it connects to the internet...
It updates to 4.36.6.17

So I'll re install the early firmware and set up over SSH to avoid the internet.

biggt, 2023-09-14:
> Are you definitely extracting the file first? I say this because the instructions say to rename the file but that is wrong. (i.e - Do not rename a .zip to .bin like I did)
> 
> Download. Exract the file. Then rename the extracted file to demo_wc3.bin Then hold down setup button while pluging the power in. Light goes purple. Then after a few mins light goes red. Firmware has been downgraded. Turn off.
> 
> No you can start copying the WZ hacks onto the SD card.

Light never goes purple on 2 different cameras. One was previously setup with Wyze app, another without (so has no concept of my home wifi)

dopeytree, 2023-09-15:
Other thing to check is the SD card definitely fat32?

gtxaspec, 2023-09-15:
restore the camera firmware following the official restore instructions using this firmware:
https://download.wyzecam.com/firmware/v3/demo_wcv3_4.36.9.139.zip

confirm on the app that the camera is running the right firmware version, then you can proceed to follow the wz_mini instructions.

dopeytree, 2023-09-15:
If you use the app to set the camera up if updates the firmware.

See post above:
I had downgraded to 4.36.2.5
When adding it to the app it updates to 4.36.6.17

gtxaspec, 2023-09-15:
The app only forces you to upgrade if you install a really old firmware.  x.139 only prompts you to upgrade, but doesn't force you.

robughblah, 2023-09-16:
There is also a setting that defaults to On, under Device Info -> Firmware Version.I don’t recall seeing it for older firmware or when it appeared.Rob

dopeytree, 2023-09-17:
Annke have a mad sale on at the mo... have been testing these c500 3k cameras & the feed is much better can tell the sub stream to be higher quality which makes object detection almost perfect for frigate. Only £32  - which is around what I paid for the wyze v3 camera and thats 1080p... https://uk.annke.com/products/c500?variant=39783279394988

biggt, 2023-09-28:
> My V3s have been sensitive to the SD card. Even if I partition it smaller than 32gb, it won't work for certain SD cards and typically won't work for the sdXC cards. Seems to prefer microsdHC. I spent hours on trying to figure out how to get the XC cards I have to work before I just bought more HC cards and they work right away. This is for Wyze firmware and the wz_mini_hacks.

I had this exact same experience. I was attempting to do this with either 128gb or 256gb cards and neither worked. I ordered a bunch of 32gb cards and using the process specified, everything works, every time. Minus the reboot bug that seems to be going on now. [https://github.com/gtxaspec/wz_mini_hacks/issues/615](https://github.com/gtxaspec/wz_mini_hacks/issues/615)

nkm8, 2024-06-01:
I am experiencing this exact same issue:
- using the correct firmware version on a v3
- Tried the following SD cards, all formatted FAT32, all with stock contents stored at root: https://github.com/gtxaspec/wz_mini_hacks/tree/master/SD_ROOT
  - 256 GB 
  - 8 GB
  - 8 GB

Same red flashing light issue with all 3 SD cards. So far, I haven't found a combination that allows the camera to boot with wz_mini_hacks installed.

virmaior, 2024-06-01:
@nkm8  
1. is there any log file generated when an SD card is there on boot?
2. does it boot when there is no SD card?
3. did you initialize the camera with the wyze QR code? (they changed the security method so that it uses the hashed version of the wifi info after a certain version; if you downgrade back to where it's raw text,the wifi info has to be re-entered).

nkm8, 2024-06-01:
@virmaior 
1. How/where would I check - on the SD card?
2. Yes
3. Yes - cloud connection works fine (and I can see the camera connect to my router) as soon as I remove the SD card

virmaior, 2024-06-01:
should be on the card at `/wz_mini/log/wz_init.log`

nkm8, 2024-06-02:
On all 3 SD cards, `/wz_mini/log/` is empty 😢

virmaior, 2024-06-02:
what does the directory tree from the root look like on the SD cards?

nkm8, 2024-06-02:
Very interesting development - I was previously using my Windows 11 laptop to format and populate the SD cards. Windows is a bit finicky with formatting large drives as FAT32, so I was using DiskGenius for the 256 GB card and native Windows for the 8 GB cards.

Instead, I tried using my Chromebook, which natively supports formatting the SD cards as FAT32. I also used the Chromebook to populate the contents and I can confirm that all 3 SD cards are now working as expected in my v3.

I am not sure what the root cause with Windows is. Thanks for all your help - greatly appreciated!