gtxaspec/wz_mini_hacks #613: md5 checksum error when compiling for v1 pan and clarification of instructions for setup

dasb00ter, 2023-09-14:
`md5sum failed check, please supply a supported demo.zip `
after `./compile_image.sh pan`

I found I think somebody else using demo_pan_v1_4.10.8.1002 and renamed that to demo.zip and it compiled. Is this common knowledge around here? I think I may be getting in over my head a little.

In any case I copy the files from sd_root and the demo.bin generated and then I am little not sure. The directions say to hold down the resent button to flash the latest modified firmware. 

I want to keep the current firmware thats on there which is an rtsp firmware 4.29.4.**.
If I pull the card  will I be back to 4.29.4.** ?? or the stock firmware 4.10.8.1002 that I compiled against?

Sorry if I am not quite as astute as some of the others around here :)

jaws35smith, 2023-09-14:
@dasb00ter the compile_image.sh script has been updated to look for the newest firmware version and check there hash values. For the Pan v1 you will have to flash a the modified firmware to the camera so it will not be rtsp 4.29.4 it would be a modified version of the most recent firmware. Currently I believe there to be a bug with getting pan v1 to work. I have been unable to get the Wifi driver to load on 3 different pan v1 cameras.

dasb00ter, 2023-09-15:
Hmmmm so maybe the download is not clean and failing the check. I've used an older firmware demo_pan_v1_4.10.8.1002 to compile with. Do u think that will not have the WiFi problem...I mean are using this in the past right?

medlhed420, 2023-10-01:
Not sure if I should open a new issue, but this may be the same issue.

Trying to compile for Cam V2 abd getting "md5sum failed check, please supply a supported demo.zip"

I have downloaded the latest firmware ZIP published on Sept 14, 2023 
[4.9.9.2351](https://download.wyzecam.com/firmware/v2/demo_4.9.9.2351.bin.zip). 

I also tried the previous 3 releases.
[4.9.9.1851](https://download.wyzecam.com/firmware/v2/demo_4.9.9.1851.bin.zip)
[4.9.9.1574](https://download.wyzecam.com/firmware/v2/demo_4.9.9.1574.bin.zip)
[4.9.9.1472](https://s3.us-west-2.amazonaws.com/download.wyzecam.com/firmware/v2/demo_4.9.9.1472.bin.zip)

All give the checksum error.

dasb00ter, 2023-10-02:
I definitely could compile with a manually downloaded firmware that placed in the proper folder and renamed demo.zip. I guess u did all the regular trouble shooting. Made sure it was the proper firmware, tried a different connection and computer. I believe I used the firmware listed on the compatible firmwares that worked with webrtc sound although I had to source the wyze cam site link format from Reddit

jokesters, 2023-10-07:
I looked in the compile_image.sh file and the versions it says its checking the MD5 for is 
4.10.9.1851
4.10.9.1574
and so on, but the versions on the wyze site says they are
4.9.9.1851
4.9.9.1574

seems to me the versions are incorect in the compile_image.sh script.  I get checksum error as well..

cabinlab, 2023-10-29:
> I just looked again and realized the 4.10 versions are for the pan v1. the only v2 in the list is 4.9.8.1002 that wyze wont let us download anymore.

They removed the link from the main firmware page to discourage downloading the older versions, but the files are still there.

[V2 4.9.8.1002.zip](https://download.wyzecam.com/firmware/v2/demo_4.9.8.1002.zip)

md5 checksum passed with that one. 👍

eulaly, 2024-01-14:
this resolved the checksum error (supported firmwares are listed in compile_image.sh starting on line 48)

bobbywaz, 2024-12-19:
This issue is still present, 

b@MIMIKYU:/mnt/c/Users/b/Desktop/wz_mini_hacks-master/v2_install$ cat compile_image.sh | grep 4.10.
        version_found="pan v1 4.10.8.1002"
        version_found="pan v1 4.10.9.1433"
        version_found="pan v1 4.10.9.1472"
        version_found="pan v1 4.10.9.1574"
        version_found="pan v1 4.10.9.1851"
        
        
These are no longer on the website and I tried making them into a URL, it just said access died.