gtxaspec/wz_mini_hacks #615: With Hack WyzeCam3 looses wyze config if power is lost very often, and boots in pairing mode (blinking red led)

johnvprcam24573455623, 2023-09-17:
I guys, I love the wz_mini_hacks  however I have a serious problem, if the cam lose power, then most of the time will reboot in wyze pairing mode (red led blinking) so no RTSP available and basically offline.

these are the versions:
T31(WYZE_CAKP2JFUS) on 4.36.9.139 running wz_mini 

This is my config
[image]
[image]
[image]

gtxaspec, 2023-09-19:
This could be an issue with your micro SD card, do you have another to test with?

gtxaspec, 2023-09-28:
do you have self hosted mode enabled on all of the cameras you are having issues on?

biggt, 2023-09-28:
Yes. This is the exact config file I am using for all of my cameras. 

```
#wz_mini configuration

#### W E B CAMERA###
##THIS MODE DISABLES EVERYTHING AND IT WILL
## WORK AS A WEB CAMERA FOR YOUR PC ***ONLY***
WEB_CAM_ENABLE="false"
WEB_CAM_BIT_RATE="8000"
WEB_CAM_FPS_RATE="25"

#####NETWORKING#####
CUSTOM_HOSTNAME="WCV3"

ENABLE_USB_ETH="false"
ENABLE_USB_ETH_MODULE_AUTODETECT="true"
ENABLE_USB_ETH_MODULE_MANUAL=""
USB_ETH_MAC_ADDR=""

ENABLE_USB_DIRECT="false"
USB_DIRECT_MAC_ADDR="02:01:02:03:04:08"

ENABLE_USB_RNDIS="false"

ENABLE_IPV6="false"

ENABLE_WIREGUARD="false"
WIREGUARD_IPV4=""
WIREGUARD_PEER_ENDPOINT=""
WIREGUARD_PEER_PUBLIC_KEY=""
WIREGUARD_PEER_ALLOWED_IPS=""
WIREGUARD_PEER_KEEP_ALIVE=""

ENABLE_IPTABLES="false"

ENABLE_NFSv4="true"

ENABLE_RTL8189FS_DRIVER="true"
ENABLE_ATBM603X_DRIVER="true"

NTP_SERVER=""

##### NETWORK INTERFACE BONDING #####
BONDING_ENABLED="false"
BONDING_PRIMARY_INTERFACE="eth0"
BONDING_SECONDARY_INTERFACE="wlan0"
BONDING_LINK_MONITORING_FREQ_MS="100"
BONDING_DOWN_DELAY_MS="5000"
BONDING_UP_DELAY_MS="5000"
BONDING_FAIL_OVER_MAC="0"

#####ACCESSORIES#####
REMOTE_SPOTLIGHT="false"

####MOTOR OPTIONS####
MOTOR_DISABLE="false"
MOTOR_DEFAULT_X=""
MOTOR_DEFAULT_Y=""

#####VIDEO STREAM#####
RTSP_LOGIN="admin"
RTSP_PASSWORD="[REDACTED]"
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="true"

RTSP_LOW_RES_ENABLED="true"
RTSP_LOW_RES_ENABLE_AUDIO="false"

RTSP_AUTH_DISABLE="false"

VIDEO_0_MAX_BITRATE="2048"
VIDEO_0_TARGET_BITRATE="2048"
VIDEO_0_ENC_PARAMETER="2"
VIDEO_0_FPS="30"

VIDEO_1_MAX_BITRATE="1024"
VIDEO_1_TARGET_BITRATE="1024"
VIDEO_1_ENC_PARAMETER="2"
VIDEO_1_FPS="15"

GO2RTC_SERVER_ENABLED="true"
GO2RTC_HI_RES_ENABLED="true"
GO2RTC_HI_RES_ENABLE_AUDIO="true"
GO2RTC_LOW_RES_ENABLED="true"
GO2RTC_LOW_RES_ENABLE_AUDIO="false"

#####RTMP STREAMING#####
#### REQUIRES RTSP ####
### AND RTSP AUDIO ###
RTMP_STREAM_ENABLED="false"
RTMP_STREAM_FEED="video1_unicast"
RTMP_STREAM_SERVICE="youtube"
RTMP_STREAM_DISABLE_AUDIO="false"
RTMP_STREAM_YOUTUBE_KEY=""
RTMP_STREAM_TWITCH_KEY=""
RTMP_STREAM_FACEBOOK_KEY=""

#####GENERAL#####
LIBCALLBACK_ENABLE="true"
ENABLE_SWAP="true"
ENABLE_USB_STORAGE="true"
ENABLE_EXT4="true"
ENABLE_CIFS="true"
DISABLE_FW_UPGRADE="true"
AUDIO_PROMPT_VOLUME="50"
ENABLE_MP4_WRITE="true"
NIGHT_DROP_DISABLE="true"
ENABLE_FSCK_ON_BOOT="true"
ENABLE_CAR_DRIVER="false"
ENABLE_LOCAL_DNS="true"
ENABLE_CRONTAB="true"
ENABLE_SELFHOSTED_MODE="true"

#####DEBUG#####
#drops you to a shell via serial, doesn't load app_init.sh
DEBUG_ENABLED="false"
DEBUG_INITRAMFS_ENABLED="false"
DEBUG_PASSWORD="false"
ENABLE_SYSLOG_SAVE="false"

####WEB####
WEB_SERVER_ENABLED="true"
WEB_SERVER_AUTH="true"
WEB_SERVER_LOGIN="admin"
WEB_SERVER_PASSWORD="[REDACTED]"

#####SCRIPTING#####
CUSTOM_SCRIPT_PATH=""
```

johnvprcam24573455623, 2023-09-28:
This is my config with the same issue:

`#wz_mini configuration

#### W E B CAMERA###
##THIS MODE DISABLES EVERYTHING AND IT WILL
## WORK AS A WEB CAMERA FOR YOUR PC ***ONLY***
WEB_CAM_ENABLE="false"
WEB_CAM_BIT_RATE="8000"
WEB_CAM_FPS_RATE="25"

#####NETWORKING#####
CUSTOM_HOSTNAME="WCV3"

ENABLE_USB_ETH="false"
ENABLE_USB_ETH_MODULE_AUTODETECT="true"
ENABLE_USB_ETH_MODULE_MANUAL=""

ENABLE_USB_DIRECT="false"
USB_DIRECT_MAC_ADDR="02:01:02:03:04:08"

ENABLE_USB_RNDIS="false"

ENABLE_IPV6="false"

ENABLE_WIREGUARD="false"
WIREGUARD_IPV4=""
WIREGUARD_PEER_ENDPOINT=""
WIREGUARD_PEER_PUBLIC_KEY=""
WIREGUARD_PEER_ALLOWED_IPS=""
WIREGUARD_PEER_KEEP_ALIVE=""

ENABLE_IPTABLES="false"

ENABLE_NFSv4="false"

ENABLE_RTL8189FS_DRIVER="true"
ENABLE_ATBM603X_DRIVER="true"

NTP_SERVER=""

##### NETWORK INTERFACE BONDING #####
BONDING_ENABLED="false"
BONDING_PRIMARY_INTERFACE="eth0"
BONDING_SECONDARY_INTERFACE="wlan0"
BONDING_LINK_MONITORING_FREQ_MS="100"
BONDING_DOWN_DELAY_MS="5000"
BONDING_UP_DELAY_MS="5000"
BONDING_FAIL_OVER_MAC="0"

#####ACCESSORIES#####
REMOTE_SPOTLIGHT="false"
REMOTE_SPOTLIGHT_HOST="0.0.0.0"

#####VIDEO STREAM#####
RTSP_LOGIN="YYYY"
RTSP_PASSWORD="XXXXXX"
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="false"
RTSP_HI_RES_ENABLE_AUDIO="false"

RTSP_LOW_RES_ENABLED="false"
RTSP_LOW_RES_ENABLE_AUDIO="false"

RTSP_AUTH_DISABLE="false"

VIDEO_0_MAX_BITRATE=""
VIDEO_0_TARGET_BITRATE=""
VIDEO_0_ENC_PARAMETER=""
VIDEO_0_FPS=""

VIDEO_1_MAX_BITRATE=""
VIDEO_1_TARGET_BITRATE=""
VIDEO_1_ENC_PARAMETER=""
VIDEO_1_FPS=""

GO2RTC_SERVER_ENABLED="true"
GO2RTC_HI_RES_ENABLED="true"
GO2RTC_HI_RES_ENABLE_AUDIO="true"
GO2RTC_LOW_RES_ENABLED="true"
GO2RTC_LOW_RES_ENABLE_AUDIO="true"

#####RTMP STREAMING#####
#### REQUIRES RTSP ####
### AND RTSP AUDIO ###
RTMP_STREAM_ENABLED="false"
RTMP_STREAM_FEED="video1_unicast"
RTMP_STREAM_SERVICE="youtube"
RTMP_STREAM_DISABLE_AUDIO="false"
RTMP_STREAM_YOUTUBE_KEY=""
RTMP_STREAM_TWITCH_KEY=""
RTMP_STREAM_FACEBOOK_KEY=""

#####GENERAL#####
LIBCALLBACK_ENABLE="true"
ENABLE_SWAP="true"
ENABLE_USB_STORAGE="false"
ENABLE_EXT4="false"
ENABLE_CIFS="false"
DISABLE_FW_UPGRADE="true"
AUDIO_PROMPT_VOLUME="50"
ENABLE_MP4_WRITE="false"
NIGHT_DROP_DISABLE="false"
DISABLE_MOTOR="false"
ENABLE_FSCK_ON_BOOT="false"
ENABLE_CAR_DRIVER="false"
ENABLE_LOCAL_DNS="false"
ENABLE_CRONTAB="false"
ENABLE_SELFHOSTED_MODE="false"

#####DEBUG#####
#drops you to a shell via serial, doesn't load app_init.sh
DEBUG_ENABLED="false"
DEBUG_INITRAMFS_ENABLED="false"
DEBUG_PASSWORD="false"
ENABLE_SYSLOG_SAVE="false"

####WEB####
WEB_SERVER_ENABLED="true"
WEB_SERVER_AUTH="true"
WEB_SERVER_LOGIN="admin"
WEB_SERVER_PASSWORD="XXXXXXX"

#####SCRIPTING#####
CUSTOM_SCRIPT_PATH=""
`

gtxaspec, 2023-09-28:
@kohrar any ideas on this?  I don't personally use the self hosted mode.

kohrar, 2023-09-28:
That's interesting. I've been using my cameras in self hosted mode without them ever connecting to the internet and through many unexpected power cycles. The only reason the startup script gets triggered is if the WiFi configs aren't available under /configs.

After reviewing @johnvprcam24573455623's logs, I think I might have a hunch on what's going wrong.

The config file under `/configs/.user_config` is read by `iCamera` on start up. I recall the `iCamera` process will reset the WiFi settings if `bindOk` isn't set to 1. My startup script will try to set this to 1, but only if that line already exists. For a new camera that has no connection to the outside, this might not be happening. 

From the logs, this `bindOk` value isn't set at all as it should have been printed out at this line on the script https://github.com/gtxaspec/wz_mini_hacks/blob/4719cb563448a7a1e9b6f65ce42f7c99f12de6c2/SD_ROOT/wz_mini/etc/init.d/S20initsetup#L52

@johnvprcam24573455623, do you have SSH access to the camera? If you do, could you check if `/configs/.user_config` contains these lines? If they don't, could you add it in and try rebooting the camera?
```
[NET]
bindOk=1
p2pid=
r_encr=
```

johnvprcam24573455623, 2023-09-28:
This is my /configs/.user_config
It looks like it has the 3 lines

```
[NET]
bindOk=1
p2pid=1TXXXXXXXXXXXX111A
r_encr=MAtXXXXXXXXXXNX

[SETTING]
bitRate=120
timezone=-7

[CAMERA_INFO]
TZ_STR=PST+0800PDT+0700,M3.2.0/02:00:00,M11.1.0/02:00:00

> @johnvprcam24573455623, do you have SSH access to the camera? If you do, could you check if `/configs/.user_config` contains these lines? If they don't, could you add it in and try rebooting the camera?
> 
> ```
> [NET]
> bindOk=1
> p2pid=
> r_encr=
> ```

```

kohrar, 2023-09-28:
That looks good. 

Can you see whether `/configs` has the wifi settings saved? There should be a `.wifipasswd` and `.wifissid` with your WiFi SSID/password there. If they're all there, does a reboot somehow still cause the camera to come up without its WiFi settings?

johnvprcam24573455623, 2023-09-28:
Now it is working when I reboot, I have 8 cameras however 6 of them went in pairing mode while I 

there is only one line in my .wifipasswd

```
yXXXXXXXXXXDL
```

kohrar, 2023-09-29:
I reviewed your logs again and noticed this:
```
#####S20initsetup#####
grep: /configs/.user_config: No such file or directory
```

That would suggest that file wasn't there at startup which would explain why it would go into the 'wifi pairing' mode.

A couple things to try for the remaining cameras that you have issues with:
1. When they go into the 'wifi pairing' mode, can you scan the QR code containing your wifi configs and see if it comes back online? If it comes online, can you try rebooting (by disconnecting the power) and see if it goes back into the 'wifi pairing' mode again? (It shouldn't since the .user_config file should be created when the QR code is scanned)

2. Alternatively, could you check if /configs/.user_config on the other 6 cameras exist and have the bindOk line set? Once you're sure they exist, try rebooting (by disconnecting the power again) and see if it comes back up.

If the .user_config file disappears, then there's something wonky with the internal flash that's causing the file to disappear when power's abruptly lost. Perhaps a corrupt MTD filesystem? If this is the case, you could try reflashing the camera again and see if the issue goes away.

That's the only thing I can think of right now which could cause the behavior that you're seeing.

biggt, 2023-09-29:
> I reviewed your logs again and noticed this:
>
> ```
> #####S20initsetup#####
> grep: /configs/.user_config: No such file or directory
> ```
>
> That would suggest that file wasn't there at startup which would [...]

When I checked whether the .user_configs and .wifipasswd and .wifissid were present, they were not. Hence the error message in my logs. 
I manually created these with my info ... the camera did not reconnect to the network. 
I attempted to re-scan a QR code and it announces that it connected to the network, yet, I can see that in a constant ping test, the camera does not come up and even on my WiFi access points management console the MAC address is not populated, thus it's not actually trying to connect.

biggt, 2023-09-29:
> > I reviewed your logs again and noticed this:
> > ```
> > #####S20initsetup#####
> > grep: /configs/.user_config: No such file or directory
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > That would suggest that file wasn't there at startup which would explain why it would go into the 'wifi pairing' mode.
> > A couple things to try for the remaining cameras that you have issues with:
> > ```
> > 1. When they go into the 'wifi pairing' mode, can you scan the QR code containing your wifi configs and see if it comes back online? If it comes online, can you try rebooting (by disconnecting the power) and see if it goes back into the 'wifi pairing' mode again? (It shouldn't since the .user_config file should be created when the QR code is scanned)
> > 
> > 2. Alternatively, could you check if /configs/.user_config on the other 6 cameras exist and have the bindOk line set? Once you're sure they exist, try rebooting (by disconnecting the power again) and see if it comes back up.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > If the .user_config file disappears, then there's something wonky with the internal flash that's causing the file to disappear when power's abruptly lost. Perhaps a corrupt MTD filesystem? If this is the case, you could try reflashing the camera again and see if the issue goes away.
> > That's the only thing I can think of right now which could cause the behavior that you're seeing.
> 
> When I checked whether the .user_configs and .wifipasswd and .wifissid were present, they were not. Hence the error message in my logs. I manually created these with my info ... the camera did not reconnect to the network. I attempted to re-scan a QR code and it announces that it connected to the network, yet, I can see that in a constant ping test, the camera does not come up and even on my WiFi access points management console the MAC address is not populated, thus it's not actually trying to connect. Restarting the camera does not help in this case. While after it claimed that it connected to the network and then I restarted the camera, it did not go back into pairing mode... it's very clearly not on the network.

Funny thing is that after this, the camera is just constantly sitting there flashing blue light and still unavailable on the network.

kohrar, 2023-09-29:
> I manually created these with my info ... the camera did not reconnect to the network.

How did you manually create this if it wasn't connected to the network? 

> I attempted to re-scan a QR code and it announces that it connected to the network, yet, I can see that in a constant ping test, the camera does not come up and even on my WiFi access points management console the MAC address is not populated, thus it's not actually trying to connect.

Could you try these steps in this exact order:
1. Press the setup button. When you hear it say "ready to connect", the camera will be in the setup mode and its WiFi settings will have been erased. If you don't hear "ready to connect", then power cycle it first before trying again.
2. Power-cycle (disconnect / reapply power) the camera. It should now come up without a connection. Press the setup button again. It should say "ready to connect".
3. Scan the QR code with the new WiFi settings. It should say "wifi connected" and connect to your network. At this point, you should be able to ping the camera.

If that is still not working, repeat the steps above without the SD card containing wz_mini_hacks and see if the behavior is any different.

Also -- try to make a new QR code using https://codepen.io/ril3y/full/gXyzmO/ just to rule out that you didn't accidentally put in the wrong credentials in your QR code.

> Funny thing is that after this, the camera is just constantly sitting there flashing blue light and still unavailable on the network.

This may be an issue with the current self hosted mode. This may be an issue with the script where it terminates after finding WiFi credentials rather than looping forever and wasting resources. I can replicate this with these steps:
1. power cycle
2. press the setup button
3. scan QR code, (it connect and is pingable; I assume scanning a QR code with bad WiFi settings would also trigger this behavior)
4. press the setup button again (which wipes its connection settings and is no longer pingable. It does not say "ready to connect" on this button press either)
5. The camera enters this failure state. LED is blinking, setup button is unresponsive, and is no longer pingable.

I can recover from this state by power cycling, pressing the setup button once, and scanning the QR code, however.

biggt, 2023-10-01:
> > I manually created these with my info ... the camera did not reconnect to the network.
> 
> How did you manually create this if it wasn't connected to the network?
I unplugged the camera, mounted the sd card on another computer and created the files

> Could you try these steps in this exact order:
>
>     1. Press the setup button. When you hear it say "ready to connect", the camera will be in the setup mode and its WiFi settings will have been [...]

It scans the QR code (hear the ding) and then just sits there for like 2 minutes trying to connect. It never connects, nothing shows on ping. 
 
> If that is still not working, repeat the steps above without the SD card containing wz_mini_hacks and see if the behavior is any different.

I took out the SD card with wz_mini_hacks, rescanned the QR code and it scans it, and shortly thereafter I see that it connects to the network (responds to ping) for like 15 seconds, then says it couldn't connect to the network, disconnects (stops being ping-able). 

> Also -- try to make a new QR code using https://codepen.io/ril3y/full/gXyzmO/ just to rule out that you didn't accidentally put in the wrong credentials in your QR code.
 - put in my credentials to this qr code generator (without wz_mini_hacks card) and same as above where it temporarily connects, then says that connection timed out, and disconnects. (I'm assuming it's looking for the Wyze connection, but even when scanning the QR code the wyze app generates doesn't allow the camera to connect properly. 
 - put in the wz_mini_hacks card and rescanned using this qr code generator (haven't touched the page since the above test) ... it scans the code, shortly thereafter says wifi connected, and then interrupts itself trying to say it can't connect... and does not respond through all of this. 
 - after this second test, i power cycled, pressed setup, scanned qr code (again haven't touched the webpage) and it sits there and doesn't do anything. 

> This may be an issue with the current self hosted mode. This may be an issue with the script where it terminates after finding WiFi credentials rather than looping forever and wasting resources. I [...]

while it was sitting there doing nothing from last test above, I pressed setup button, says ready to connect but is unable to scan the qr code. it doesn't ding, doesn't say anything. I repeated this 3 times and it will not scan qr code again.

Habile2019, 2023-10-15:
Just to say I see this very behaviour. Seemingly randomly the entire /configs content 'disappears'. I can tell when this has occurred as the timezone is lost and the timestamp shows UTC on the camera image.

When this happens I usually ssh in and copy the configs from `/opt/wyze_config_backup/configs/` to `/configs`, otherwise I know on reboot it will need to be paired again using the app.

I'll keep trying to debug what wipes out `/configs`.

awdark, 2023-11-02:
That's an interesting thing to notice with the time zone.  I can confirm my camera was in UTC and the folder was basically empty,

johnvpr, 2023-11-07:
it just happened again to me this is the log:

`+ echo 'welcome to wz_mini initramfs'
welcome to wz_mini initramfs
+ mount -t devtmpfs devtmpfs /dev
+ mount -t proc proc /proc
+ mount -t sysfs sysfs /sys
+ sleep 1
+ echo 'Check for T20 platform'
Check for T20 platform
+ '[' -b /dev/mtdblock10 ]
+ echo 'Not T20'
Not T20
+ '[' -d /sys/class/gpio/gpio38 ]
+ echo 38
+ echo out
+ echo 1
+ '[' -d /sys/class/gpio/gpio39 ]
+ echo 39
+ echo out
+ echo 1
+ mkdir -p /sdcard
+ mount -t vfat /dev/mmcblk0p1 /sdcard -o 'rw,umask=0000,dmask=0000'
+ '[' -f /sdcard/wz_mini/wz_mini.conf ]
+ source /sdcard/wz_mini/wz_mini.conf
+ WEB_CAM_ENABLE=false
+ WEB_CAM_BIT_RATE=8000
+ WEB_CAM_FPS_RATE=25
+ CUSTOM_HOSTNAME=WCV3
+ ENABLE_USB_ETH=false
+ ENABLE_USB_ETH_MODULE_AUTODETECT=true
+ ENABLE_USB_ETH_MODULE_MANUAL=
+ ENABLE_USB_DIRECT=false
+ USB_DIRECT_MAC_ADDR=02:01:02:03:04:08
+ ENABLE_USB_RNDIS=false
+ ENABLE_IPV6=false
+ ENABLE_WIREGUARD=false
+ WIREGUARD_IPV4=
+ WIREGUARD_PEER_ENDPOINT=
+ WIREGUARD_PEER_PUBLIC_KEY=
+ WIREGUARD_PEER_ALLOWED_IPS=
+ WIREGUARD_PEER_KEEP_ALIVE=
+ ENABLE_IPTABLES=false
+ ENABLE_NFSv4=false
+ ENABLE_RTL8189FS_DRIVER=true
+ ENABLE_ATBM603X_DRIVER=true
+ NTP_SERVER=
+ BONDING_ENABLED=false
+ BONDING_PRIMARY_INTERFACE=eth0
+ BONDING_SECONDARY_INTERFACE=wlan0
+ BONDING_LINK_MONITORING_FREQ_MS=100
+ BONDING_DOWN_DELAY_MS=5000
+ BONDING_UP_DELAY_MS=5000
+ BONDING_FAIL_OVER_MAC=0
+ REMOTE_SPOTLIGHT=false
+ REMOTE_SPOTLIGHT_HOST=0.0.0.0
+ RTSP_LOGIN=john
+ RTSP_PASSWORD=yS2CnYo2L
+ RTSP_PORT=8554
+ RTSP_HI_RES_ENABLED=false
+ RTSP_HI_RES_ENABLE_AUDIO=false
+ RTSP_LOW_RES_ENABLED=false
+ RTSP_LOW_RES_ENABLE_AUDIO=false
+ RTSP_AUTH_DISABLE=false
+ VIDEO_0_MAX_BITRATE=
+ VIDEO_0_TARGET_BITRATE=
+ VIDEO_0_ENC_PARAMETER=
+ VIDEO_0_FPS=
+ VIDEO_1_MAX_BITRATE=
+ VIDEO_1_TARGET_BITRATE=
+ VIDEO_1_ENC_PARAMETER=
+ VIDEO_1_FPS=
+ GO2RTC_SERVER_ENABLED=true
+ GO2RTC_HI_RES_ENABLED=true
+ GO2RTC_HI_RES_ENABLE_AUDIO=true
+ GO2RTC_LOW_RES_ENABLED=true
+ GO2RTC_LOW_RES_ENABLE_AUDIO=true
+ RTMP_STREAM_ENABLED=false
+ RTMP_STREAM_FEED=video1_unicast
+ RTMP_STREAM_SERVICE=youtube
+ RTMP_STREAM_DISABLE_AUDIO=false
+ RTMP_STREAM_YOUTUBE_KEY=
+ RTMP_STREAM_TWITCH_KEY=
+ RTMP_STREAM_FACEBOOK_KEY=
+ LIBCALLBACK_ENABLE=true
+ ENABLE_SWAP=true
+ ENABLE_USB_STORAGE=false
+ ENABLE_EXT4=false
+ ENABLE_CIFS=false
+ DISABLE_FW_UPGRADE=true
+ AUDIO_PROMPT_VOLUME=50
+ ENABLE_MP4_WRITE=false
+ NIGHT_DROP_DISABLE=false
+ DISABLE_MOTOR=false
+ ENABLE_FSCK_ON_BOOT=false
+ ENABLE_CAR_DRIVER=false
+ ENABLE_LOCAL_DNS=false
+ ENABLE_CRONTAB=false
+ ENABLE_SELFHOSTED_MODE=false
+ DEBUG_ENABLED=false
+ DEBUG_INITRAMFS_ENABLED=false
+ DEBUG_PASSWORD=false
+ ENABLE_SYSLOG_SAVE=false
+ WEB_SERVER_ENABLED=true
+ WEB_SERVER_AUTH=true
+ WEB_SERVER_LOGIN=admin
+ WEB_SERVER_PASSWORD=XXXXXX
+ CUSTOM_SCRIPT_PATH=
+ '[' -f /sdcard/wz_mini/wz_mini.conf ]
+ source /sdcard/wz_mini/wz_mini.conf
+ WEB_CAM_ENABLE=false
+ WEB_CAM_BIT_RATE=8000
+ WEB_CAM_FPS_RATE=25
+ CUSTOM_HOSTNAME=WCV3
+ ENABLE_USB_ETH=false
+ ENABLE_USB_ETH_MODULE_AUTODETECT=true
+ ENABLE_USB_ETH_MODULE_MANUAL=
+ ENABLE_USB_DIRECT=false
+ USB_DIRECT_MAC_ADDR=02:01:02:03:04:08
+ ENABLE_USB_RNDIS=false
+ ENABLE_IPV6=false
+ ENABLE_WIREGUARD=false
+ WIREGUARD_IPV4=
+ WIREGUARD_PEER_ENDPOINT=
+ WIREGUARD_PEER_PUBLIC_KEY=
+ WIREGUARD_PEER_ALLOWED_IPS=
+ WIREGUARD_PEER_KEEP_ALIVE=
+ ENABLE_IPTABLES=false
+ ENABLE_NFSv4=false
+ ENABLE_RTL8189FS_DRIVER=true
+ ENABLE_ATBM603X_DRIVER=true
+ NTP_SERVER=
+ BONDING_ENABLED=false
+ BONDING_PRIMARY_INTERFACE=eth0
+ BONDING_SECONDARY_INTERFACE=wlan0
+ BONDING_LINK_MONITORING_FREQ_MS=100
+ BONDING_DOWN_DELAY_MS=5000
+ BONDING_UP_DELAY_MS=5000
+ BONDING_FAIL_OVER_MAC=0
+ REMOTE_SPOTLIGHT=false
+ REMOTE_SPOTLIGHT_HOST=0.0.0.0
+ RTSP_LOGIN=john
+ RTSP_PASSWORD=yS2CnYo2L
+ RTSP_PORT=8554
+ RTSP_HI_RES_ENABLED=false
+ RTSP_HI_RES_ENABLE_AUDIO=false
+ RTSP_LOW_RES_ENABLED=false
+ RTSP_LOW_RES_ENABLE_AUDIO=false
+ RTSP_AUTH_DISABLE=false
+ VIDEO_0_MAX_BITRATE=
+ VIDEO_0_TARGET_BITRATE=
+ VIDEO_0_ENC_PARAMETER=
+ VIDEO_0_FPS=
+ VIDEO_1_MAX_BITRATE=
+ VIDEO_1_TARGET_BITRATE=
+ VIDEO_1_ENC_PARAMETER=
+ VIDEO_1_FPS=
+ GO2RTC_SERVER_ENABLED=true
+ GO2RTC_HI_RES_ENABLED=true
+ GO2RTC_HI_RES_ENABLE_AUDIO=true
+ GO2RTC_LOW_RES_ENABLED=true
+ GO2RTC_LOW_RES_ENABLE_AUDIO=true
+ RTMP_STREAM_ENABLED=false
+ RTMP_STREAM_FEED=video1_unicast
+ RTMP_STREAM_SERVICE=youtube
+ RTMP_STREAM_DISABLE_AUDIO=false
+ RTMP_STREAM_YOUTUBE_KEY=
+ RTMP_STREAM_TWITCH_KEY=
+ RTMP_STREAM_FACEBOOK_KEY=
+ LIBCALLBACK_ENABLE=true
+ ENABLE_SWAP=true
+ ENABLE_USB_STORAGE=false
+ ENABLE_EXT4=false
+ ENABLE_CIFS=false
+ DISABLE_FW_UPGRADE=true
+ AUDIO_PROMPT_VOLUME=50
+ ENABLE_MP4_WRITE=false
+ NIGHT_DROP_DISABLE=false
+ DISABLE_MOTOR=false
+ ENABLE_FSCK_ON_BOOT=false
+ ENABLE_CAR_DRIVER=false
+ ENABLE_LOCAL_DNS=false
+ ENABLE_CRONTAB=false
+ ENABLE_SELFHOSTED_MODE=false
+ DEBUG_ENABLED=false
+ DEBUG_INITRAMFS_ENABLED=false
+ DEBUG_PASSWORD=false
+ ENABLE_SYSLOG_SAVE=false
+ WEB_SERVER_ENABLED=true
+ WEB_SERVER_AUTH=true
+ WEB_SERVER_LOGIN=admin
+ WEB_SERVER_PASSWORD=XXXX
+ CUSTOM_SCRIPT_PATH=
+ '[[' false '==' true ]]
+ echo 'initramfs debug disabled'
initramfs debug disabled
+ initram_init
+ mkdir /wz
+ mount -t squashfs /dev/mtdblock2 /wz
+ '[[' false '==' true ]]
+ '[' '!' -f /sdcard/wz_mini/etc/init.d/wz_init.sh ]
+ echo 'Loading wz_mini...'
Loading wz_mini...
+ mkdir -p /wz/dev
+ mkdir -p /wz/sys
+ mount --move /dev /wz/dev
+ mount --move /sys /wz/sys
+ mount --move /proc /wz/proc
+ mount --move /sdcard /wz/opt
+ cp /tmp/wz_mini_initramfs.log /wz/opt/
`

stncttr908, 2023-11-27:
Also having this problem with several cameras. The time zone resetting is a giveaway, so I try to copy the config out of /opt/wyze_config_backups over to /configs, which fixes it up for a while, but then it seems like /configs gets blasted seemingly at random. I might have to put a cron entry in just to keep refreshing that directory. This sometimes seems to prevent the need to re-pair the camera with the Wyze app, sometimes not.

vic4news, 2023-11-28:
I am experiencing this problem as well. Most of the files in /configs disappear randomly.  It is easy to tell if this has happened because the time displayed on the camera is wrong. If the camera loses power or is rebooted when the /configs files are missing then the camera will enter pairing mode.  Copying the files from the backup always fixes the issue for me as long as the camera wasn’t restarted.

kzaoaai, 2023-11-29:
I noticed that the files disappear from /configs/ when I access the webserver page at cgi-bin/config.cgi

They will disappear the second I access it, without any settings modifications.

gtxaspec, 2023-11-29:
@virmaior checkout @kzaoaai's comment above, I don't have access to a wz_mini setup at the moment for testing.

Habile2019, 2023-11-30:
> @kzaoaai That's really bad behavior. I'll try to figure this out.
> 
> can you tell me
> 
> 1. what version wz_mini you'er running?
> 2. whether you're using the cam in wifi mode or over ethernet or usb?

If it helps, as I also see this behaviour, here's my version:

- T31(WYZE_CAKP2JFUS) on 4.36.9.139 running wz_mini Sat Sep 30 11:46:23 AM PDT 2023
- WiFi

I'm wondering if `/configs` is getting unmounted somehow.

kzaoaai, 2023-11-30:
@virmaior 
1. Latest, Nov 20 build
2. Built-in WIFI

virmaior, 2023-11-30:
@Habile2019 

I experience periodic `/configs` clobbering on some but not all of my cameras -- even over spans of time where I don't use `/cgi-bin/config.cgi`

I'll try to look through the code but won't be at a place where I can for about 12 hours.

My current hypothesis is that the cameras can get overloaded in a way that somehow causes this, because those cameras are using the longest micro-usb cables.

If @kzaoaai 's cameras get this *every single time* that `/cgi-bin/config.cgi` is accessed that actually makes this easier to debug as I can only sporadically create this my self.
Does it still clobber if you change line 16 of `/cgi-bin/config.cgi` to exit?

```
#!/bin/sh
# This serves a rudimentary webpage based on wz_mini.conf
. /opt/wz_mini/www/cgi-bin/shared.cgi

test_area_access config
title="$camver on $camfirmware running wz_mini $hackver as $HOSTNAME"
updated=false

echo "HTTP/1.1 200"
echo -e "Content-type: text/html\n\n"
echo ""

exit
```
?
if so that basically isolates the problem to `/wz_mini/www/cgi-bin/shared.cgi`

also, does it crash if you use `jpeg.cgi` ?

vic4news, 2023-11-30:
Current version:
T31(WYZE_CAKP2JFUS) on 4.36.9.139 running wz_mini Sun Oct 15 12:16:27 AM PDT 2023
But was occurring on earlier versions of wz_mini as well. 

Using WiFi

Habile2019, 2023-11-30:
> My current hypothesis is that the cameras can get overloaded in a way that somehow causes this, because those cameras are using the longest micro-usb cables.

Thanks. I'll try to pay attention to `dmesg` and uptime etc. when it occurs again. My events are also usually independent from `/cgi-bin/config.cgi`, and to be honest haven't done it in a while. Possibly helped by turning off event recording, not sure.

kzaoaai, 2023-11-30:
so I inadvertently cleared cache from Safari by quitting and re-opening, and was presented with the screen saying something like "WEBSERVER Options missing, use WEB_SERVER_OPTIONS="cam config car jpeg multicam diag status...".

I added this line to the conf file, and now the issue is not happening everytime I access the config.cgi page.

> My current hypothesis is that the cameras can get overloaded in a way that somehow causes this,

However, leaving this page open in the foreground does definitely cause the config files to disappear more often. Maybe having the page open adds stress causing this.

virmaior, 2023-11-30:
@kzaoaai 
leaving the page open the only additional strain is the `jpeg.cgi` image refreshing (nothing else on the page connects to the server after initial load until you submit something).

virmaior, 2023-12-02:
some further progress on this.

for cameras that are still using v4l2rtspserver

`echo jpeg | /usr/bin/nc localhost 4000` (https://github.com/mnakada/atomcam_tools/blob/4568a2c5a7a6f0cf9155a627031ec906034e60ba/overlay_rootfs/var/www/cgi-bin/get_jpeg.cgi#L2) runs significantly faster that `cmd jpeg`

would like to figure if go2rtc has a similar fast image capture to jpeg.

stncttr908, 2023-12-03:
> it works for normal streams as well.

Thanks, I must be doing something wrong, because it returns 

> exit status 1

When I call http://[camera hostname]:1984/api/frame.jpeg?src=1080p (I assume that's the stream name here)

gtxaspec, 2023-12-03:
@stncttr908 are you using the latest versions?

virmaior, 2023-12-15:
@gtxaspec  is  https://github.com/gtxaspec/wz_mini_hacks/issues/10 FiveLeavesLeft's code implemented?

I've been looking to figure out _how_ jpeg files are created by `cmd jpeg` to see if there's an issue and I just noticed that FiveLeavesLeft states his code writes to `/config` because its rw ... and well /config is getting clobbered if you the jpeg code is called too frequently.

I haven't used C since about 2002 but if I'm grasping the code correctly (https://github.com/mnakada/atomcam_tools/blob/4568a2c5a7a6f0cf9155a627031ec906034e60ba/custompackages/package/ingenic_samples/include/imp/imp_encoder.h#L656 , https://github.com/mnakada/atomcam_tools/blob/4568a2c5a7a6f0cf9155a627031ec906034e60ba/libcallback/jpeg.c )

the centerpiece is `ret = IMP_Encoder_GetStream(ENC_JPEG_CHANNEL, &stream, 1);` where &stream is being passed such that the data gets put in there, ENC_JPEG_CHANNEL is presumably a constant and then the result is getting written somewhere.

Habile2019, 2023-12-15:
> the centerpiece is `ret = IMP_Encoder_GetStream(ENC_JPEG_CHANNEL, &stream, 1);` where &stream is being passed such that the data gets put in there, ENC_JPEG_CHANNEL is presumably a constant and then the result is getting written somewhere.

Interesting. I can only add that I do believe the issue only occurs after accessing the web interface where the snapshot appears, so it's a good candidate!

I also know it's a lot easier to break stuff unintentionally in `C` than it is in `bash`!

gtxaspec, 2023-12-15:
This is how we capture jpeg:
https://github.com/gtxaspec/libcallback/blob/master/jpeg.c
we don't do anythign with `/config` at all, it's all done in memory.  Even if we were to hammer the endpoint for jpeg images, I don't think this has anything to do with the configs issue though.

amandamana, 2024-02-02:
I experienced /configs/ getting repeated wiped on a couple v3s. I can't reliably reproduce it but am sure it's related to the webserver. There may be some state that it gets into, perhaps after editing the wz_mini.conf or issuing the reboot command, or perhaps related to some LAN "flood" traffic or browser requests from open pages. In one case, after one v3 started clobbering /configs/, it seems to trigger the issue another camera. I've disabled the web servers for now. I'll update this comment if it happens with them off or if I find time to figure out how to trigger the wipe using the web server.

virmaior, 2024-02-02:
> I experienced /configs/ getting repeated wiped on a couple v3s. I can't reliably reproduce it but am sure it's related to the webserver. 

> There may be some state that it gets into, perhaps after editing the wz_mini.conf or issuing the reboot command, or perhaps related to some LAN "flood" traffic or browser requests from open pages.

I don't think it's related to the `reboot` command or the editing of the config at this point.  In part, because I've seen the behavior and don't edit the config and one contributor pointed out that they  get this problem *when they leave the page open*.  This led to my previous hypothesis that `jpeg.cgi` and some issue in the screenshot render was causing it, but after adding in some debug code, it seems like the clobber happens outside of production of the jpeg.

My current working theory is that when the busybox httpd server gets overloaded something is wiping `/configs`, but I don't know why that would be the case  Or alternately the TCP/IP stack has some sort of issue.

Other possibility is that the (b)ash code I wrote for the web page is clobbering it but nothing in that code touches `/configs` leaving me lost as to how it would clobber it. 

> In one case, after one v3 started clobbering /configs/, it seems to trigger the issue on another camera. 

This is an intriguing claim.  It's hard to think how this would happen.

> I've disabled the web servers for now. I'll update this comment if it happens with them off or if I find time to figure out how to trigger the wipe using the web server.

mrdarrengriffin, 2024-10-29:
Any further progress on this? Running in isolated mode, goes into pairing mode upon reboot. Thanks!

Using a v3 T31 WYZE_CAKP2JFUS - FW 4.36.13.0416

mrdarrengriffin, 2024-10-29:
I just realised the wrong firmware is being applied on boot. Not sure why, I have the correct demo-wcv3.bin on there. I'll come back to this once I've fixed that

mrdarrengriffin, 2024-10-29:
I solved my issue by removing the demo-wcv3.bin from the SD card after the firmware downgrade

Cightline, 2024-11-20:
> I solved my issue by removing the demo-wcv3.bin from the SD card after the firmware downgrade

I think this was my issue as well. I'll be removing the leftover *.bins from now on.