gtxaspec/wz_mini_hacks #621: wz_mini.conf: Windows line ending repair testers needed.

gtxaspec, 2023-09-19:
Latest release includes the `S00config` init.d script to check for windows line endings in wz_mini.conf and corrects them.   This should help users against invalid configurations.

I am away from a windows machine at the moment, if anyone can test, it would be appreciated!

archandanime, 2023-09-26:
I did these steps on my Wyze Cam v3:
1. Pull `wz_mini.conf` from Camerera with scp
```bash
scp -O root@192.168.10.20:/opt/wz_mini/wz_mini.conf wz_mini-orig.conf
```
2. Reboot to Windows, open the file with Notepad, copy all and paste to a new file `wz_mini-win.conf`

3. Reboot to Linux, 2 files have different hashes:
```bash
> md5sum wz_mini-orig.conf wz_mini-win.conf
fa618ee11f64a6d39433fc7e8c088de6  wz_mini-orig.conf
c315e6146efbfa56168d343de26d51c0  wz_mini-win.conf
```
Inspect 2 files with hexdump:
```bash
> head -n 10 wz_mini-orig.conf | hexdump -C > hexdump+wz_mini-orig.conf
> head -n 10 wz_mini-win.conf | hexdump -C > hexdump_wz_mini-win.conf
> diff hexdump_wz_mini-win.conf hexdump+wz_mini-orig.conf
2,15c2,14
< 00000010  72 61 74 69 6f 6e 0d 0a  0d 0a 0d 0a 23 23 23 23  |ration......####|
< 00000020  20 57 20 45 20 42 20 43  41 4d 45 52 41 23 23 23  | W E B CAMERA###|
< 00000030  0d 0a 23 23 54 48 49 53  20 4d 4f 44 45 20 44 49  |..##THIS MODE DI|
< 00000040  53 41 42 4c 45 53 20 45  56 45 52 59 54 48 49 4e  |SABLES EVERYTHIN|
< 00000050  47 20 41 4e 44 20 49 54  20 57 49 4c 4c 0d 0a 23  |G AND IT WILL..#|
< 00000060  23 20 57 4f 52 4b 20 41  53 20 41 20 57 45 42 20  |# WORK AS A WEB |
< 00000070  43 41 4d 45 52 41 20 46  4f 52 20 59 4f 55 52 20  |CAMERA FOR YOUR |
< 00000080  50 43 20 2a 2a 2a 4f 4e  4c 59 2a 2a 2a 0d 0a 57  |PC ***ONLY***..W|
< 00000090  45 42 5f 43 41 4d 5f 45  4e 41 42 4c 45 3d 22 66  |EB_CAM_ENABLE="f|
< 000000a0  61 6c 73 65 22 0d 0a 57  45 42 5f 43 41 4d 5f 42  |alse"..WEB_CAM_B|
< 000000b0  49 54 5f 52 41 54 45 3d  22 38 30 30 30 22 0d 0a  |IT_RATE="8000"..|
< 000000c0  57 45 42 5f 43 41 4d 5f  46 50 53 5f 52 41 54 45  |WEB_CAM_FPS_RATE|
< 000000d0  3d 22 32 35 22 0d 0a 0d  0a                       |="25"....|
< 000000d9
---
> 00000010  72 61 74 69 6f 6e 0a 0a  0a 23 23 23 23 20 57 20  |ration...#### W |
> 00000020  45 20 42 20 43 41 4d 45  52 41 23 23 23 0a 23 23  |E B CAMERA###.##|
> 00000030  54 48 49 53 20 4d 4f 44  45 20 44 49 53 41 42 4c  |THIS MODE DISABL|
> 00000040  45 53 20 45 56 45 52 59  54 48 49 4e 47 20 41 4e  |ES EVERYTHING AN|
> 00000050  44 20 49 54 20 57 49 4c  4c 0a 23 23 20 57 4f 52  |D IT WILL.## WOR|
> 00000060  4b 20 41 53 20 41 20 57  45 42 20 43 41 4d 45 52  |K AS A WEB CAMER|
> 00000070  41 20 46 4f 52 20 59 4f  55 52 20 50 43 20 2a 2a  |A FOR YOUR PC **|
> 00000080  2a 4f 4e 4c 59 2a 2a 2a  0a 57 45 42 5f 43 41 4d  |*ONLY***.WEB_CAM|
> 00000090  5f 45 4e 41 42 4c 45 3d  22 66 61 6c 73 65 22 0a  |_ENABLE="false".|
> 000000a0  57 45 42 5f 43 41 4d 5f  42 49 54 5f 52 41 54 45  |WEB_CAM_BIT_RATE|
> 000000b0  3d 22 38 30 30 30 22 0a  57 45 42 5f 43 41 4d 5f  |="8000".WEB_CAM_|
> 000000c0  46 50 53 5f 52 41 54 45  3d 22 32 35 22 0a 0a     |FPS_RATE="25"..|
> 000000cf
 ```
even they seem to look the same:
```bash
#wz_mini configuration

#### W E B CAMERA###
##THIS MODE DISABLES EVERYTHING AND IT WILL
## WORK AS A WEB CAMERA FOR YOUR PC ***ONLY***
WEB_CAM_ENABLE="false"
WEB_CAM_BIT_RATE="8000"
WEB_CAM_FPS_RATE="25"

```
They are basically the same file but on Linux newline is `0a`, on Windows is `0d 0a`
4. Copy the new wz_mini.conf to the Camera over scp and reboot

----------------
Result: The camera does not work anymore even waiting 5 minutes. I did the same thing on my second camera but the result is the same.
I'm going to check boot log from serial connection to see what caused the problem.

archandanime, 2023-09-26:
I have kernel panic on my terminal when booting with `wz_mini.conf` with Windows' newlines:
[wz_mini_initramfs.log](https://github.com/gtxaspec/wz_mini_hacks/files/12726278/wz_mini_initramfs.log)
[wcv3_boot.log](https://github.com/gtxaspec/wz_mini_hacks/files/12726280/wcv3_boot.log)
Then I replaced `wz_mini.conf` with the previous one, it worked normally.

gtxaspec, 2023-09-26:
Ok thank you, I'll have a fix later today.

archandanime, 2023-09-26:
On my Linux laptop, dos2unix worked:
```bash
> md5sum wz_mini-win.conf
c315e6146efbfa56168d343de26d51c0  wz_mini-win.conf
> dos2unix wz_mini-win.conf
dos2unix: converting file wz_mini-win.conf to Unix format...
> md5sum wz_mini-win.conf wz_mini-orig.conf
fa618ee11f64a6d39433fc7e8c088de6  wz_mini-win.conf
fa618ee11f64a6d39433fc7e8c088de6  wz_mini-orig.conf
```
I think you... forgot to include dos2unix binary in SD_ROOT @@

gtxaspec, 2023-09-26:
The problem is dos2unix runs after initramfs... Before the file has been converted

archandanime, 2023-09-26:
Could you elaborate, please? Did you mean that initramfs's busybox doesn't include dos2unix but busybox on SD_ROOT does?

gtxaspec, 2023-09-26:
S00config (the init.d script) is run at camera boot.  When the initramfs tries to read the config file, it does this in the "pre-boot" environment so to speak, in which init.d has not run yet, so the repair script has not been run.  The script being in a corrupted state with windows line endings breaks the initramfs loading script.

archandanime, 2023-09-29:
Maybe we can let initramfs init run `dos2unix` before reading the file. busybox initramfs init can be compiled with `dos2unix` or we can use wz_mini busybox to run `/sdcard/wz_mini/bin/busybox dos2unix /sdcard/wz_mini/wz_mini.conf`, I think so.