gtxaspec/wz_mini_hacks #623: Web Interface missing options

sandstormkeshav, 2023-09-22:
When I use the web interface of the camera, this is the only option I see. If I manually visit the other uri paths such as /cgi-bin/cam.cgi it works fine, but these links are missing from the homepage. Do you have any ideas on what I can try?

[image]

Thank you!

gtxaspec, 2023-09-22:
@virmaior I noticed this too.  If you have time to take a look... thank you =D

virmaior, 2023-09-22:
using the wz_mini.conf,

 you need to add a line:

```
WEB_SERVER_OPTIONS="cam config car jpeg"
```

this line both populates `index.html` and should be something that's getting tested to see if the options are supposed to be usable or not.

...
if you've got that line and this still happens, then I've got to figure out why!

(I did just make some recent changes that are useful to me and possibly others by adding a test to see if the last minute copied over to the SD)

tdashmike, 2023-09-24:
> using the wz_mini.conf,
> 
> you need to add a line:
> 
> ```
> WEB_SERVER_OPTIONS="cam config car jpeg status"
> ```
> 
> this line both populates `index.html` and should be something that's getting tested to see if the options are supposed to be usable or not.
> 
> ... if you've got that line and this still happens, then I've got to figure out why!
> 
> (I did just make some recent changes that are useful to me and possibly others by adding a test to see if the last minute copied over to the SD)

I'm not sure what the status option does but it does show the links. I tried adding multicam but the it's not linking to the correct html file. I ended up just copying the links in index.html to index.top.html and fixed the "sever" typo. Thank you!

virmaior, 2023-09-24:
good find!

I've added "multicam.html" into the list of options it will auto-populate.
status.cgi
so far has `test=recording` = will figure out whether the camera is saving minutes to the SD card (I have some SD cards that intermittently seem to be failing and going read only).
was also hoping to add `test=irledon` and `test=nightmode` but I was not able to figure out the GPIO functionality.

once working I'm planning to add these into the multicam tool -- to add the ability to coordinate night mode / led states

tdashmike, 2023-09-25:
I'm using your multicam code btw, I'm not sure if it's in the latest release.

gtxaspec, 2023-09-25:
PR #624 applied to latest master, let us know or reopen if it still doesn't work.

closed 2023-09-25 by gtxaspec