gtxaspec/wz_mini_hacks #625: Upgrade fail: md5sum: WARNING: 1 of 351 computed checksums did NOT match

archandanime, 2023-09-26:
I just ran upgrade-run.sh from my Wyze cam V3 but it failed with error:
```
SD_ROOT/wz_mini/etc/network.d/S10httpd: FAILED
SD_ROOT/wz_mini/etc/network.d/s10httpd: OK
...
md5sum: WARNING: 1 of 351 computed checksums did NOT match
```
So I took a look at `file.chk`
```
c36d67c69a7e86be6202b2d6e95a63ee  SD_ROOT/wz_mini/etc/network.d/S10httpd
758921d3c465ea9ecd578639dd33f774  SD_ROOT/wz_mini/etc/network.d/s10httpd
```
They have different hashes because they are different files, yes. I believe this issue was caused after `upgrade-run.sh` was run, it extracted `MASTER.tar.gz` on an FAT32 partition, then `S10httpd` is overwritten by `s10httpd` because filenames are not case-sensitive on FAT32:
```
[root@WCV3-Indoor:network.d]# ls
S01wlanhw           S08hostname         S14storemac         S17go2rtc
S03ipv6             S09dnsmasq          S15v4l2rtspserver   s10httpd
```

archandanime, 2023-09-26:
I look at the commit history and the file was added by commit [f23672b](https://github.com/gtxaspec/wz_mini_hacks/commit/f23672b6391633df57697628ff727b29e4cc3f5c), did you create a new `s10httpd` instead of modifying `S10httpd`?

gtxaspec, 2023-09-26:
@archandanime fixed in latest master

archandanime, 2023-09-26:
I ran `upgrade-run.sh`:
```
File verification successful!
Move staging directory to perform upgrade
Installing latest upgrade-run from repo
Launching latest upgrade-script
Exit old script
[root@WCV3-Indoor:opt]# Resume upgrade-run, latest version
check for old directory path
enable wifi drivers if disabled in config

[root@WCV3-Indoor:opt]# Backup user config

[root@WCV3-Indoor:opt]# Rebooting into UPGRADE MODE
Connection to 192.168.10.20 closed by remote host.
Connection to 192.168.10.20 closed.
```
Then the device is not connected to Wi-Fi anymore, I don't know if it failed to boot. I waited 5 minutes and re-plugged the power brick but it still didn't connect to Wi-Fi, rebooting Wi-Fi router didn't help either.
I pulled put the SD card and read `upgrade_wz_mini.log`:
```
2_install/fw_tool.sh: OK
v2_install/v2_kernel.bin: OK
v2_install/.gitignore: OK
+ '[' 0 -eq 0 ]
+ echo 'File verification successful!'
File verification successful!
+ '[' -f /opt/wz_mini/tmp/.T20 ]
+ umount -l /opt/wz_mini/tmp
+ ls -l /opt/wz_mini/
total 131232
drwxrwxrwx    2 root     root         16384 Sep 26  2023 bin
drwxrwxrwx   12 root     root         16384 Jan  1  1980 etc
drwxrwxrwx    3 root     root         16384 Jul 20  2023 lib
drwxrwxrwx    2 root     root         16384 Jan  1 00:00 log
drwxrwxrwx    2 root     root         16384 Jul 20  2023 mnt
drwxrwxrwx    2 root     root         16384 Jul 20  2023 root
-rwxrwxrwx    1 root     root     134217728 Jan  1  1980 swap
drwxrwxrwx    2 root     root         16384 Jul 20  2023 tmp
drwxrwxrwx    4 root     root         16384 Jul 20  2023 usr
drwxrwxrwx    4 root     root         16384 Jul 20  2023 www
-rwxrwxrwx    1 root     root          2664 Sep 26  2023 wz_mini.conf
+ rm -rf /opt/wz_mini/bin /opt/wz_mini/etc /opt/wz_mini/lib /opt/wz_mini/log /opt/wz_mini/mnt /opt/wz_mini/root /opt/wz_mini/swap /opt/wz_mini/tmp /opt/wz_mini/usr /opt/wz_mini/www /opt/wz_mini/wz_mini.conf
rm: can't stat '/opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/kernel/drivers/usb/storage/\F3\E6\E1?	\EE?2.\FF': I/O error
rm: can't remove '/opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/kernel/drivers/usb/storage/?\BAs?^3\F4\B5.B?': Read-only file system
rm: can't remove '/opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/kernel/drivers/usb/storage/?)f?\F1\FC?.cF?': Read-only file system
rm: can't stat '/opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/kernel/drivers/usb/storage/w;????].x
e': I/O error
...
rm: can't remove '/opt/wz_mini/www/multicam.html': Read-only file system
rm: can't remove '/opt/wz_mini/wz_mini.conf': Read-only file system
+ sync
+ mv /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/bin /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/etc /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/lib /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/log /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/mnt /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/root /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/swap.gz /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/tmp /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/usr /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/www /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/wz_mini.conf /opt/wz_mini/
mv: can't rename '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/bin': Read-only file system
mv: can't rename '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/etc': Read-only file system
...
rm: can't remove '/opt/Upgrade/wz_mini_hacks/v2_install/fw_tool.sh': Read-only file system
rm: can't remove '/opt/Upgrade/wz_mini_hacks/v2_install/v2_kernel.bin': Read-only file system
+ sync
+ reboot
```
I think there's something wrong with my SD card, I formatted the SD card, downloaded lastest wz_mini.tar.gz, extracted, copied config files and powered the camera on. It worked.

---------------
On my second Wyze Cam V3, I also ran `upgrade-run.sh`, it has no issue at all. I'm leaving the issue from the first camera in case someone needs.

Thank you for the fix!

closed 2023-09-26 by archandanime

gtxaspec, 2023-09-26:
@virmaior FYI looks like you may need to fix the case on S10httpd vs s10httpd in your repo. i fixed it in master.

virmaior, 2023-09-26:
not sure how that propagated through ... but I've restored my repo from yours (also not really sure how an uppercase got in there since it was WyzeV3 -> RPi -> OSX -> Github .... )