gtxaspec/wz_mini_hacks #632: Wyze V3 go2rtc frame link fails

vfilby, 2023-10-06:
Using  go2rtc, the links to get a jpeg frame return a 500 error, and the following error in the go2rtc log.  Reading up on go2rtc it looks like it should just grab the first stream and transcode, but I am not knowledgeable enough about either go2rtc or ffmpeg to start tweaking values.

Te error message suggested increasing the  analyze and probe durations so I tried 500 instead of 50 and that didn't change the results.

Anyone else experiencing this problem? Any potential solutions or leads?

```
03:25:40.749 TRC [api] GET /api/frame.jpeg?src=1080p <device ip>:63033
03:25:40.756 DBG [exec] run url="exec:ffmpeg -fflags +genpts+flush_packets+nobuffer+bitexact -flags +low_delay -thread_queue_size 100 -analyzeduration 500 -probesize 500 -pix_fmt yuv420p -r 30 -i /dev/video1  -flags +global_header -movflags +empty_moov+default_base_moof+frag_keyframe -framerate 30 -video_size 1920x1080 -pix_fmt yuv420p -c:v copy -copyts -rtsp_transport tcp -f rtsp {output}"
ffmpeg version 4.4.4 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 11.4.0 (Buildroot 2023.02.2)
  configuration: --enable-cross-compile --cross-prefix=/root/buildroot/buildroot-2023.02.2/output/host/bin/mipsel-buildroot-linux-musl- --sysroot=/root/buildroot/buildroot-2023.02.2/output/host/mipsel-buildroot-linux-musl/sysroot --host-cc=/usr/bin/gcc --arch=mipsel --target-os=linux --disable-stripping --pkg-config=/root/buildroot/buildroot-2023.02.2/output/host/bin/pkg-config --enable-static --disable-shared --prefix=/usr --enable-avfilter --disable-version3 --enable-logging --enable-optimizations --disable-extra-warnings --enable-avdevice --enable-avcodec --enable-avformat --enable-network --disable-gray --enable-swscale-alpha --disable-small --enable-dct --enable-fft --enable-mdct --enable-rdft --disable-crystalhd --disable-dxva2 --enable-runtime-cpudetect --disable-hardcoded-tables --disable-mipsdsp --disable-mipsdspr2 --disable-msa --enable-hwaccels --disable-cuda --disable-cuvid --disable-nvenc --disable-avisynth --disable-frei0r --disable-libopencore-amrnb --disable-libopencore-amrwb --disable-libdc1394 --disable-libgsm --disable-libilbc --disable-libvo-amrwbenc --disable-symver --disable-doc --enable-gpl --enable-nonfree --enable-ffmpeg --disable-ffplay --disable-libv4l2 --enable-avresample --enable-ffprobe --disable-libxcb --enable-postproc --enable-swscale --disable-encoders --enable-encoder=flv --enable-encoder=aac --enable-encoder=pcm_s16be --enable-encoder=pcm_s16le --enable-encoder=pcm_alaw --enable-encoder=pcm_mulaw --enable-encoder=libmp3lame --enable-encoder=libfdk_aac --enable-encoder=png --enable-encoder=ljpeg --enable-encoder=mjpeg --enable-encoder=libx264 --enable-encoder=libopus --disable-decoders --enable-decoder=h264 --enable-decoder=pcm_u8 --enable-decoder=flv --enable-decoder=pcm_s16be --enable-decoder=pcm_s16le --enable-decoder=pcm_alaw --enable-decoder=pcm_mulaw --enable-decoder=mp3 --enable-decoder=png --enable-decoder=ljpeg --enable-decoder=mjpeg --enable-decoder=h264 --enable-decoder=libopus --disable-muxers --enable-muxer=mp4 --enable-muxer=h264 --enable-muxer=rtsp --enable-muxer=segment --enable-muxer=stream_segment --enable-muxer=flv --enable-muxer=image2 --enable-muxer=hls --enable-muxer=mp3 --enable-muxer=pcm_s16le --disable-demuxers --enable-demuxer=h264 --enable-demuxer=rtsp --enable-demuxer=image2 --enable-demuxer=hls --enable-demuxer=mp3 --enable-demuxer=pcm_s16le --disable-parsers --enable-parser=h264 --enable-parser=png --disable-bsfs --enable-bsf=h264_metadata --disable-protocols --enable-protocol=file --enable-protocol=tcp --enable-protocol=udp --enable-protocol=pipe --enable-protocol=rtmp --enable-protocol=rtmps --enable-protocol=hls --enable-protocol=https --disable-filters --enable-filter=null --enable-filter=anullsrc --enable-filter=aresample --enable-filter=scale --enable-filter=volume --enable-filter=adelay --enable-indevs --enable-alsa --enable-outdevs --enable-pthreads --enable-zlib --disable-bzlib --enable-libfdk-aac --disable-libcdio --disable-gnutls --disable-openssl --disable-libdrm --disable-libopenh264 --disable-vaapi --disable-vdpau --disable-mmal --disable-omx --disable-omx-rpi --disable-libopencv --enable-libopus --disable-libvpx --disable-libass --disable-libbluray --disable-libmfx --disable-librtmp --enable-libmp3lame --disable-libmodplug --disable-libspeex --disable-libtheora --disable-iconv --disable-libfreetype --disable-fontconfig --disable-libopenjpeg --enable-libx264 --disable-libx265 --disable-libdav1d --disable-x86asm --disable-mmx --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-avx --disable-avx2 --disable-armv6 --disable-armv6t2 --disable-vfp --disable-neon --disable-mipsfpu --disable-asm --disable-altivec --extra-libs=-latomic --disable-pic --cpu=generic --enable-indev=alsa --enable-outdev=alsa --enable-libmp3lame --enable-version3 --enable-mbedtls
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[h264 @ 0x772b7450] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x772b7450] decode_slice_header error
[h264 @ 0x772b7450] no frame!
[video4linux2,v4l2 @ 0x772b9820] decoding for stream 0 failed
[video4linux2,v4l2 @ 0x772b9820] Could not find codec parameters for stream 0 (Video: h264, none, 1920x1080): unspecified pixel format
Consider increasing the value for the 'analyzeduration' (500) and 'probesize' (500) options
Input #0, video4linux2,v4l2, from '/dev/video1':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: h264, none, 1920x1080, 30 fps, 30 tbr, 1000k tbn, 2000k tbc
```

Relevant wz config:

```
#####VIDEO STREAM#####
RTSP_LOGIN="admin"
RTSP_PASSWORD=""
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="false"
RTSP_HI_RES_ENABLE_AUDIO="false"

RTSP_LOW_RES_ENABLED="false"
RTSP_LOW_RES_ENABLE_AUDIO="false"

RTSP_AUTH_DISABLE="true"

GO2RTC_SERVER_ENABLED="true"
GO2RTC_HI_RES_ENABLED="true"
GO2RTC_HI_RES_ENABLE_AUDIO="true"
GO2RTC_LOW_RES_ENABLED="true"
GO2RTC_LOW_RES_ENABLE_AUDIO="true"
```

gtxaspec, 2023-10-06:
from what I can tell from the go2rtc docs, `/api/frame.jpeg?src=1080p ` is only supported on mjpeg sources.  you'll have to use `api/frame.mp4?src=1080p`

vfilby, 2023-10-08:
According to this comment on go2rtc, as of v1.5 it should automatically transcode to jpeg from the first stream, it doesn't have to be an mpeg source.

https://github.com/AlexxIT/go2rtc/issues/417#issuecomment-1537064737

> JPEG snapshots
>
> All streams now support JPEG snapshots. For MJPEG streams frame is taken directly. For H264/H265 cameras automatic transcoding with FFmpeg is used.

gtxaspec, 2023-10-09:
our ffmpeg is compiled without jpeg support, i'll take a look

gtxaspec, 2023-10-13:
ok this seems to be fixed in the latest go2rtc, i'll update it soon.

vfilby, 2023-10-13:
Great! I'll wait for the next release to test further.

gtxaspec, 2023-10-13:
latest master includes the new version of go2rtc.

gtxaspec, 2023-10-17:
@vfilby can you confirm this is fixed?

vfilby, 2023-10-22:
So, it doesn't seem to be working for me.  When I try to hit the jpeg frame link, here is the logs from go2rtc.  I have included the go2rtc logs and some version sanity checking below to make sure I am not doing something completely wrong.

```
21:13:24.210 TRC [api] GET /api/frame.jpeg?src=1080p 10.0.1.97:57317
21:13:24.211 DBG [exec] run url="exec:ffmpeg -fflags +genpts+flush_packets+nobuffer+bitexact -flags +low_delay -thread_queue_size 100 -analyzeduration 50 -probesize 50 -pix_fmt yuv420p -r 30 -i /dev/video1  -flags +global_header -movflags +empty_moov+default_base_moof+frag_keyframe -framerate 30 -video_size 1920x1080 -pix_fmt yuv420p -c:v copy -copyts -rtsp_transport tcp -f rtsp {output}"
ffmpeg version 4.4.4 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 11.4.0 (Buildroot 2023.02.2)
  configuration: --enable-cross-compile --cross-prefix=/root/buildroot/buildroot-2023.02.2/output/host/bin/mipsel-buildroot-linux-musl- --sysroot=/root/buildroot/buildroot-2023.02.2/output/host/mipsel-buildroot-linux-musl/sysroot --host-cc=/usr/bin/gcc --arch=mipsel --target-os=linux --disable-stripping --pkg-config=/root/buildroot/buildroot-2023.02.2/output/host/bin/pkg-config --enable-static --disable-shared --prefix=/usr --enable-avfilter --disable-version3 --enable-logging --enable-optimizations --disable-extra-warnings --enable-avdevice --enable-avcodec --enable-avformat --enable-network --disable-gray --enable-swscale-alpha --disable-small --enable-dct --enable-fft --enable-mdct --enable-rdft --disable-crystalhd --disable-dxva2 --enable-runtime-cpudetect --disable-hardcoded-tables --disable-mipsdsp --disable-mipsdspr2 --disable-msa --enable-hwaccels --disable-cuda --disable-cuvid --disable-nvenc --disable-avisynth --disable-frei0r --disable-libopencore-amrnb --disable-libopencore-amrwb --disable-libdc1394 --disable-libgsm --disable-libilbc --disable-libvo-amrwbenc --disable-symver --disable-doc --enable-gpl --enable-nonfree --enable-ffmpeg --disable-ffplay --disable-libv4l2 --enable-avresample --enable-ffprobe --disable-libxcb --enable-postproc --enable-swscale --disable-encoders --enable-encoder=flv --enable-encoder=aac --enable-encoder=pcm_s16be --enable-encoder=pcm_s16le --enable-encoder=pcm_alaw --enable-encoder=pcm_mulaw --enable-encoder=libmp3lame --enable-encoder=libfdk_aac --enable-encoder=png --enable-encoder=ljpeg --enable-encoder=mjpeg --enable-encoder=libx264 --enable-encoder=libopus --disable-decoders --enable-decoder=h264 --enable-decoder=pcm_u8 --enable-decoder=flv --enable-decoder=pcm_s16be --enable-decoder=pcm_s16le --enable-decoder=pcm_alaw --enable-decoder=pcm_mulaw --enable-decoder=mp3 --enable-decoder=png --enable-decoder=ljpeg --enable-decoder=mjpeg --enable-decoder=h264 --enable-decoder=libopus --disable-muxers --enable-muxer=mp4 --enable-muxer=h264 --enable-muxer=rtsp --enable-muxer=segment --enable-muxer=stream_segment --enable-muxer=flv --enable-muxer=image2 --enable-muxer=hls --enable-muxer=mp3 --enable-muxer=pcm_s16le --disable-demuxers --enable-demuxer=h264 --enable-demuxer=rtsp --enable-demuxer=image2 --enable-demuxer=hls --enable-demuxer=mp3 --enable-demuxer=pcm_s16le --disable-parsers --enable-parser=h264 --enable-parser=png --disable-bsfs --enable-bsf=h264_metadata --disable-protocols --enable-protocol=file --enable-protocol=tcp --enable-protocol=udp --enable-protocol=pipe --enable-protocol=rtmp --enable-protocol=rtmps --enable-protocol=hls --enable-protocol=https --disable-filters --enable-filter=null --enable-filter=anullsrc --enable-filter=aresample --enable-filter=scale --enable-filter=volume --enable-filter=adelay --enable-indevs --enable-alsa --enable-outdevs --enable-pthreads --enable-zlib --disable-bzlib --enable-libfdk-aac --disable-libcdio --disable-gnutls --disable-openssl --disable-libdrm --disable-libopenh264 --disable-vaapi --disable-vdpau --disable-mmal --disable-omx --disable-omx-rpi --disable-libopencv --enable-libopus --disable-libvpx --disable-libass --disable-libbluray --disable-libmfx --disable-librtmp --enable-libmp3lame --disable-libmodplug --disable-libspeex --disable-libtheora --disable-iconv --disable-libfreetype --disable-fontconfig --disable-libopenjpeg --enable-libx264 --disable-libx265 --disable-libdav1d --disable-x86asm --disable-mmx --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-avx --disable-avx2 --disable-armv6 --disable-armv6t2 --disable-vfp --disable-neon --disable-mipsfpu --disable-asm --disable-altivec --extra-libs=-latomic --disable-pic --cpu=generic --enable-indev=alsa --enable-outdev=alsa --enable-libmp3lame --enable-version3 --enable-mbedtls
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[h264 @ 0x779f3450] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 0x779f3450] decode_slice_header error
[h264 @ 0x779f3450] no frame!
[video4linux2,v4l2 @ 0x779f5820] decoding for stream 0 failed
[video4linux2,v4l2 @ 0x779f5820] Could not find codec parameters for stream 0 (Video: h264, none, 1920x1080): unspecified pixel format
Consider increasing the value for the 'analyzeduration' (50) and 'probesize' (50) options
[... cut ...]
  - exec:ffmpeg -hide_banner -f alsa -ac 1 -i dsnooper2 -c:a libfdk_aac -afterburner 1 -channels 1 -b:a 128k -profile:a aac_he -ar 16000 -strict experimental -muxdelay 0 -muxpreload 0 -ac:a 1 -f rtsp {output}
# libopus has higher cpu requirements, audio may studder.
#  - exec:ffmpeg -hide_banner -f alsa -ac 1 -i dsnooper2 -c:a libopus -muxdelay 0 -muxpreload 0 -ac:a 1 -af adelay=0|0 -f rtsp {output}
```

Couple checks to make sure I updated the software correctly:

```
[root@WCV3:bin]# cat app.ver
Sun Oct 15 12:16:27 AM PDT 2023
[root@WCV3:bin]# cd ../../bin
[root@WCV3:bin]# ./go2rtc_linux_mipsel -version
Current version:  1.8.0
```

gtxaspec, 2024-01-02:
go2rtc updated to 1.8.5, is this still happening?

vfilby, 2024-01-02:
I can't check I had to reuse this camera elsewhere offsite and I won't have access to it for a few months.

Dan-tha-man, 2024-03-09:
I am running go2rtc 1.8.5 and the frame.jpeg does not work for me

ziggycatuk, 2024-03-12:
Not working with 1.8.5 here either when updating frigate (docker)

However if I run a docker container just for go2rtc,I can access the still jpeg image.