gtxaspec/wz_mini_hacks #634: Feature request: Floodlight output (call webhook, script, or mqtt)

loopy321, 2023-10-11:
Thanks for the amazing work here.

I have a feature request for output when the PIR sensor is triggered.   I see in the logs the following when the PIR sensor is triggered, but was hoping to be able to call a url, script, or mqtt:
```
iCamera: [flood_light.c:2092][trigger_mode_night_auto] motion & nightMode ok! turn ON light!
iCamera: [sdk,0735](local_sdk_close_night_light) stateFlag:2, LastState:1
iCamera: [flood_light.c:1302][flood_light_open]bright set [dst:255] [old:0]
iCamera: [SDK-ALARM]dbg: algo[0] state goto: THREAD_STATE_SETTING
iCamera: [SDK-ALARM]dbg: algo[0] state goto: THREAD_STATE_IDEL get_algo_result_handle
iCamera: [handle_video.c:118]dbg: ##### localsdk alarm switch off, byid:2 #####
iCamera: [flood_light.c:848](floodlight_dynamic_disable_alarm) disable...
iCamera: [flood_light.c:791](floodlight_set_light_on_isp_mode) ok...
iCamera: @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
iCamera: @@@@@@@@@@@@@ FLOODLIGHT TURN ON @@@@@@@@@@@@@
iCamera: @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
iCamera: [protocol.c:2616]-------->>  to[255], with[700 ms]
iCamera: [nightvision.c:126]in: state:[2]
iCamera: [handle_audio.c:104]dbg: ##### ada audio switch off, byid:1 #####
iCamera: [nightvision.c:167]out: state:[2]
iCamera: [nightvision.c:126]in: state:[1]
iCamera: [nightvision.c:167]out: state:[1]

```
Any tips on how to use the flood_light.c call?

virmaior, 2023-10-18:
directly no, but the flood_light.c call appears to occur in response to the device sensor switching to night mode.

`cat /proc/jz/isp/isp-m0` provides sensor status info 

I'm using it to detect night mode at `https://github.com/gtxaspec/wz_mini_hacks/blob/master/SD_ROOT/wz_mini/www/cgi-bin/status.cgi`

that seems to be the same thing you're looking for here.

loopy321, 2023-10-18:
> `cat /proc/jz/isp/isp-m0` provides sensor status info
> 
> I'm using it to detect night mode at `https://github.com/gtxaspec/wz_mini_hacks/blob/master/SD_ROOT/wz_mini/www/cgi-bin/status.cgi`
> 
> that seems to be the same thing you're looking for here.

Thank you!  Now I can poll that script.  Ideally it would push the change - I guess I could use a logger script to grep the change.

loopy321, 2023-10-19:
I have the following script working to call a webhook on my homeassistant host and added an rc.d startup entry to `/opt/wz_mini/etc/rc.local.d` in case it is helpful to others or you have suggestion for improvement:
```
#! /opt/wz_mini/bin/bash

logread -f | while read;do echo "$REPLY" | grep -q " FLOODLIGHT TURN" &&
if [[ $REPLY = *ON* ]]
then
  curl -k --header "Content-Type: application/json"   --request POST \
    --data '{"floodlight": true}'   http://homeassistant:8123/api/webhook/floodlight
else
  curl -k --header "Content-Type: application/json"   --request POST \
    --data '{"floodlight": false}'   http://homeassistant:8123/api/webhook/floodlight
fi
done
```