gtxaspec/wz_mini_hacks #636: Updated VLC crashing on Wyze Stream

skilau, 2023-10-13:
Updated VLC today, (they pushed a new version: 3.0.19 Vetinari), and opening any Wyze stream causes VLC to crash.

Older VLCs have always worked fine.

It is unclear why VLC crashes, but since VLC is the gold standard, we should try to figure out why it is dying on our stream.

gtxaspec, 2023-10-13:
@skilau is this using v4l2rtspserver or go2rtc

gtxaspec, 2023-10-13:
vlc 3.0.19 works for me using go2rtc with the link "rtsp://<ip>:8554/1080p?mp4" as an example

skilau, 2023-10-13:
Ah, sorry.  I should have done a better job reporting this.

I am on a bit of older firmware for the Mini hacks, (it has been so rock-solid, I didn't want to upgrade from it)
So I am sure it would be on the old v4l2rtspserver.

I hadn't intended on upgrading, since the old v4l2rtspserver firmware has been rock-solid, but I think this probably pushes me forward to upgrading.

One question.
I have sat on the older firmware for the Wyze Cam Floodlight, since you had to build the special driver to bring into the kernel for controlling the Spotlight.
Issue:
https://github.com/gtxaspec/wz_mini_hacks/issues/172

Does the newer firmware include this driver/kernel with this fix?

Thanks for your continued support!

gtxaspec, 2023-10-13:
no, it doesn't yet.  Waiting for newer kernel version hopefully soon, though you should be able to use the same kernel as before.  Make a backup of wz_mini just in case =D