gtxaspec/wz_mini_hacks #637: Upgrade script doesn't save go2rtc.yml

clee604, 2023-10-14:
I just upgraded a wyze V2 camera and the upgrade script doesn't seem to retain the previous go2rtc.yml and seems to overwrite with default file. I noticed because the V2 requires updating the video source in the go2rtc configuration.

I believe this also happens on the V3 since previously I've had to reapply my api and rtsp authentication settings but I haven't tried upgrading just yet to confirm this.

gtxaspec, 2023-10-14:
good find, will fix.

clee604, 2023-10-14:
Confirmed this also happens on V3

gtxaspec, 2023-10-15:
fixed in latest master

clee604, 2023-10-15:
Just tried an upgrade and the configuration wasn't saved. I believe in your code you have go2rtc.yaml when the file is named go2rtc.yml

gtxaspec, 2023-10-15:
good catch my dev system is named differently.  updated master.

gtxaspec, 2023-10-17:
@clee604 can you confirm it's fixed? =D

archandanime, 2023-10-17:
It worked on my v3 [upgrade_wz_mini.log](https://github.com/gtxaspec/wz_mini_hacks/files/12928405/upgrade_wz_mini.log). go2rtc.yml file is preserved after upgrade, I'm not sure if it also works on v2 cameras.
```
...
+ cp /opt/Upgrade/preserve/cron/root /opt/wz_mini/etc/cron/
+ cp /opt/Upgrade/preserve/resolv.dnsmasq /opt/wz_mini/etc/resolv.dnsmasq
+ cp /opt/Upgrade/preserve/go2rtc.yml /opt/wz_mini/etc/go2rtc.yml
+ rm -rf /opt/Upgrade
+ sync
+ reboot
```

gtxaspec, 2023-10-17:
The script should be the same on the v2... Thanks 🙂

closed 2023-10-17 by gtxaspec

clee604, 2023-10-17:
I upgraded 5 cameras and 4 worked fine retaining the go2rtc.yml. The last errored out but I think it's a separate issue unrelated to this issue