gtxaspec/wz_mini_hacks #640: Missing a pre-requisite, is this a valid alternative?

andlewis, 2023-10-19:
After reviewing the prerequisites for installation, I noticed I'm missing a cat. I do have a dog though. The interface is basically the same (food in, poop out). Can you confirm compatibility, and if it's not a valid configuration, can we get that implemented?

archandanime, 2023-10-20:
I think that prerequisite is optional and a dog works too(I haven't tested to confirm).

On the [Prerequisites](https://github.com/gtxaspec/wz_mini_hacks#prerequisites) it says a cat is for emotional support during setup. I think gtxaspec does own(or has not tested with) a dog to confirm if a dog can also provide emotional support.

It can be a good idea to make a "List of working animals" on [Discussion](https://github.com/gtxaspec/wz_mini_hacks/discussions) for people to add their favorite pets that work with the setup ;) However the prerequisite is optional, so we don't need that.

gtxaspec, 2023-10-20:
no! Cats only.  and now you're banned from the internet! =D

endertable, 2023-10-22:
Guys, there is a way around not have a cat,  I have a stuffed kitty toy on my mantle.  It's a kludge workaround, but I have had minimal issues with it, especially with FW 4.36.9.139 and under.