gtxaspec/wz_mini_hacks #642: Wyze App can't connect when using latest wz_mini_hacks

cad1231, 2023-10-22:
I am having a new problem that just started today -- after I updated to the latest version of this mod. I have a WyzeCam v3. I am using this mod with go2rtc to stream to homeassistant, and also have others who access the camera with the Wyze app. Today, the Wyze app stopped connecting to the camera. RTSP continues to stream via go2rtc and I can access the camera via webUI, SSH, WebRTC, RTSP, etc. But nothing from the Wyze app. 

If I remove the SD card and reboot, the wyze app works perfectly. Any ideas what I should be checking for?

archandanime, 2023-10-22:
Could you run `gather_wz_logs.sh` and share the log file?

cad1231, 2023-10-22:
It magically started working again today. Really weird. I'll post back if it acts up again.

closed 2023-10-22 by cad1231