gtxaspec/wz_mini_hacks #656: Does this project work on Wyze OG and Wyze Doorbell v2?

zaneschepke, 2023-10-28:


endertable, 2023-10-29:
Wyze OG - no it doesn't.  These use completely different OSs, not Linux.  I believe they are more of Arduino type systems, OS on a chip.

Wyze Doorbell v2 - just came out so the jury is still out, but I would be very surprised if not the same thing as Wyze has managed to keep hacks out of their latest hardware releases by using the new infrastructures.

I should be getting mine tomorrow, so maybe can get more info..

gtxaspec, 2023-10-29:
confirmed the DB2 is a T31 chip, so maybe.

endertable, 2023-10-29:
The update tar.  Linux, but it looks more like it is following the security of Cam V3 Pro.  An all-in-one file system...

[4.51.0.6875.zip](https://github.com/gtxaspec/wz_mini_hacks/files/13199442/4.51.0.6875.zip)

gtxaspec, 2023-12-29:
the doorbell v2, like the pan v3, has signature verification enabled on the SOC.  no dice. :(

closed 2023-12-29 by gtxaspec