gtxaspec/wz_mini_hacks #662: V3 blue light but no stream to RTSP or go2rtc

madjetey, 2023-11-05:
V3 camera running 4.36.9.139 isn't working anymore after sitting unused for 92 days.  Prior to this, wz_mini ran fine as did [docker wyze bridge](https://github.com/mrlt8/docker-wyze-bridge).  After resuming use after 92 days, only the wyze app streams normally.  RTSP or go2rtc streams don't work at all.

Config below:
```
#wz_mini configuration

#### W E B CAMERA###
##THIS MODE DISABLES EVERYTHING AND IT WILL
## WORK AS A WEB CAMERA FOR YOUR PC ***ONLY***
WEB_CAM_ENABLE="false"
WEB_CAM_BIT_RATE="8000"
WEB_CAM_FPS_RATE="25"

#####NETWORKING#####
CUSTOM_HOSTNAME="WCV3"

ENABLE_USB_ETH="false"
ENABLE_USB_ETH_MODULE_AUTODETECT="true"
ENABLE_USB_ETH_MODULE_MANUAL=""
USB_ETH_MAC_ADDR=""

ENABLE_USB_DIRECT="false"
USB_DIRECT_MAC_ADDR="02:01:02:03:04:08"

ENABLE_USB_RNDIS="false"

ENABLE_IPV6="false"

ENABLE_WIREGUARD="false"
WIREGUARD_IPV4=""
WIREGUARD_PEER_ENDPOINT=""
WIREGUARD_PEER_PUBLIC_KEY=""
WIREGUARD_PEER_ALLOWED_IPS=""
WIREGUARD_PEER_KEEP_ALIVE=""

ENABLE_IPTABLES="false"

ENABLE_NFSv4="false"

ENABLE_RTL8189FS_DRIVER="true"
ENABLE_ATBM603X_DRIVER="true"

NTP_SERVER=""

##### NETWORK INTERFACE BONDING #####
BONDING_ENABLED="false"
BONDING_PRIMARY_INTERFACE="eth0"
BONDING_SECONDARY_INTERFACE="wlan0"
BONDING_LINK_MONITORING_FREQ_MS="100"
BONDING_DOWN_DELAY_MS="5000"
BONDING_UP_DELAY_MS="5000"
BONDING_FAIL_OVER_MAC="0"

#####ACCESSORIES#####
REMOTE_SPOTLIGHT="false"

####MOTOR OPTIONS####
MOTOR_DISABLE="false"
MOTOR_DEFAULT_X=""
MOTOR_DEFAULT_Y=""

#####VIDEO STREAM#####
RTSP_LOGIN="login"
RTSP_PASSWORD="password"
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="false"
RTSP_HI_RES_ENABLE_AUDIO="false"

RTSP_LOW_RES_ENABLED="false"
RTSP_LOW_RES_ENABLE_AUDIO="false"

RTSP_AUTH_DISABLE="false"

VIDEO_0_MAX_BITRATE="2048"
VIDEO_0_TARGET_BITRATE="1024"
VIDEO_0_ENC_PARAMETER="2"
VIDEO_0_FPS=""

VIDEO_1_MAX_BITRATE=""
VIDEO_1_TARGET_BITRATE=""
VIDEO_1_ENC_PARAMETER=""
VIDEO_1_FPS=""

GO2RTC_SERVER_ENABLED="true"
GO2RTC_HI_RES_ENABLED="true"
GO2RTC_HI_RES_ENABLE_AUDIO="true"
GO2RTC_LOW_RES_ENABLED="true"
GO2RTC_LOW_RES_ENABLE_AUDIO="true"

#####RTMP STREAMING#####
#### REQUIRES RTSP ####
### AND RTSP AUDIO ###
RTMP_STREAM_ENABLED="false"
RTMP_STREAM_FEED="video1_unicast"
RTMP_STREAM_SERVICE="youtube"
RTMP_STREAM_DISABLE_AUDIO="false"
RTMP_STREAM_YOUTUBE_KEY=""
RTMP_STREAM_TWITCH_KEY=""
RTMP_STREAM_FACEBOOK_KEY=""

#####GENERAL#####
LIBCALLBACK_ENABLE="true"
ENABLE_SWAP="true"
ENABLE_USB_STORAGE="false"
ENABLE_EXT4="false"
ENABLE_CIFS="false"
DISABLE_FW_UPGRADE="true"
AUDIO_PROMPT_VOLUME="50"
ENABLE_MP4_WRITE="true"
NIGHT_DROP_DISABLE="true"
ENABLE_FSCK_ON_BOOT="false"
ENABLE_CAR_DRIVER="false"
ENABLE_LOCAL_DNS="false"
ENABLE_CRONTAB="false"
ENABLE_SELFHOSTED_MODE="false"

#####DEBUG#####
#drops you to a shell via serial, doesn't load app_init.sh
DEBUG_ENABLED="false"
DEBUG_INITRAMFS_ENABLED="false"
DEBUG_PASSWORD="false"
ENABLE_SYSLOG_SAVE="false"

####WEB####
WEB_SERVER_ENABLED="true"
WEB_SERVER_AUTH="true"
WEB_SERVER_LOGIN="login"
WEB_SERVER_PASSWORD="password"

#####SCRIPTING#####
CUSTOM_SCRIPT_PATH=""

```

Log below:
```
There is no log 😵‍💫 
```

bjs-pdx, 2023-11-05:
Check your camera firmware via the Wyze App.  They pushed a forced update in the last couple of days.

madjetey, 2023-11-05:
It hadn't changed before posting but to be safe, I reinstalled 4.36.1.139 a few hours ago & it's still in effect right now.  App has a new firmware alert for 4.36.10.4054 which I'm ignoring

bjs-pdx, 2023-11-05:
By any chance did the camera IP change?

madjetey, 2023-11-05:
Nah. It's reserved in dhcp & I confirmed it's the same IP saved there as is in the app

bjs-pdx, 2023-11-06:
Hate to suggest it, but reinstall?

madjetey, 2023-11-06:
Didn't just reformat and reinstall wz_mini with a new config but even reflashed the firmware.

Only thing I haven't tried is upgrading the firmware to 4.36.10.4054 as wyze want then downgrading back to 4.36.1.139 before reinstalling again

gtxaspec, 2023-11-06:
@madjetey just wanted to clear up a few points:

1. When you noticed that wz_mini wasn't working, do you know the current firmware which was installed on the device?
2. Can you confirm you downgraded to x.139 with a fresh install of wz_mini on the sd card?
3. Can you check the partitioning of the SD card, make sure it's MBR, not GPT.

madjetey, 2023-11-07:
1. It was still 4.36.1.139.  I hadn't had a forced update applied.
2. I didn't downgrade so much as I reflashed the same 4.36.1.139 firmware. I also downloaded and install the latest version of wz_mini on the sd card
3. Also confirmed that the sd card partition is healthy FAT32 MBR

virmaior, 2023-11-07:
can you ssh into the camera or not?
(i.e. is just the streams down or is the entire wz_mini not running)

madjetey, 2023-11-07:
No I can't. Portscan also shows no ports open

madjetey, 2023-11-07:
Yup. All commands work through the app locally & over the internet. Even restart commands in wyze docker bridge work (no streaming there either though)

UnexDev, 2023-11-12:
Having same issue after (maybe) forced update. Just came back from a very long trip and the camera is not being picked up by my Homebridge plugin.

virmaior, 2023-11-12:
@UnexDev  you mean *after* a downgrade or just as a result of wyze pushing an update?

@madjetey  if wyze itself wasn't working, I would say:  after a certain version Wyze started using encoded versions of the ssid / pw and that if you downgrade before that it creates an issue where it doesn't successfully connect to any wifi and icamera depends on that moment happening to work at all. / it's not exactly a fun debugging method but if you modify the code for init.d section, you can get some logs etc. running before icamera starts. 

but if wyze is working, then I don't know what's going on. One sinister possibility is that the upgrade modified the bootloader and made it incompatible with the wz_mini_hack's system for loading.

madjetey, 2023-11-12:
@virmaior to be clear about the debugging process; am I removing the # from start of wz_init.sh line 3?  I should note that it latches onto the wifi just fine.  Watching myself through the app as I type this

I wonder if wyze can force update the bootloader without the firmware because I don't recall the camera updating beyond 4.36.9.139 aside early in Nov 2022 when I first booted then successfully downgraded and installed wz_mini_hack.

endertable, 2023-11-12:
With everything you are describing, I would try the fat32 SD card reformat again.  It really sounds like the camera is not loading the wz_mini kernel from the SD card..  The only reasons would be the formatting of the card, WYZE updated the BL, or something is corrupt on your wz-mini installation...  Beleive it or not, not all fat32 formatters are created equal.  Never use Windows for this.  Search and use a utility named 'guiformat-x64.exe' or 'guiformat.exe'  I have successfully used it for SD cards as big as 128G to work with wz_mini.

virmaior, 2023-11-13:
@madjetey that wouldn't be my first step.
I'd first delete `/opt/wz_mini/log/wz_init` from the SD Card.
Then put it back in and see if any file `/opt/wz_mini/log/wz_init` exists
if so, we know that the wz_mini starts up its process; if not, we know that it doesn't.
If it does start its process, then the question is how far does it get through the different processes and slogging through by using `touch` to produce files at different possible end points would be my strategy.

madjetey, 2023-11-13:
Woah.  It doesn't exist.  Matter of fact aside the camera generated folders (record & time_lapse) and the wz_mini.conf I modified for my setup, the only files newer than 25th Oct 2023 (the last time I downloaded and installed wz_mini) are the contents of .ota (all from yesterday 12th Nov 2023).

bjs-pdx, 2023-11-13:
Did you try a full factory reset of the camera?  I seem to recall that being required if you upgrade the firmware and want to go back down to the supported version.

madjetey, 2023-11-13:
Tried but no change. Also tried reflashing 4.36.9.139 but now it doesn't seem to work. Looking for 4.36.9.131 to downgrade and return. See if that does something. I'm not looking forward to going up the firmware ladder and downgrading incase there are surprises in the bootloader.

Anyone got the bin files?

gtxaspec, 2023-11-14:
can you take a screenshot of the file listing of the root from sdcard?

madjetey, 2023-11-14:
[image]

And the contents of the tutklog that hadn't appeared until now
```
{
  "IOTC Version":"3.4.4.3-0-g85fdb4a_openssl_Mips_T31_4.7.2_uclibc",
  "System Time":"2023-11-14 00:02:32",
  "TCP Relay Status":"OFF",
  "My Nat Type":3,
  "Login Protocl":"UDP",
  "LANConnectionTimeout(ms)":1500,
  "P2PConnectionTimeout(ms)":1500,
  "SessionAliveTimeout(ms)":30000,
  "Master Resolve Status":"RESOLVE_SUCCESS",
  "P2PServerList Exist":true,
  "VPG":[
    49193,
    62209,
    61954
  ],
  "MSG_HELLO1_R Received":true,
  "MSG_DEVICE_REGISTER_R Received":true,
  "IOTC_Device_Login() Return":"0",
  "Port Change Count":0,
  "Current Session Count":0,
  "connectionHandler Alive":true,
  "tutk_SockTaskMng Alive":false,
  "Server IP":"23.81.208.134:10001",
  "Recv Count of MSG_NOTIFY_DEVICE_REGISTER":1,
  "Last RTT(ms) of Alive Msg":249,
  "Send Count of MSG_DEVICE_REGISTER_ALIVE":28,
  "Recv Count of MSG_DEVICE_REGISTER_ALIVE_R":27,
  "Lost Rate of Alive Msg":3,
  "Last Return Value of Send MSG_DEVICE_REGISTER_ALIVE":48,
  "Last MSG_DEVICE_REGISTER_ALIVE Send Time":"00:02:11.123",
  "Last MSG_DEVICE_REGISTER_ALIVE_R Recv Time":"00:02:11.372"
}
```

JzJad, 2023-11-25:
Having just ran through all this, I had flashed the firmware from Lukejb14/Wyze-RTSP, then loaded the SD card with wz_mini_hacks, then did a factory reset/hold setup while its powered on for 10-15 seconds, get a voice prompt saying something about swap then when it boots its good to go. However any config change I make I seem to have to do a factory reset again for it to take affect.

SolAZDev, 2024-04-13:
@JzJad  sorry for the tag, but would this mean that a factory reset, and going through the normal procedure should fix the issue? WebCam has a similar issue where its a blue light but nothing's shown, so I assume the two might be related somehow