gtxaspec/wz_mini_hacks #663: what is dropbear_ed25519_host_key in the ssh folder?

johnvpr, 2023-11-07:
Found this file in my ssh folder:
[image]

[image]

what is this file?

Thanks

gtxaspec, 2023-11-07:
It's the host key generated by the ssh server.  It's used to validate communication with clients.

https://phoenixnap.com/kb/how-does-ssh-work

kronkinatorix, 2023-12-21:
I was able to ssh into the camera just fine before this file showed up, but now I'm getting permission denied errors. I can't seem to find how to generate a public key using dropbear? Advice would be appreciated.