gtxaspec/wz_mini_hacks #668: upgrade-run.sh gets stuck at Rebooting to UPGRADE MODE

Scags104, 2023-11-15:
The upgrade script has now taken down 2 of my cameras and put them in some odd boot loop im not sure know how to fix

`SD_ROOT/wz_mini/lib/libcallback_t20.so: OK
SD_ROOT/wz_mini/lib/libasound.so.2: OK
SD_ROOT/wz_mini/lib/libaudioProcess.so: OK
SD_ROOT/wz_mini/wz_mini.conf: OK
SD_ROOT/wz_mini/tmp/.gitignore: OK
v2_install/compile_image.sh: OK
v2_install/fw_tool.sh: OK
v2_install/v2_kernel.bin: OK
v2_install/.gitignore: OK
File verification successful!
Move staging directory to perform upgrade
Installing latest upgrade-run from repo
Launching latest upgrade-script
Exit old script
Resume upgrade-run, latest version
check for old directory path
[root@Wyze_Driveway_Camera:etc]# enable wifi drivers if disabled in config
Backup user config
Rebooting into UPGRADE MODE`

...stays here for a few minutes and then:
`client_loop: send disconnect: Connection reset`

They havent reconnected back to my network since this happened after a few power cycles either. Do i need to reimage the whole SD card?

archandanime, 2023-11-15:
I believe so, I countered this issue multiple times and it seems that formatting SD card is the only reliable solution.

KillahB33, 2023-12-19:
My cameras came back up but even after setting up the go2rtc it was giving me black screen on the stream so will have to go with a fresh install.

aejeffor, 2024-01-23:
Same behavior here. Upgrade a handful of camera from oct ‘22 release and they error out because go2rtc.yaml doesn’t exist in that older version. I tested in a newer aminj version that did have that particular file (July ‘23).   Similar behavior, although there is no error. It simply states “rebooting into upgrade mode” and then bricks the camera.