gtxaspec/wz_mini_hacks #671: MSE stream is getting squashed

day1118, 2023-11-20:
I have deployed `4.36.9.139` on Wyze Cam v3. My computer is running Ubuntu 22.04 (multiple browsers tested)

When I look at the WebRTC stream from `go2rtc`, it looks perfect.
When I look at the MSE stream from `go2rtc`, it looks squashed in on the sides.
(url: http://camera:1984/stream.html?src=1080p&mode=mse)

When I view the same url from my phone or Mac, the MSE stream looks correct.

Screenshot is of WebRTC & MSE link side by side (notice the additional black border on the mse image the door is very narrow). How can I troubleshoot if this is a server or client side issue? Is there a setting somewhere I've missed to set the correct resolution or aspect ratio?

[image]

Thanks

gtxaspec, 2023-11-20:
I believe It's a browser bug, or a go2rtc bug in the mse web code.  Have you tried a different browser?

day1118, 2023-11-20:
> I believe It's a browser bug, or a go2rtc bug in the mse web code. Have you tried a different browser?

I've tried chrome, brave & firefox

gtxaspec, 2023-11-20:
Same issue here, but it doesn't happen on windows or mobile... Hmm

day1118, 2023-11-21:
I should add that when I view MSE streams from an Armcrest camera, or Wyze camera via [Wyze Bridge](https://github.com/mrlt8/docker-wyze-bridge), the aspect ratio looks correct which is why I have assumed the issue is server side.

Any suggestions how we investigate and pinpoint this?

computergreek, 2024-12-10:
i'm having this exact issue too. this appears to be a bug with [wyze_mini_hacks](https://github.com/blakeblackshear/frigate/discussions/14459#discussioncomment-10999700).