gtxaspec/wz_mini_hacks #672: Flashing Blue LED

cdn4lf, 2023-11-20:
I'm running this in self-hosted mode and am unable to disable the flashing blue LED.

Is there a way to do this?

virmaior, 2023-11-21:
if memory serves this can be done by setting a GPIO.

See https://git.i386.io/wyze/v3-unlocker/-/commit/0a260c4ca34433b4c0263404e964cec3a6d8d196

```
# Blue led on
# Red led off
echo 38 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio38/direction
echo 1 > /sys/class/gpio/gpio38/value
# Blue led off
echo 39 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio39/direction
echo 0 > /sys/class/gpio/gpio39/value
echo 1 > /sys/class/gpio/gpio39/value

```

cdn4lf, 2023-11-21:
I got around this by removing the SD card, disabling the light in the Wyze app then reinserting the SD card.  No more light.

closed 2023-11-21 by cdn4lf