gtxaspec/wz_mini_hacks #678: 2ANJHWYZEC2 model its compatible with hack or not?

Toei79, 2023-12-01:
v2 version model 2ANJHWYZEC2, just google and other hack works on it,  check compatability list here , i dont see the model

i tried, but stuck on blue yellow light.
i feel prefer this h instead other one
thanks

Semag, 2023-12-06:
@Toei79  - I did not figure it out. I'm wondering if there is a different kernel in the C2 version? The compile script is just overwriting the existing "kernel.bin" with the "v2_kernel" that comes with WZ Mini - and it seems like that sends the camera into a bootloop when it tries to flash from the SD Card. However, remove the SD Card and the camera works fine after that.

Semag, 2023-12-06:
@gtxaspec  i can confirm that the boards are really close to the github i linked above: ingenic T20 and wyzec2 senp02 v1.1, isc5s-B02. (see pics)
[image]
[image]

I don't think i have the cable to get the UART logs though

gtxaspec, 2023-12-06:
confirm, do the markings on the chip end in BX02?

Semag, 2023-12-06:
Negative, it's BX01.

gtxaspec, 2023-12-06:
hmm...ok, its different from the S1, the S1 uses the L SOC which is 64mb of dram, while the X is 128M, so without the uart serial logs, can't tell whats going on

Semag, 2023-12-06:
@gtxaspec *sigh* I'll see if I can find my serial USB somewhere around here, I don't think i've used it in like 14 years, haha.

Answer-1, 2023-12-17:
Having the same issue with 2 more recent V2's, the older ones work like a charm.
Seems like even Wyze had this issue and releases an updated firmware to address this.
Found interesting infos there: https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/issues/1667#issuecomment-770461311

gtxaspec, 2023-12-17:
serial logs from the UART would most certainly tell us whats going on... I couldn't find any concrete information on what exactly they changed for compatability

Semag, 2023-12-17:
I'm still trying to find my serial cable to get the logs. However, based on the thread from @Answer-1 -  I went ahead and grabbed the RTSP firmware and unpacket it compared to the 4.9.8.1022 firmware for WZ Mini hacks.  Interestingly, the *kernel* is identical between the two files.  GTXASpec - i'm wondering if i use the RTSP Firmware and swap out the kernel with your custom kernel, if it's the same equivalent of WZ_mini_hacks.

md5Sums for the 4.9.8 FW
50deec4b53dde03c3efe98535e2f5b16  4.9.8.1022FW/fw_dir/appfs.bin
e9e8f7b18e57845bf6835a9c4b5788f1  4.9.8.1022FW/fw_dir/driver.bin
bee7951307dc53edf59599cd15294a88  4.9.8.1022FW/fw_dir/kernel.bin
c4ac3804e50cae555a21d2afbc63f26a  4.9.8.1022FW/fw_dir/rootfs.bin

**md5Sums for the RTSP Fw**
a6c5e074ccd95a1b99c9aec05bfad9e3  RTSP_4.28.4.49/fw_dir/appfs.bin
8575be9608a78e2ef4c8a53fa0dfca9d  RTSP_4.28.4.49/fw_dir/driver.bin
bee7951307dc53edf59599cd15294a88  RTSP_4.28.4.49/fw_dir/kernel.bin
d5701e43ca2c09b07b361cb4160c015f  RTSP_4.28.4.49/fw_dir/rootfs.bin

gtxaspec, 2023-12-17:
It'll crash probably, you need Wi-Fi drivers compiled too