gtxaspec/wz_mini_hacks #680: WyzeV3 upgrade-run.sh stops wz_mini_hacks

johnvpr, 2023-12-06:
I reset  my WyzeV3 download to SD the required files/folder from:
https://github.com/gtxaspec/wz_mini_hacks/archive/refs/heads/master.zip

then once installed configured in the wyze app 

then  ssh to the camera and run upgrade-run.sh
 
Now I cant access via SSH and the web UI is down.

Am I doing something wrong?

is there a command to see on which release the hack is running on?

Thanks

John

virmaior, 2023-12-06:
if you can't access via SSH or the web interface, most likely the hack isn't running.

My guess is:

> then once installed configured in the wyze app

my understanding is that the wyze app now auto upgrades to an incompatible version.

If not that, then the first question is whether the SD card is FAT32 or ExFAT.  Once booted, ExFAT is fine, but the wz_mini hack needs FAT32 because it depends on a vulnerability of the bootloader which can only read FAT32.

virmaior, 2023-12-07:
> so how do I tell which wz_mini_hacks release is on the camera?

if you can use the webserver then it displays it on the config.cgi page.
if you can ssh, then you can get it using `cat /opt/wz_mini/usr/bin/app.ver`

if you can't do either, then I think you're going to have to pull out the SD card and find the `/wz_mini/usr/bin/app.ver`  file ... but that probably means either you don't have this set up right or it's not running at all.

johnvpr, 2023-12-08:
So I am on. 4.36.9.139
[image]

if I go here I am not able to see the release number:
https://github.com/gtxaspec/wz_mini_hacks/wiki/Releases#changelog

is an other place where I can see the release 4.36.9.139 when was released?

Thanks!

johnvpr, 2023-12-08:
Ok, got it, the release is this one:
[image]

Thanks!

virmaior, 2023-12-08:
yes.

ktetreault, 2023-12-20:
Hello, are you able to link over to the firmware that you put on there? I cant seem to find the required version anywhere, only a .61

virmaior, 2023-12-20:
https://github.com/kohrar/Wyze-Firmwares/tree/master/v3