gtxaspec/wz_mini_hacks #683: UI dont load and get wz_mini.conf is missing the line WEB_SERVER_OPTIONS

johnvpr, 2023-12-08:
some camera return this if I go to http://10.10.40.217/
[image]

however loads if go to:
http://10.10.40.217/cgi-bin/config.cgi
[image]

bjs-pdx, 2023-12-10:
Did you follow this section of the config file:

Web Server

WEB_SERVER_ENABLED="true"
WEB_SERVER_AUTH="true"
WEB_SERVER_LOGIN="admin"
WEB_SERVER_PASSWORD=""

Enables the local webserver, for configuration, car control, or to retreive an image snapshot via a web browser. Available at : http://<Camera IP>/ Thank you @virmaior! Similar to the RTSP default password, if you don't set the password, the password will be set to the unique MAC address of the camera, in all uppercase, including the colons. If you do not want to have an authentication prompt for the local webserver, set WEB_SERVER_AUTH to false.

WEB_SERVER_OPTIONS="cam config car jpeg"`

This adds access controls limiting which parts of the web server are exposed.

    cam = cam.cgi
    config = config.cgi
    car = car.sh
    jpeg = jpeg.cgi

virmaior, 2023-12-15:
yeah, more so the error is backwards. Without the `WEB_SERVER_OPTIONS` line, it should prevent you from accessing any of the pages.
It is telling you to add it ...