gtxaspec/wz_mini_hacks #687: /dev/mmcblk0p1 does not mount to /media/mmc even though wz_mini_hacks is loaded

husain3, 2023-12-17:
I have 2 WyzeCam V3s that intermittently mounts `/dev/mmcblk0p1` to `/media/mmc`. I have a script to reboot every night and sometimes `/media/mmc` get mounted and sometimes it just doesn't. 

I can't record events when nothing is mounted.

I did try to manually run:

`mount -t vfat -o rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro /dev/mmcblk0p1 /media/mmc`

but I would then get this error:

`mount: mounting /dev/mmcblk0p1 on /media/mmc failed: Resource busy`

I've been troubleshooting this problem for months and I'm stumped on how to fix it.

gtxaspec, 2023-12-17:
what firmware version are you running on the camera?

husain3, 2023-12-17:
Firmware version is: 4.61.0.3

I ran `gather_wz_logs.sh` and in mount.log the mount to /media/mmc doesn't show up. Am I looking at the right log file?

gtxaspec, 2023-12-17:
just run `mount` and share that if you can

husain3, 2023-12-17:
Here you go! The last line is an NFS share to rsync my record folder (if it mounts)

```
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=46572k,nr_inodes=11643,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /opt type vfat (ro,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mtdblock2 on / type squashfs (ro,relatime)
/dev/mmcblk0p1 on /bin/busybox type vfat (ro,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/fstab type vfat (ro,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/inittab type vfat (ro,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/profile type vfat (ro,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /lib/modules type vfat (ro,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/hostname type vfat (ro,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /opt/wz_mini/tmp type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
/dev/mtdblock3 on /system type squashfs (ro,relatime)
tmpfs on /etc/shadow type tmpfs (rw,relatime)
/dev/mmcblk0p1 on /system/bin/factorycheck type vfat (ro,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,relatime,mode=755)
media on /media type tmpfs (rw,relatime)
/dev/mtdblock6 on /configs type jffs2 (rw,relatime)
none on /tmp/Upgrade type tmpfs (rw,relatime,size=4k,nr_inodes=1)
192.168.1.2:/media/wyze_cam_captures on /media/nfs_drive type nfs (rw,relatime,vers=3,rsize=524288,wsize=524288,namlen=255,hard,nolock,proto=tcp,port=2049,timeo=600,retrans=2,sec=sys,mountaddr=192.168.1.2,mountvers=3,mountproto=tcp,local_lock=all,addr=192.168.1.2)
```

husain3, 2023-12-17:
Thanks for your swift response by the way. I didn't think anyone would've responded.

husain3, 2023-12-17:
I was running through the logs of iCamera and came across this:

`Jun 21 23:48:22 assis: 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.`
`Jun 21 23:48:22 assis:  Automatically removing dirty bit.`

I'm not sure if the assis process actually removes the dirty bit (because /media/mmc still didn't mount when I rebooted). But I ran `fsck.vfat -a /dev/mmcblk0p1` from `/opt/wz_mini/bin` and rebooted. Now it mounts.

Please note: I had to run `fsck.vfat` rather than `fsck`.

I will keep monitoring this bug and run the above command when it happens again. If it always fixes the problem I could write a BASH script to automate. Is `fsck.vfat` used automatically anywhere?

husain3, 2023-12-18:
While playing around to replicate the bug, I've noticed that the dirty-bit usually gets written to the SD Card when it's recording something and reboot is called. On boot, `tf_prepare` gets called to automatically remove the dirty-bit and **in most cases** it successfully does so. I am assuming the bug I am encountering is the case where `tf_prepare` cannot remove the dirty-bit automatically and thus cannot mount to /media/mmc.

After running `fsck.vfat -a /dev/mmcblk0p1` and then `reboot` the SD Card mounts successfully afterward. 

However, scouring the GitHub Issues forum I notice there's an option in `wz_mini.conf` called `ENABLE_FSCK_ON_BOOT="true"` that would do this automatically. I will replicate the dirty-bit switch a few more times and check for success rate at SD Card repair and mount at boot.

virmaior, 2023-12-20:
@husain3 this might be a very critical piece to a puzzle related to reboots and SD corruption.

Maybe we need to use a different process to reboot.

husain3, 2023-12-20:
> Maybe we need to use a different process to reboot.

Good point. As it stands right now my cron job reboots using the `reboot` command. I was poking around in the shell scripts that came in the firmware and noticed this is also how reboot is done.

We can't modify iCamera that writes to the SD Card, but perhaps we could write a script that lazily unmounts `/dev/mmc` and won't reboot until it sees it is successfully unmounted? Lazy unmount meaning that it would only unmount once the SD Card isn't busy anymore.

husain3, 2023-12-20:
Ok nevermind, I tried a dry run of that method (unmount when SD Card is not busy) and I still flipped the dirty-bit.