gtxaspec/wz_mini_hacks #709: Two Way Audio Not working

yankees5963, 2024-01-13:
in Go2RTC there is a webrtc option for two way audio, i can hear the audio coming in, but theres no option to talk into a mic. Same issue within frigate card in homeassistant where the mic button is enabled but not showing up (indicating the camera cannot use mic).

I suppose best solution would be to not depend on go2rtc and create a ONVIF option for frigate to talk directly to the camera (saw it was suggested but not actively worked).

gtxaspec, 2024-01-13:
two way audio is not supported for wz_mini.  It's supported on ONLY devices which support RTSP backchannel.

yankees5963, 2024-01-13:
could that feature be added given the current system? Sort of like a remote cmd command that sends the shortly recorded audio to the speaker? I would imagine that would have a good bit of latency though.

gtxaspec, 2024-01-13:
Not with the current wyze system, no.  It's working with ope ipc with custom components, though

AlexxIT, 2024-01-24:
Not ONLY RTSP backchannel... :)
https://github.com/AlexxIT/go2rtc?tab=readme-ov-file#two-way-audio