gtxaspec/wz_mini_hacks #711: go2rtc stopped working ?!?

Answer-1, 2024-01-18:
Running wz on two v2 and 3 v3's.  They all got new SD cards, got flashed and were all working perfectly.
Until, out of nowhere, one of them would not work in Frigate anymore using go2rtc. Started debugging a bit and the hack was still running fine, ssh and web GUI access.  I could see the stream from the config page so I knew it was somewhat working, but the go2rtc streams was not. Under the go2rtc page, I tried "manually" streaming the feed and got a "segmentation fault, core dumped" error from the browser.
At that point, I netstat port 8554 and say Frigate establishing a connection, but that connection wasn't staying established and just got into time_wait. Frigate was at first reporting the DESCRIBE wasn't correct, then a 404.
I disabled go2rtc, enabled v4l2 and that stream is working (although seemingly more buggy, but working).
So, I'm not understanding why go2rtc isn't working anymore. Since the first one, I've got a second one with the exact same issue.
I'll probably try to reflash the SD card and start from new, but still, pretty weird stuff.

Any idea on what this could be?  Wyze Firmware firmware wasn't updated.

flan7, 2024-02-28:
I am facing this same issue, but only on two of my cameras. Likely after a recent frigate update.

Answer-1, 2024-02-28:
Happened again after 2-3 weeks running time. Same thing, camera was in a boot loop. I did saw some "Segmentation fault" errors in a log somewhere, but lost it. Back to running WyzeBridge for those cams, it's pretty stable since I fixed the wifi issues with the cams.

jmbulan, 2024-02-28:
Are these v2 or v3?

Answer-1, 2024-02-29:
I was testing Dafang on my v2's so the two that exhibited the issue are v3's.
Looks like some files are getting corrupted on the SD card thus rendering the cam non bootable. A quick format, recopy of the files and it gets back up and running.

ejeffords, 2025-01-06:
I've experienced the same issue on one of my cams recently.  In my diagnostics on boot, I'm seeing "drivers/rtc/hctosys.c: unable to open rtc device (rtc0)"

I'm sure a flash of the card and reset of everything would fix it, but still would like to know what triggers this error.