gtxaspec/wz_mini_hacks #716: ONVIF Support

nschimme, 2024-01-24:
Can we implement ONVIF support?

There is some work already done as part of the Dafang-Hacks project: https://github.com/Dafang-Hacks/Main/tree/master/onvif

virmaior, 2024-01-31:
Was going to try to bring the MQTT code that someone put into a fork a while back ago as an option.

I've looked at ONVIF and the documentation as far as I can tell suffers from the "sandbox problem" (or rather its like - emacs -- would you like a text editor with your emacs operating system?)

nschimme, 2024-02-01:
I would suggest we limit the features to the bare minimum. Motion and PTZ are nice to haves but could be extensions.

gord888, 2024-11-11:
Howdy folks - i realize this is an old thread, but if you're still looking to get your wz_mini_hacks Wyzecam working with ONVIF i did figure out a solution. 

There's a software here:  https://www.happytimesoft.com/products/onvif-server/index.html that implements a VERY basic ONVIF server.  You can change the configuration of the ONVIF server so that it responds with the RTSP stream that points to your WyzeCam.  I'm currently using this method so i can get the WyzeCam to work with my Reolink NVR and it's been working alright.

nschimme, 2024-11-11:
I've since moved on to Thingini which has an ONVIF server implementation

Danimal4326, 2024-11-13:
Check out this project for a self contained OSS onvif server:
https://github.com/p10tyr/rtsp-to-onvif