gtxaspec/wz_mini_hacks #720: Wyze v2 uses wrong device nodes in Go2RTC

reidprichard, 2024-02-02:
I just installed on a Wyze v2 (have had it running on three v3s) and found the RTSP streams didn't work. After a bit of digging, I found that the device nodes in the Go2RTC config were set to the defaults of /dev/video1 and /dev/video2 (1080p and 360p, respectively) rather than /dev/video6 and /dev/video7 as the comment in the Go2RTC config helpfully stated. Fixing the nodes got everything working.

I'm not sure if it's possible to auto-detect the device type and set the nodes in the Go2RTC config accordingly. If it's not, perhaps a note in the docs stating that this change is necessary for the v2?

Mathews2115, 2024-02-10:
I think it's commented somewhere here but for v2 and pan v1 cams, you'll have to go update the go2rtc config in `wz_mini\etc\go2rtc`.  You can manually change them to video6/7 there.

gtxaspec, 2024-02-10:
see https://github.com/gtxaspec/wz_mini_hacks/wiki/Video-Streaming

reidprichard, 2024-02-16:
Ahh, I see that now. Would it be easy to add a go2rtc config item to select camera type and update this behind the scenes? Or maybe I'm the only one using the v2 and it's not worth the trouble. At any rate, hopefully this issue makes the solution more googleable.

bjs-pdx, 2024-05-26:
I just found this.  Was having problems with a V2.  Might be worthwhile noting it in the wiki somewhere (config file entries?).  It isn't exactly obvious that you need to change the go2rtc.yaml file to make this work.

ctrl50, 2025-02-22:
took me two days to find this and update the file accordingly.

On kubuntu 24. when I ssh into the camera and open go2rtc with nano it cuts off the lines of code with the ending > before it gets to the video1, and video2 configs. have to fullscreeen the terminal and then open the file before it eveb shows it.

PErhaphs its just my setup., but i overlooked those little details untill i found this information.

Clairifying the comment in that file by adding something such as "you must change video1 and video2 in the following code block for v2 cameras" would help, I seen where it said it must be changed, but didnt see the video1,2, coding due to t h e above reasons.