gtxaspec/wz_mini_hacks #729: ENABLE_MP4_WRITE

unoukujou, 2024-02-25:
Can somebody explain more about ENABLE_MP4_WRITE?

What is it doing? Is this for the continuous recording? If i enable, then Where do I choose the location to save?

gtxaspec, 2024-02-25:
Did you check the wiki?
https://github.com/gtxaspec/wz_mini_hacks/wiki/Configuration-File#mp4_write

unoukujou, 2024-02-25:
> Did you check the wiki? https://github.com/gtxaspec/wz_mini_hacks/wiki/Configuration-File#mp4_write

Yes. But just didn't understand, is this setting meaningless when using the internal SD card for storage? If I set it to true, where do I set the location path to save to?

endertable, 2024-02-25:
@unoukujou  This setting only has to do with writing to your SD card.  Normally, on newer cameras, the binary writes the 1 minute mp4 video file to /tmp, then moves it to the SD card.  This setting allows you to override that process and have it write the mp4 on the SD card directly, like some of the older cams did, e.g. the V2.  Anything on the /tmp directory uses precious RAM and the video files can take up 10MB+

unoukujou, 2024-02-25:
Good to know! Thanks!