gtxaspec/wz_mini_hacks #738: jzmmc_v1.2 jzmmc_v1.2.0: timeout

tdashmike, 2024-03-03:
Are these errors normal, I'm getting a ton of these but the camera seems to be working ok. I'm on wz_mini Wed Oct 25 12:53:39 AM PDT 2023. Firmware Version: 4.36.9.139.

```
46465] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x0000F200 blks:6/128 clk:enable clk_gate:enable
[99259.676449] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x00008C00 blks:57/128 clk:enable clk_gate:enable
[99441.316464] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x00009C00 blks:49/128 clk:enable clk_gate:enable
[99624.816496] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x00005C00 blks:81/128 clk:enable clk_gate:enable
[99913.666447] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x0000F200 blks:6/128 clk:enable clk_gate:enable
[99989.826451] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x0000F400 blks:5/128 clk:enable clk_gate:enable
[99990.336465] jzmmc_v1.2 jzmmc_v1.2.0: timeout 1000ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x0000EC00 blks:9/128 clk:enable clk_gate:enable
[100880.276477] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:3584 state:2 STAT:0x1E010580 DMALEN:0x00000000 blks:6/7 clk:enable clk_gate:enable
[100891.826449] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x0000F200 blks:6/128 clk:enable clk_gate:enable
[100913.956447] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:18432 state:2 STAT:0x1E000580 DMALEN:0x00000600 blks:32/36 clk:enable clk_gate:enable
[100914.466463] jzmmc_v1.2 jzmmc_v1.2.0: timeout 1000ms op:25 w sz:18432 state:2 STAT:0x9E011140 DMALEN:0x00000000 blks:36/36 clk:enable clk_gate:enable
[101676.036444] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:3584 state:2 STAT:0x1E010580 DMALEN:0x00000000 blks:6/7 clk:enable clk_gate:enable
[101783.656479] jzmmc_v1.2 jzmmc_v1.2.0: timeout 500ms op:25 w sz:65536 state:2 STAT:0x1E000580 DMALEN:0x0000F200 blks:6/128 clk:enable clk_gate:enable
```

gtxaspec, 2024-03-03:
bad power supply, not enough voltage to drive everything plus the sd card.
also could be bad sd card.

tdashmike, 2024-03-03:
Thanks, I’ll try a few things.