gtxaspec/wz_mini_hacks #746: ntpd not updating system time

kevdliu, 2024-03-14:
I'm having trouble setting the camera's system time from my ntp server. I see that the camera defaults to the firmware built time on boot, but ntpd is not updating the time to the current time. Running `ntpd -ndp 192.168.1.1` outputs a string of 
```
ntpd: sending query to 192.168.1.1
ntpd: reply from 192.168.1.1: offset:+57972971.758940 delay:0.001649 status:0x24
```
so it looks like ntpd is able to talk to my NTP server. I've read that ntpd adjusts the time gradually, but the offset doesn't seem to be getting any smaller over a period of a few hours. 

I can set the time manually using `date -s` but that doesn't persist across reboots. 

I'm using a T31 running firmware 4.36.9.139. wz_mini was built on 2024-01-09_23:55:50. 

Any help would be greatly appreciated!

amandamana, 2024-03-16:
It's not intuitive and not well documented, but I believe you need to specify the port along with the NTP server IP. 
https://github.com/gtxaspec/wz_mini_hacks/issues/673#issuecomment-1821339141

kevdliu, 2024-03-18:
> It's not intuitive and not well documented, but I believe you need to specify the port along with the NTP server IP. [#673 (comment)](https://github.com/gtxaspec/wz_mini_hacks/issues/673#issuecomment-1821339141)

I tried that to but unfortunately it didn't help

AwesomenessZ, 2024-04-17:
Using your command worked for me after letting it run for about a minute:
```root@sideyard:~]# ntpd -ndp 10.3.0.1
ntpd: sending query to 10.3.0.1
ntpd: reply from 10.3.0.1: offset:+60592569.440024 delay:0.093317 status:0x24 strat:3 refid:0x0f6930c0 rootdelay:0.133806 reach:0x01
ntpd: sending query to 10.3.0.1
ntpd: reply from 10.3.0.1: offset:+60592569.482305 delay:0.009428 status:0x24 strat:3 refid:0x0f6930c0 rootdelay:0.133806 reach:0x03
ntpd: setting time to 2024-04-17 04:29:41.367211 (offset +60592569.482305s)
ntpd: sending query to 10.3.0.1
ntpd: reply from 10.3.0.1: offset:+0.001388 delay:0.005082 status:0x24 strat:3 refid:0x0f6930c0 rootdelay:0.133806 reach:0x07
ntpd: sending query to 10.3.0.1
ntpd: reply from 10.3.0.1: offset:-0.000359 delay:0.002740 status:0x24 strat:3 refid:0x0f6930c0 rootdelay:0.133806 reach:0x0f
ntpd: sending query to 10.3.0.1
ntpd: reply from 10.3.0.1: offset:+0.001609 delay:0.006582 status:0x24 strat:3 refid:0x0f6930c0 rootdelay:0.133806 reach:0x1f
```

Using  (and installed this camera on) `T31(WYZE_CAKP2JFUS) on 4.36.9.139`

Using your command helped me discover that the FQDN does not work on this field, so "router.lan:123" does not work, but "10.3.0.1:123" does for me.

kevdliu, 2024-04-25:
I just tried it again and it's working now. Maybe there was a network issue the first time or maybe I just didn't wait for long enough. Anyway thanks for confirming that the command worked which led me to try it again myself!

closed 2024-04-25 by kevdliu