gtxaspec/wz_mini_hacks #749: Wyze Cam V3 stops working after one to two minutes, becomes unresponsive

eliluong, 2024-03-26:
I am having an issue setting up wz_mini_hacks on a Wyze Cam V3. The camera has been flashed with firmware 4.36.9.139 and initial setup done with the Wyze app.

I was able to get an RTSP stream to display for a minute or two, but then it will stop working. I am no longer able to access the Web UI. If I power cycle the camera, I am unable to access anything again. It will start with red light, then red-blue, and then no light. Only by wiping the SD card and letting it set up wz_mini_hacks again will I be able to access the Web UI (for a few minutes). My goal is to record to an NVR.

If I remove wz_mini_hacks, I can access the camera with docker-wyze-bridge.

My configuration is attached. I'm unsure where to look, as none of the log files indicate anything. Thank you.

[wz_config.txt](https://github.com/gtxaspec/wz_mini_hacks/files/14750882/wz_config.txt)

virmaior, 2024-03-26:
if ssh into the camera during that brief window, and do `ls /configs` what are the results?

eliluong, 2024-03-26:
This is the output of `ls /configs`:

`
[root@WYZEV3-1:~]# ls /configs/
TZ          app.ver     os-release
`

After a few minutes, I still see blinking blue light, so I ran the same command again. But after disconnecting from SSH, I will not be able to reconnect.

`
[root@WYZEV3-1:~]# ls /configs/
Segmentation fault
`

This was done after setting up the camera with Wyze app, then loading fresh wz_mini_hacks copy.

clee604, 2024-03-26:
one thing I would suggest is trying another sd card to eliminate that as a possible cause

virmaior, 2024-03-26:
it's possible that either the SD card or the internal storage is faulty on your camera.

eliluong, 2024-03-27:
> it's possible that either the SD card or the internal storage is faulty on your camera.

Thanks. I will try different SD card, like above mentioned. But, do you know what I can do to tell if internal storage is faulty?

virmaior, 2024-04-10:
no idea on the internal storage.
another possibility is that the power to the camera is insufficient and leading to instability.

hnws, 2024-08-06:
It's difficult to believe, but it happened to me that the changing the power adapter fixed this issue for me.
The camera behavior is the same as this issue. And the faulty power adapter is used indoor for 5 months.