gtxaspec/wz_mini_hacks #752: Wyze Cam V4 Support

digitydogs, 2024-03-30:
Please add support for the new V4 cams if possible, and if not please let us know and post that info on the wiki as well.

virmaior, 2024-04-03:
wasn't able to figure out what the processor is but if it's not an ingenic, no dice for this project.

UnKnoWn-Consortium, 2024-04-11:
It would be great if the new v4 is supported but I won't keep my hopes up given how tightened down v3 Pan and v3 Pro are.

digitydogs, 2024-04-11:
We already know they are using the proper chip type, so the only real potential hang up is if they use secure boot.  If it has secure boot I can assure you there will be people looking into breaking it open.

claudobahn, 2024-04-11:
See https://github.com/gtxaspec/wz_mini_hacks/issues/760 - secure boot is enabled as already confirmed by gtxaspec

incredimike, 2024-10-26:
Any progress on V4 support?

virmaior, 2024-10-27:
> Any progress on V4 support?

I don't think that's going to happen for wz_mini_hacks for two reasons
1. development has basically ended and moved to thingino
2. V4 has secure boot meaning that the processor has a hard-coded check to only use wyze produced firmwares (https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/oem-secure-boot). Unless/until someone finds a way around that, nothing like wz_mini_hacks will work.

gtxaspec, 2024-10-27:
Some folks are trying to experiment with an SOC swap on the v4... looks like this is the only way for now.  Stay tuned.