gtxaspec/wz_mini_hacks #753: File '/opt/wz_mini/wz_mini.conf' is unwritable

mrkonyk, 2024-04-01:
Trying to edit conf on Wyze Cam V2. 

Cameras working just can't edit "/opt/wz_mini/wz_mini.conf". Tried, "mount | grep "/opt/wz_mini" and "ls -l /opt/wz_mini/wz_mini.conf"

Here's the results, tmpfs on /opt/wz_mini/tmp type tmpfs (rw,relatime) and -rwxrwxrwx    1 root     root          2753 Mar 31 03:37 /opt/wz_mini/wz_mini.conf

Anyone come across this?

virmaior, 2024-04-04:
`tmpfs on /opt/wz_mini/tmp` but what is the mnt for `/opt`?

on a v3, I see `/dev/mmcblk0p1 on /opt type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
`

RASTRILL0, 2024-04-07:
I have this problem on both my V2 cameras. I have to use this command to edit the conf file:

`mount -o remount,rw /dev/mmcblk0p1 /opt`