gtxaspec/wz_mini_hacks #758: WIKI Information for Config File Contains an error

bjs-pdx, 2024-04-08:
I was struggling to setup a new V3 camera with the most recent version of the files.  I was modifying the configuration file to enable features for the web server, but after I did, wz_mini_hacks wouldn't load.  I was copying the setting from the wiki but those settings contain an additional character (a single quote) so the camera would never boot (note the last character in the line below).  The extra character barely visible.

WEB_SERVER_OPTIONS="cam config car jpeg"`

[image]

xOnlyFadi, 2024-04-09:
what version is ur v3 camera on?

gtxaspec, 2024-04-09:
fixed!

closed 2024-04-09 by gtxaspec

bjs-pdx, 2024-04-09:
> what version is ur v3 camera on?

[image]