gtxaspec/wz_mini_hacks #759: Bash Script Loading Problem on reboot

xOnlyFadi, 2024-04-09:
Hi I have a problem with my motion script to run when reboot and run in background i tried custom path location and then tried `/opt/wz_mini/etc/rc.local.d` still no motion log on my webhook only run it manually it works

here is my bash script
```bash
#!/opt/wz_mini/bin/bash

while true
do
  ret=$(cmd waitmotion 1000 | tr -d '\0')
  #printf "$ret\n"
  if [[ "$ret" == "detect"* ]] ; then
    echo "Reporting motion!"
    curl "webhook"
    sleep 1
  else
    echo "No motion found!"
    sleep 1
  fi
done
```

corbolais, 2024-07-27:
Hi @xOnlyFadi ,

Here, the scripts are located in '/opt/wz_mini/etc/rc.local.d/', file name like 'motion.sh' or 'motion'. For the latter to work, you have to patch  '/opt/wz_mini/etc/init.d/wz_init.sh' line 52ff. to look like:

```
*)
# No sh extension, so fork subprocess.
sh $i start
```
(just put the "sh" in front of the line)

HTH

xOnlyFadi, 2024-08-03:
Hi, @corbolais 

Can you show an example inside `/opt/wz_mini/etc/init.d/wz_init.sh`