gtxaspec/wz_mini_hacks #760: Wyze V4 HL_CAM4

gtxaspec, 2024-04-10:
```
U-Boot SPL 2013.07 (Nov 27 2023 - 16:59:23)
Board info: T41NQ

apll_freq = 1104000000 
mpll_freq = 1440000000 
vpll_freq = 1188000000 
sdram init start
DDR clk rate 720000000
DDRP_INNOPHY_PLL_CTRL: 00000028
DDRP_INNOPHY_PLL_CTRL: 00000020
T41NQ ddr phy skew set﻿
mem data:
81000000:   00000000  ffffffff  00000000  ffffffff
81000010:   00000000  ffffffff  00000000  ffffffff
81000020:   00000000  ffffffff  00000000  ffffffff
81000030:   00000000  ffffffff  00000000  ffffffff
image entry point: 0x80100000
Board: ISVP (Ingenic XBurst2 T41 SoC)
DRAM size is 128 MiB
DTB: 0x8013aefc
Top of RAM usable for U-Boot at: 88000000
Reserving 427k for U-Boot at: 87f94000
Reserving 32772k for malloc() at: 85f93000
Reserving 32 Bytes for Board Info at: 85f92fe0
Reserving 124 Bytes for Global Data at: 85f92f64
Reserving 128k for boot params() at: 85f72f64
Stack Pointer at: 85f72f48
Now running in RAM - U-Boot at: 87f94000
MMC:   MSC: 0
Using default environment

Data transfer:    serial
gpio_request lable = wifi_enable_gpio gpio = 78
gpio_request lable = red_gpio gpio = 39
gpio_request lable = blue_gpio gpio = 38
gpio_request lable = white_light gpio = 48
gpio_request lable = white_light power en gpio = 47
gpio_request lable = 850_light_gpio gpio = 46
gpio_request lable = SPK_able_gpio gpio = 63
gpio_request lable = TF_SD_M_gpio gpio = 59
gpio_request lable = TF_sdcard_en_gpio gpio = 57
gpio_request lable = TF_CD_N_gpio gpio = 58
gpio_request lable = SD_power_en_gpio gpio = 61
gpio_request lable = wifi_enable_gpio gpio = 78
Hit any key to stop autoboot:  0 
gpio_request lable = sdupgrade gpio = 62
SF: Detected XT25F128B, flash size: 16MB, manufacturer id: 0b

SF: Detected XT25F128B, flash size: 16MB, manufacturer id: 0b

### JFFS2 loading 'upGradeFlag' to 0x85f72140
Scanning JFFS2 FS: . done.
find_inode failed for name=upGradeFlag
load: Failed to find inode
### JFFS2 LOAD ERROR<0> for upGradeFlag!
gpio_request lable = TF_cd_gpio gpio = 58
SF: Detected XT25F128B, flash size: 16MB, manufacturer id: 0b

--->probe spend 6 ms
SF: 4653056 bytes @ 0x30000 Read: OK
--->read spend 1502 ms
## Loading kernel from FIT Image at 80600000 ...
   Using 'conf' configuration
   Verifying Hash Integrity ... sha256,rsa2048:kernel-dev+ sha256,rsa2048:kernel-dev+ OK
   Trying 'kernel' kernel subimage
     Description:  ingenict41 kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x80604240
[... 333 lines cut ...]
[    3.365225] [atbm_log]:ble start
[    3.369140] [atbm_log]:[ALTOBEAM-ble_recv] start
[    3.374051] [atbm_log]:ble subscribe
[    3.374062] [atbm_log]:[ALTOBEAM-ble_xmit] start
[    3.383028] [atbm_log]:ble_platform_probe(828c3880)
[    3.388310] [atbm_log]:is registered as 'ALTOBEAM'
[    3.393488] [atbm_log]:default country code:00,support channel=14
[    3.400009] [atbm_log]:[atbm_wtd]:set wtd_probe = 1
net.core.wmem_max = 26214400
net.core.wmem_default = 26214400
Updating device time to:
Wed Nov 29 02:31:16 UTC 2023
[VER]
appver=4.52.1.7738
gitsha=f973f7f633-dirty
#######################
#   IS USER STATUS   #
#######################
kernel.core_pattern = |/system/bin/ucoredmp_collector.sh --pid %p --signal %s --name %e --time %t --output-dir /media/mmc/cores
kernel.core_pipe_limit = 1
net.unix.max_dgram_qlen = 128

Welcome
t41 login: 
```

gtxaspec, 2024-04-10:
secure-boot enabled, verified.  investigating further.

closed 2024-04-10 by gtxaspec

digitydogs, 2024-04-12:
May be of some use in getting past the secure boot..  There are a few gits dedicated to ingenic booting ;) 

https://github.com/OpenIPC/u-boot-ingenic/blob/master/README