gtxaspec/wz_mini_hacks #769: Unable to Remove Current Settings

saihgupr, 2024-05-15:
I'm attempting to remove the current value of VIDEO_0_MAX_BITRATE, which I previously set but now want to remove. When I delete the entry in the configuration on the web server and click the update button, it still reverts to the original value. I suspect it has to do with a null object, but I'm not sure.

How can I remove this value without taking the camera down to edit the config on the SD card?

endertable, 2024-05-16:
you can ssh in and  delete the entry using vi.

Right afterwards, while still in ssh, issue an immediate sync then reboot, e.g.:

`sync; reboot ; exit`

saihgupr, 2024-05-16:
@endertable Thank you for your response! Unfortunately, I forgot to copy over the public SSH key, so I don't think I can SSH in. I tried connecting the USB cable directly to my computer to access the serial console, but it was unsuccessful.

saihgupr, 2024-05-20:
@endertable Thanks again for your input. I manned up and took down the camera to enable SSH. All is well again.

saihgupr, 2024-05-23:
If anyone finds this in the future, another way to solve this problem is to set ENABLE_FILESERVER to true. Then you can access your wz_mini.conf at http://[ip]:5000/wz_mini and edit it from there.

closed 2024-06-18 by saihgupr