gtxaspec/wz_mini_hacks #775: Unable to SSH or get into Web Server Menu

CowSkullBolo, 2024-07-06:
Hi all,

For the past couple of days I have been doing my best to get things loaded onto my two V3 cameras and I finally gave up and decided to come here. I have followed all of the directions and tried several different options in the config file to see if I can get this going. I'm primarily working with MacOS but have access to Windows/Linux with VM. SD Cards are 32gb. I think I have read the instructions several times, but I done so much troubleshooting and research that I'm starting to go cross-eyed/mildly crazy and it's possible I might be missing something.

Troubleshooting:
1.) Check browser/router settings re: http access
2.) Disable AdGuard and VPN
3.) Disable Little Snitch
4.) Try SSH and check http page from separate computer
5.) Try SSH and check http page by VM on Windows/Debian
6.) Format card in FAT32 with Apple Disk Utility, Raspberry Pi Imager, and SD Card Formatter
7.) Go to Staples and get a new 32gb SD card
8.) Downgrade firmware to 9.131, upgrade to 9.139, load hacks
9.) Download BBEdit for config edits for safe measures
10.) Ensure camera has ip reservation

I have followed the ssh-keygen instructions and have checked it several times.

I'm able to ping the camera, but I can't SSH in. 
`ssh: connect to host xxx.xxx.x.xxx port 22: Connection refused`

I don't think I'm a computer wiz or anything, but I have some experience and frequently ssh into my router and a single board computer that I am planning to run MotionEye on. 

The only thing that I can think that may be going wrong is:
1.) My router settings
2.) My Wyze v3 is not booting the hack for some reason. 

I'm leaning on #2. Please let me know if I'm missing any helpful information.

robughblah, 2024-07-06:
Hey CowSkullBolo,
I have a mac with Sonoma, use BBEdit and have Little Snitch running and use wz_mini hacks to ssh only into several V3 cameras.
Typically the SD format nails you, or config file terminations or content. Then there is just typing wrong things. Been there, done that.

Your post inspired me to repair the USB cable on a spare V3 and write up some initial troubleshooting steps for you. The firmware version can be current if you only 
desire to ssh or sftp into your camera, which is all i need.  

Note: don’t do a Wyze camera firmware upgrade while running wz_mini_hacks, it will eat it’s brain and you will need to reflash the firmware and reinstall wz_mini-hacks.

Try these steps, note where it goes wrong and attach your full-list.txt.gz file for additional kicks at the can.

Rob

July 6 2024- SD CARD and WYZE MINI HACKS INSTALL and SSH VERIFICATION for MAC
#power off camera. Insert SD card into your mac

#get a list of all disks your mac can see. 
Want to obtain the Volume name, and verify TYPE = Windows_FAT_32
Open mac Terminal app (in /Applications/Utility)
-type:
df -h
diskutil list

Example:
***@***.*** ~ % df -h
Filesystem        Size    Used   Avail Capacity iused ifree %iused  Mounted on
... (lines omitted)
/dev/disk4s1      15Gi   115Mi    15Gi     1%       1     0  100%   /Volumes/WYZE

***@***.*** ~ % diskutil list
... (lines omitted)
/dev/disk4 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *15.9 GB    disk4
   1:             Windows_FAT_32 WYZE                    15.9 GB    disk4s1

#do a repair on your SD card. Note the IDENTIFIER value for the last command.
-type:
diskutil repairVolume 'IDENTIFIER'

Example:
***@***.*** ~ % diskutil repairVolume disk4s1
Started file system repair on disk4s1 (WYZE)
Checking file system and repairing if necessary and if possible
Volume was successfully unmounted
Performing fsck_msdos -y /dev/rdisk4s1
** /dev/rdisk4s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
** Phase 3 - Checking for Orphan Clusters
Warning: 979 files, 15432352 KiB free (482261 clusters)
File system check exit code is 0
Restoring the original state found as mounted
Finished file system repair on disk4s1 (WYZE)

#get a list of all files on your SD card in case you need to send it for more troubleshooting
-type:
ls -alR /Volumes/WYZE > full-list.txt
gzip  full-list.txt

#check the crlf termination of critical files wz_mini.conf and authorized keys, 
Use 'Mounted On' value from 'df -h' above.
-type:
file 'Mounted On'/wz_mini/wz_mini.conf
file 'Mounted On'/wz_mini/etc/ssh/authorized_keys

Example:
***@***.*** ~ % file /Volumes/WYZE/wz_mini/wz_mini.conf
/Volumes/WYZE/wz_mini/wz_mini.conf:     ASCII text

#you do not want to see terminators info in the next two lines
Example with deliberately wrong crlf terminations:
/Volumes/WYZE/wz_mini/wz_mini.conf-DOS: ASCII text, with CRLF line terminators
/Volumes/WYZE/wz_mini/wz_mini.conf-MAC: ASCII text, with CR line terminators

***@***.*** ~ % file  /Volumes/WYZE/wz_mini/etc/ssh/*
/Volumes/WYZE/wz_mini/etc/ssh/authorized_keys: ASCII text, with very long lines (569)

NETWORK VERIFICATION
#do not do this with the camera at a remote location.
#soft eject the SD card form your mac. Plug it into the camera. Power the camera on.
#the camera blink and say:
WZ MINI. Initializing first boot.
Extracting swap. Please wait.

#after a couple of minutes, lets test network connectivity with ping to ensure have correct IP
Open Wyze app, select camera, open Settings, click on Device Info.
record IP, Network and firmware version

Example:
IP=192.168.6.228
Network=apelike
Firmware Version=4.36.12.9751

#open mac Terminal app (in /Applications/Utility)
-type:
ping 'your cameras IP'

Example:
***@***.*** ~ % ping 192.168.6.228
PING 192.168.6.228 (192.168.6.228): 56 data bytes
Request timeout for icmp_seq 0
64 bytes from 192.168.6.228: icmp_seq=0 ttl=64 time=1071.414 ms
64 bytes from 192.168.6.228: icmp_seq=1 ttl=64 time=66.411 ms
64 bytes from 192.168.6.228: icmp_seq=2 ttl=64 time=1.409 ms
64 bytes from 192.168.6.228: icmp_seq=3 ttl=64 time=10.041 ms

#unplug camera. does ping commence timeouts?
Example:
64 bytes from 192.168.6.228: icmp_seq=9 ttl=64 time=4.430 ms
Request timeout for icmp_seq 11
Request timeout for icmp_seq 12

#plug camera into USB, does ping resume after a minute
Example:
***@***.*** ~ % ping 192.168.6.228
PING 192.168.6.228 (192.168.6.228): 56 data bytes
64 bytes from 192.168.6.228: icmp_seq=70 ttl=64 time=4.350 ms
64 bytes from 192.168.6.228: icmp_seq=71 ttl=64 time=2.175 ms
64 bytes from 192.168.6.228: icmp_seq=72 ttl=64 time=2.914 ms
^C

SSH TEST
#very exciting if you got this far. Open mac Terminal app (in /Applications/Utility)
-type:
ssh root@'Your IP'

Example:
***@***.*** ~ % ssh ***@***.***
The authenticity of host '192.168.6.228 (192.168.6.228)' can't be established.
ED25519 key fingerprint is SHA256:ufmewoni6aL1b62WCZTWaUWQ2brqRvnrBz/FvTRyA80.
This key is not known by any other names.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.6.228' (ED25519) to the list of known hosts.
COLUMNS=120;LINES=36;export COLUMNS LINES;

***@***.***:~]# uname -a
Linux WCV3 3.10.14__isvp_swan_1.0__ #120 PREEMPT Wed Jul 13 00:01:23 PDT 2022 mips GNU/Linux

***@***.***:~]# exit
Connection to 192.168.6.228 closed.
***@***.*** ~ %
***@***.*** ~ %
***@***.*** ~ % ssh ***@***.***
COLUMNS=120;LINES=36;export COLUMNS LINES;
***@***.***:~]#

***@***.***:~]# cd /media/mmc
***@***.***:mmc]# ls
alarm                   record                  wyze_config_backup      wz_mini_initramfs.log
factory_t31_ZMC6tiIDQN  time_lapse              wz_mini
***@***.***:mmc]#

CowSkullBolo, 2024-07-07:
Thanks for taking the time to help out! After reading your reply I went back and started over from scratch. Unfortunately, I'm still not having any luck. 

Running Sonoma 14.4.1
Wyze Firmware Version: 4.36.9.139
Wyze Cam v3

When connecting the wz_mini SD card, the camera loads up as solid red, then blinks red for a minute, and then goes solid blue. The camera gives no audible indication that the hack has been successfully installed. 

With all of this information, I'm wondering if the hidden Spotlight files are fudging things up. 

Another thing I want to note is that I did a little poking after I read your troubleshoot guide and noticed that the card I was working with yesterday was formatted as "DOS_FAT_32 SDCARD." I had been using a mixture of Disk Utility, SD Card Formatter and Pi Imager prior to this. I went into into my Windows VM and formatted it to Windows_FAT_32 SDCARD with DiskGenius. Disk Utility only gives the option of MBR, GUID and Apple Partition Map as options, and I know it needs to be DOS. I think I may be getting confused at this step.

I do like the idea of having a way to ssh into my camera, but I'm willing to do without if you think upgrading to a current firmware version if you think the hack would successfully load. Let me know if I should try anything different or if you need any additional info. I'm determined to get this going and happy to try again.

[full-list.txt](https://github.com/user-attachments/files/16117902/full-list.txt)

robughblah, 2024-07-07:
Good news CowSkullBolo,

Your full-list.txt shows where you have diverted from potential success.

#your SD card has the wz_mini_hack in a subdirectory called SD_ROOT
/Volumes/SDCARD/SD_ROOT:
total 4128
-rwx------@ 1 myadminuser  staff  1998399 Feb 24 03:14 factory_t31_ZMC6tiIDQN
drwx------@ 1 myadminuser  staff    16384 Jul  6 17:35 wz_mini

They must be at the root level of the SD card, and should look like:
***@***.*** ~ % ls -l /Volumes/WYZE
total 4608
-rwx------@ 1 rob  staff  1998399 28 May 03:50 factory_t31_ZMC6tiIDQN
drwx------@ 1 rob  staff    32768  6 Jul 14:47 wz_mini

SD_ROOT is a designation, not a folder. Relocate both of those items and delete your SD_ROOT folder. Then you have one less bloackage to loading wz_mini_hacks.

To your other points:

>  I'm wondering if the hidden Spotlight files are fudging things up
They are not, but if your SD card is mounted on your mac, you can open System Settings -> Siri & Spotlight. Scroll down to click on Spotlight Privacy. Add your SD volume to the exclusion list. then Spotlight will cease to make new .Spotlight-v100 folders.

To delete existing mac meta files, you can do this if you wish. Do not make any typing mistakes!
***@***.*** ~ % cd /Volumes/WYZE

***@***.*** WYZE % find . -name "._*"
./._factory_t31_ZMC6tiIDQN
./wz_mini/._usr
./wz_mini/._.DS_Store
./wz_mini/._bin
./wz_mini/._etc
./wz_mini/._swap.gz
./wz_mini/._www
./wz_mini/._mnt
./wz_mini/._root
./wz_mini/._lib
./wz_mini/._log
./wz_mini/._wz_mini.conf
./wz_mini/._tmp
./wz_mini/._wz_mini.conf-DOS
./wz_mini/._wz_mini.conf-MAC
./wz_mini/etc/rc.local.d/._.gitignore
./wz_mini/etc/wireguard/._.gitignore
./._wz_mini

***@***.*** WYZE % find . -name "._*" | xargs rm
***@***.*** WYZE % find . -name "._*”
***@***.*** WYZE %

>  Disk Utility only gives the option of MBR, GUID and Apple Partition Map as options, and I know it needs to be DOS. I think I may be getting confused at this step.
Disk Utility shows my SD card to be MBR. The WYZE filesystem on it is MS-DOS (FAT32).  I used the SD Card Formatter.app.

> I do like the idea of having a way to ssh into my camera, but I'm willing to do without if you think upgrading to a current firmware version if you think the hack would successfully load.
I just meant that for my V3, loading wz_mini_hacks and using ssh seems independent of the Wyze firmware version on the camera.
Other users indicate that streaming requires a specific lower firmware version.

> I'm determined to get this going and happy to try again.

Have another happy go at head banging.
Rob

CowSkullBolo, 2024-07-08:
Well now I feel silly. I was definitely confused and as copying over the entire SD_ROOT file. At least I learned a lot of things in the process of my mistake! Thank you for your help! Everything is working beautifully now.

JamesGriebler8, 2025-08-30:
Just want to say CowSkullBolo, that I did the exact same thing. Your comment thread helped me out. Thank you!