gtxaspec/wz_mini_hacks #785: Can I use mini hacks for this device - dafang 1s (see photos from the box)?

invisible999, 2024-08-13:
Hello everyone,

Back >4 years ago I got this Dafang 1s camera (as it was in the description) from Aliexpress. Visually it looks like identically as Wyze cam 2. It was sitting in the box for 4 years and I finally decided to put it into action. However I am not sure if the firmware can be applied to it.

This is the sticker on the camera itself
[image]

And this is the one side of the box:
[image]

If anyone knows more about this camera, would be happy to know.
Thanks

gtxaspec, 2024-08-13:
I would recommend using [thingino](http://thingino.com/).  check the discord or telegram for additional support.  It's a newer up-to-date firmware with additional features.

invisible999, 2024-08-13:
Hey thanks, never heard about it. Do you have persnal experience with this firmware?