gtxaspec/wz_mini_hacks #786: Wyze Forcing New Firmware

bjs-pdx, 2024-08-19:
It looks like Wyze is forcing new firmware onto cameras.  Just noticed a V2 that was changed. 

What is the best way to stop this.  It’s going to break things.

ltoolio, 2024-08-19:
What FW ver did you get forced to?

bjs-pdx, 2024-08-19:
V2 - 4.9.9.3006

bjs-pdx, 2024-08-19:
Here is the information from the config page:

[image]

endertable, 2024-08-24:
The actual setting we need is what you have in your /opt/wz_mini/wz_mini.conf file.  Look for a setting:

`DISABLE_FW_UPGRADE=`

to see if it is true or false.

Fortunately, for the V2 cam, wz_mini still works with the latest FW, but you may want to check your V3 cams for this setting as a FW update could break things on those for wz_mini.  So far, Wyze has only tried updates on my V2's not on my V3's. Although most of my V3's are on FW 4.36.10.3406 or higher.  I think they are ok if you are on at least 4.36.9.139 or higher for the V3 and anything less they may try forced updates, possibly so they don't have to support so many different API calls.

bjs-pdx, 2024-09-07:
So Wyze is forcing updated firmware in order to use the cameras in the app:
[image]

dudududodododedede, 2024-09-07:
I'm seeing this too when I'm trying to view the cameras in the app
Has anyone tried rolling back to an older version of the app?

bjs-pdx, 2024-09-07:
> I'm seeing this too when I'm trying to view the cameras in the app Has anyone tried rolling back to an older version of the app?

You can’t rollback on iOS.

tismofied, 2024-09-08:
> So Wyze is forcing updated firmware in order to use the cameras in the app: ![IMG_1033](https://private-user-images.githubusercontent.com/14082937/365399075-ebef88f0-05e0-4875-88ed-873dd70def09.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU3NTk2OTgsIm5iZiI6MTcyNTc1OTM5OCwicGF0aCI6Ii8xNDA4MjkzNy8zNjUzOTkwNzUtZWJlZjg4ZjAtMDVlMC00ODc1LTg4ZWQtODczZGQ3MGRlZjA5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA4VDAxMzYzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg0ZmUxOGVlMzc3ZjVhZTYxMTBmMGUyNTU3ZmEzOTllMjlkM2JmYjZmN2QyZmY1NWE0NTJiMWVlMzEyMGQ3MjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IaJoL1RSeRuP9A298tnTOrDOgtZUhJR356KFYL9n0e0)

I am seeing the exact same thing 😭

gtxaspec, 2024-09-08:
someone on the old firmware, try:

Please write the following in `/system/bin/app.ver` file and restart.
```
[VER]
appver=4.36.13.0416```

from https://github.com/mnakada/atomcam_tools/issues/115#issuecomment-2336583807
see if it works.

endertable, 2024-09-08:
On iOS you can also switch to another app real quick and switch back, and the box will be gone for the rest of the session.

bjs-pdx, 2024-09-08:
Had to give up and accept the inevitable upgrade.  Every time I tried to view a camera in the app the forced upgrade window would pop up with no way to close it.  I was able to sneak past it by powering off the camera and, when I powered it back on, quickly clicking the setup gear icon.  I had to reformat the SD Cards then run the update.  Of course after that, no more wyze_mini_hacks.

This really sucks.

gtxaspec, 2024-09-08:
@bjs-pdx you could try and downgrade, and try the fix i posted above

bjs-pdx, 2024-09-08:
Not sure a downgrade is going to do anything.  Wyze app checks the firmware and forces the upgrade.  How does your suggested fix change that?

gtxaspec, 2024-09-08:
the fix physically changes the firmware version reported by the device.  icamera reads from that file to tell the fw ver

bjs-pdx, 2024-09-08:
Okay...I see.  But still don't know what breaks in the Wyze app when running older firmware.  They are forcing this for some hidden reason but can't tell what that may be.

bjs-pdx, 2024-09-08:
Just to be clear, I think this project is great.  I’m just done with Wyze. Seems like every month or so they make a change and things break. Tired of fighting a losing battle. Going to switch to Amcrest for my camera needs. Much more reliable.

gtxaspec, 2024-09-08:
You can convert the wyze devices to a standalone firmware we've been working on... www.thingino.com.  just FYI.

bjs-pdx, 2024-09-08:
Not really something I want to do. Liked the idea of added functionality without losing the native functions.

cad1231, 2024-09-08:
> someone on the old firmware, try:
> 
> Please write the following in `/system/bin/app.ver` file and restart.
> 
> V2:
> 
> ```
> [VER]
> appver=4.9.9.3006
> ```
> 
> V3:
> 
> ```
> [VER]
> appver=4.36.13.0416
> ```
> 
> taken from [mnakada/atomcam_tools#115 (comment)](https://github.com/mnakada/atomcam_tools/issues/115#issuecomment-2336583807)
> 
> see if it works.

That file appears to be read-only. How can I write to it?

Thanks!

endertable, 2024-09-08:
You can’t on the V3 camera, but you should be able to alter the file that is in the /configs directory.  Supposedly, that is the file that I camera reads…

cad1231, 2024-09-08:
I tried changing /configs/app.ver - it doesn't do anything, just reverts to the old app.ver (even in /configs) on reboot.

cad1231, 2024-09-09:
@gtxaspec do you have any other ideas? The thread on atomcam-tools that you linked said that the person had to update the firmware for this trick to work. I upgraded using the shell script, but changes to /configs/app.ver still don't stick after a reboot. I have the wyze cam v3. Thanks!

gtxaspec, 2024-09-09:
i'll have to setup a wyze device ( i dont currently have any setup with stock firmware )  I'll try to restore one soon and test

sideup66, 2024-09-09:
Figured out the dns issue. Seems like wyze took a page out of our book :) and now runs dnsmasq by default on the device. Ours fails because the address is bound by their instance, plus on the newest firmware the build of dnsmasq in firmware is newer than the one present on wz_mini. I was able to address the DNS noise issue by killing the watchdog process they run for it + dnsmasq, running the copy as in the init file for resolv.conf, then starting the Wyze copy of dnsmasq with our config file (using -C argument.) after work ill submit a PR tonight with the fix so it can be integrated into the tree. Can include an if statement easily enough to check the existence of the system dnsmasq and if not present, simply use the wyze_mini one.

cad1231, 2024-09-09:
> i'll have to setup a wyze device ( i dont currently have any setup with stock firmware ) I'll try to restore one soon and test

Thanks!

sideup66, 2024-09-10:
See here for a updated dnsmasq script that is compatible with the newer firmware beyond 4.36.9.139. Also sent @gtxaspec a PR for upstream so its included here.
https://github.com/gtxaspec/wz_mini_hacks/pull/789

endertable, 2024-09-10:
@sideup66 hi, what FW version did you upgrade to that had all these changes?  Does the newer firmware allow wz_mini functionality again?

sideup66, 2024-09-10:
I am on 4.36.13.0416. Aside from dnsmasq, I am not sure if the way firmware is pushed may have changed. That should be investigated as well.

On the newer versions, you won't have RTSP still but wz_mini still has root shell on the device. I am simply updating the Dnsmasq script in my PR to utilize the built-in dnsmasq binary from wyze if it exists, it seems more up-to-date currently than the one present in wz_mini and I included logic for both, so the script should be able to detect one way or the other and effectively let you configure dnsmasq with our configuration file.

qip, 2024-09-11:
To continue view my wyze V3 on the app, I've the following quick fix:

In `/media/mmc/wz_mini/etc/rc.d/S10firmware`, add: 
```sh
echo -e "[VER]\nappver=4.36.13.0416" > /opt/wz_mini/tmp/patched_app.ver
mount --bind /opt/wz_mini/tmp/patched_app.ver /system/bin/app.ver
```
somewhere between `if [[ "$DISABLE_FW_UPGRADE" == "true" ]]; then` and `else`.

After that, restart the camera.

This fix is effective for immediate needs (I just want to check if my package is delivered on my phone right now), I guess it could be better to move this code to `rc.local.d` or somewhere more appropriate.

cad1231, 2024-09-11:
@qip thank you, that worked!

sideup66, 2024-09-11:
> To continue view my wyze V3 on the app, I've the following quick fix:
> 
> In `/media/mmc/wz_mini/etc/rc.d/S10firmware`, add:
> 
> ```shell
> echo -e "[VER]\nappver=4.36.13.0416" > /opt/wz_mini/tmp/patched_app.ver
> mount --bind /opt/wz_mini/tmp/patched_app.ver /system/bin/app.ver
> ```
> 
> somewhere between `if [[ "$DISABLE_FW_UPGRADE" == "true" ]]; then` and `else`.
> 
> After that, restart the camera.
> 
> This fix is effective for immediate needs (I just want to check if my package is delivered on my phone right now), I guess it could be better to move this code to `rc.local.d` or somewhere more appropriate.

Now heres another question, if one is full self-hosting this is a non-issue. However, if you are relying on wyze services for anything....there may be a good reason to force the firmware update (im going to take a stab and say they possibly moved to newer certificates in the new firmware that are about to expire in the old). Do we know what that is? My last talk with them was a vague "oh a bug was fixed".  I worry that if they are forcing people off the old firmware, cloud level things may change that will break functionality of the older fw.

 Also, we should test the scripts in newer fw to ensure things are working. Despite losing RTSP in newer, we still have full root shell on the device and those supporting scripts are still useful. I already fixed dnsmasq for newer above and sent a PR to @gtxaspec  for it. We should confirm the firmware blocker still works as well. My fear is everyone will hack the old firmware to cling on, then wyze will break something at the cloud level and everyone will whine and be forced to upgrade. We should have the devices setup for all scenarios in either sense IMO.

gtxaspec, 2024-09-11:
they could easily change the stuff at the cloud level to break this, for some reason they haven't, yet.  I would take anything customer service or media reps say with a grain of salt, they simply aren't privy to real engineering information.  Firmware blocker should still work as well, the method to upgrade fw has not changed.  eventually, one day, it will break, but not yet.

sideup66, 2024-09-11:
No problem, I am just considering both sides here. A certificate update could break the legacy firmware theoretically.

If one is interested in not using wyze at all, they should just move to their own firmware at this point. dafang is pretty mature and so is thingino with constant updates.

gtxaspec, 2024-09-11:
If i recall correctly from analyzing the binary before, they have the ability to remotely update certificates, so hopefully this wouldn't be the case.

I'll add the fix mentioned above to the official release later today.  Hopefully wz_mini still has a long life to live =D

sideup66, 2024-09-11:
It definitely has a long time to go yet. Even if RTSP is gone in newer firmware, its still insanely valuable for giving root access to the camera shell.

Id be curious to see a comparison of the latest firmware to 4.36.139 to exactly see what has changed in the icamera binary and why. We know about rtsp, but what else? It may give clues to why they forced it.

bjs-pdx, 2024-09-11:
I'm disheartened by what Wyze did.  They forced the update with no easy way to avoid it and fix the problem.  I had two cameras offline due to the update and was annoyed that the Wyze pop-up screen prevented any access to the camera or any ability to close that pop-up.  The only way I found around it was to force close the app.  Even then, every time I tried to access the live view of my V3 cameras the pop-up returned making them totally useless.  

For now, I'm running the Wyze firmware without this software.  I had to do manual firmware updates via sd card on two of the cameras.  I finally got around the pop-up by turning the cameras off, then turning them back on and quickly pressing the settings icon.  I then reformatted the sd cards in the cameras containing the wyze_mini_hacks software, rebooted and allowed the firmware to update.  

Too much trouble for now.  I'll watch for updates here and consider running this again if things seem stable.

gtxaspec, 2024-09-14:
pushed a fix for firmware version, please test and report back.

sideup66, 2024-09-16:
So far so good on my fix. However, I would say to change the delay for the resolv.conf copy to 5 seconds. Seems like its inconsistent lower, I am still testing here but noticed improvement doing that.

JuniorSad, 2024-09-17:
Quelqu'un n'aurai pas un hardware pour bypass iCloud

qip, 2024-09-21:
While the fix allows for viewing the camera on wzye app, I encountered issues with event recording and notifications (at least on V3).
Specifically, whenever an event is uploaded to the cloud, the camera restarts and resets to default settings (event recording, detection zones, logo/timestamp, etc.), and in the app the event is saved without thumbnail.

Probably remote API has changed? I assume wyze API returned an error code and iCamera just resets everything.
In `wz_init.log.0` the last few lines are:
```
Deleting logs older than 5 boots...
sh: /opt/wz_mini/log: out of range
sh: /opt/wz_mini/log/wz_init.log: out of range
sh: /opt/wz_mini/log/v4l2rtspserver.log: out of range
sh: /opt/wz_mini/log/daemon.log: out of range
sh: /opt/wz_mini/log/go2rtc.log: out of range
sh: /opt/wz_mini/log/dmesg.log: out of range
umount: can't remount /dev/mmcblk0p1 read-only
umount: can't remount /dev/mtdblock6 read-only
umount: media busy - remounted read-only
umount: tmpfs busy - remounted read-only
umount: devpts busy - remounted read-only
umount: can't remount tmpfs read-only
umount: /dev/mtdblock3 busy - remounted read-only
umount: can't remount tmpfs read-only
umount: can't remount /dev/mmcblk0p1 read-only
umount: can't remount /dev/mmcblk0p1 read-only
umount: can't remount devtmpfs read-only
^MThe system is going down NOW!
^MSent SIGTERM to all processes
^MSent SIGKILL to all processes
^MRequesting system reboot
```

DrVlikhell, 2024-09-22:
At first I thought the new fix for spoofing the firmware version didn't work, but after a little digging I learned I had to set DISABLE_FW_UPGRADE="false" to "true" in wz_mini.conf. Once I did that it works fine.

akivacp, 2024-09-23:
> At first I thought the new fix for spoofing the firmware version didn't work, but after a little digging I learned I had to set DISABLE_FW_UPGRADE="false" to "true" in wz_mini.conf. Once I did that it works fine.

is this the solution to @qip  issue? because suddenly all my cameras seem to be going in a bootloop.

sideup66, 2024-09-28:
Wanted to report my final findings regarding this. As a person who "jumped the fence" and went to the newer wyze firmware....

After upgrading, I initially had a weird issue where my camera's  record light would typically get stuck after recording an event. Was able to fix that by resetting the camera. I learned that well, after the firmware jump there were multiple reports of weirdness from other v3 owners who were able to fix it by performing the reset. Wyze mini still works, albeit without any of the RTSP features, but if you need root shell to run services and effectively make the experience of it not suck, it works well :).

The other interesting thing of note is that in the newer firmware (my v3 is on 4.36.13.0416 at the time of this writing). I notice that wyze actually took a page from our book, and added a local dnsmasq server on the camera. That build is at 2.88 while the wz-mini build is 2.86. Not a HUGE difference but a slightly newer version. Therefore, I had the update merged to determine whether the onboard dnsmasq was present and restart it using the wz-mini config rather than its builtin config if the local dns is enabled. otherwise, use the wz-mini's build.

One thing of note in that fix is that the initial 3 second setting i pushed with the commit might be too short. If needed, I can submit another PR pushing it out to 5 seconds, or even 10 if preferred. But its such a small change if Gtxaspec wants to do it im fine with that also. Or maybe we come up with a faster way to do it. Also, I would suggest makign the firmware spoof a separate feature from blocking firmware updates. In my own setup, I am using the firmware update block as given Wyze's history of messy updates, id rather not update again unless forced and/or a major breaking bug occurs to the device but rather not spoof the firmware version, just incase there is a change in the way cloud is communicating or functioning with my device and that spoofed version throws things off....

At the time of this report my camera seems stable.

Few TLDR notes.
1. if you DO decide to upgrade the firmware to the new version.... DO IT LOCALLY. Download the version you want, put it on a blank microsd and do a local flash. the cloud update will take you through ~3 flashes before the final. Also, a local update will be the most reliable.
2. IF YOU DO UPDATE.....RESET your device after, then set it up again in wyze. The firmware in this seems to be a more drastic jump in their setup and may not place nice with the older configuration. In my experience, it just seemed to clear up a lot of weirdness in my camera and things are more stable in my experience.
3. after updating, get the latest wz_mini firmware update so if you are using dnsmasq locally on the device you can take advantage of it. By default the update does break that script.

dasb00ter, 2024-10-01:
So I have few pan cam that i had firmware update disable= true. It seems to have blocked the update but I have the nag in the wyze app(which my wife uses...sigh). I have run update-run.sh  and it looks as if it has updated mini hacks. Is there something I need to do so this works as I still get the nag. I guess first how do tell if I have in fact had a successful run of upgrade-run.sh? Is there a config setting for "spoofing the firmware"

Thanks

sideup66, 2024-10-16:
I plan to make a followup PR soon. Im testing out some further changes I made to the dnsmasq script. It seems something from the wyze app side kept trying to overwrite the change from the script, I was able to block this by remounting the bind to make it read-only. 

Im testing due to an unrelated issue with the LED not functioning correctly on mine. I think I solved it, but im trying to be sure.

Only other follow up from all of this I am going to send a PR for is to break out the firmware spoofing potentially into its own independent script that can be controlled from the conf file. Not a fan of it being enabled with fw disable as some cams may be on newer firmware, thus breaking their reporting and potential cloud communications.

sideup66, 2024-10-18:
Made PR https://github.com/gtxaspec/wz_mini_hacks/pull/801 with my aforementioned fixes.

dasb00ter, 2024-10-19:
> I plan to make a followup PR soon. Im testing out some further changes I made to the dnsmasq script. It seems something from the wyze app side kept trying to overwrite the change from the script, I [...]

I had an interesting thing happen with one camera(vs2). It was fully functional but I had the blue led in the back flashing. It was doing so even with the light disabled in the app. I think I rebooted and it was fine. It was just a problem BC it's in a dark area and was bothersome to somebody

sideup66, 2024-10-19:
I've seen that once or twice. Not super common but usually indictive of a
past or present connectivity issue. If mine was doing it with me not around
not a huge deal as cron does a nightly reboot which would have cleared it.

ericloyd, 2024-10-24:
I know I'm coming late to this party, but I only recently discovered how great wz_mini is for V3 cams and so I bought two of them to use for livestreaming, monitoring and generally doing public outreach for astrophotography.  Originally, I flashed the beta RTSP firmware from Wyze, obtained from Wayback Machine.  It worked great with my two test cams on 32GB cards.  I decided to get six more cams to do a full roll-out of setting up live streams for the telescopes (both looking at the rigs and looking at the sky).  I purchased 128GB cards to go in all those cams.  I use RTSP to stream all that to OBS that then streams to YouTube and/or Twitch.

Then I found out that I had to ugprade the firmware to solve the micro SD card corruption problem.  I went to 4.36.9.131 and then .139 to make sure I didn't go past the RTSP barrier.  

Now I see, like everyone else, that the app is forcing me to ugprade further.  Honestly, I don't care.  I don't use the app, I don't need its features, and if I do, I'll use Home Assistant.  But I just downloaded the latest wz_mini which seems to have the "echo -e "[VER]\nappver=4.36.13.0416" > /opt/wz_mini/tmp/patched_app.ver" line in S10firmware, and yet the Wyze app is still telling me I should update.

While I don't need the app, it's faster to check on cam stats there than it is to pop up an RTSP stream (though I could use VLC for that on the Android phone).  Do I need to do something else to have this fix work for convincing the app that I don't need to upgrade?  wz_mini web page reports the .0416 firmware so it looks like it's working, but the app still complains.

Also, the whole point of wz_mini for me was the RTSP streams.  If I bite the bullet and push past the upgrade barrier, I know I lose that.  Would I still be able to pull a live stream via the Go2RTC server, or does that rely on the RTSP under the hood?

Sorry for the questions, I'm just trying to make sure I didn't make a $400 mistake with all these cameras and cards.

sideup66, 2024-10-24:
Hi!
If you move past the rtsp barrier no, no custom rtsp streaming would work
unfortunately.... If you are looking to exclusively use it for that I'd
recommend dafang or thingino which fully replace wyze stock firmware, and
fully support streaming.

I personally am sticking to wz_mini because though I don't use rtsp I do
use the hack to provide some stability, security, and usability patches to
my otherwise stock v3.

At this point the logical solution is to start testing the new firmware and
look for a workaround to rtsp. It's a matter of time imo before wyze kills
the legacy firmware with some cloud change I think.... Not to discourage
anyone.

ericloyd, 2024-10-24:
Okay, thanks for that.  I was unaware of the other firmwares.  I took a quick look, but it does not seem like they do the recording that the Wyze firmware does.  Which is something I like; being able to go back in time to look at a specific event/time is great for monitoring remote telescopes.

Honestly, I'd be fine with just using the .139 firmware to get the RTSP and then wz_mini to get the streaming capabilities and other things that it provides, but now I have a new problem - the camera keeps crashing on me.  It's like it ran fine on a 32GB test microSD card with the old beta RTSP firmware from Wyze, but now that I've got the .139 on it, on a 128GB card, it's just completely unstable.  

Maybe I'll return all these 128GB cards and get 32GB cards instead.  They seem to work with the old RTSP firmware just fine but man, what a pain.

sideup66, 2024-10-24:
You could possibly have them record somewhere and be able to do time lapse.
As for the firmware message, make sure the wz_mini system is actually
loading on your SD card...usually I know it's not if ssh isn't running

sideup66, 2024-10-24:
An alternative if you move to New firmware is to setup new firmware, still
run wz_mini and copy off the video folder, could even smb mount it I think,
then playback using whatever. Even when rtsp is not running the ssh
functions as well as file transfer capabilities of wz_mini still work fine
on newer firmware

ericloyd, 2024-10-24:
Yes, wz_mini is definitely loading.  I have monitoring software that SSHes to the box and checks to make sure that /opt/wz_mini/log is writeable.  That's how I found the corrupt microSD card problem. I'm actually going to see if I can find the 4.49.1.47 firmware, since the person on the microSD corruption thread said that solved their problem. 

And while I could move images around with new firmware, if I can't stream with it, then it's useless.  That's the whole point - to be able to stream these things to Youtube and Twitch for real-time outreach in the field of Astrophotography.

ericloyd, 2024-10-24:
Wait.  I can't go to 4.49.1.47, because that breaks the RTSP barrier.  I'm going to downgrade from .139 to .131 and see if that solves my crashing problem.

gtxaspec, 2024-10-24:
4.49.x is for the pan, not for the v3, FYI.

ericloyd, 2024-10-24:
Thanks.  Just put .131 on and let it run for a few minutes.  No crashes.  Now I've got wz_mini on and just first booted.  Waiting to see if it crashes or gives me an sd card corruption.

ericloyd, 2024-10-24:
So far, with just RTSP enabled (low res for now) and the web interface but no Go2RTC, the 4.36.9.131 firmware is holding steady without crashes and without sd card corruptions.  If this works, I may just lock it here and call it a day.

gtxaspec, 2024-10-24:
if you have no interest in using the app, you may want to switch to thingino, which provides RTSP too.  There's other folks interested in astronomy which are using it.

ericloyd, 2024-10-24:
I saw that briefly, but wasn't sure if it still did the automatic recording that the Wyze app does, and I have no clue on how to figure out what my specific SoS and Wifi controller is.  So I punted and came back to this.

[9 more replies]