gtxaspec/wz_mini_hacks #794: Floodlight Cam Can't Access Floodlight Controls

tmsteinhardt, 2024-09-20:
I have the custom floodlight kernel on the floodlight cam but I can't seem to edit the floodlight controls as they are all locked out from being edited.  I can set the "REMOTE SPOTLIGHT" option to true but have no option to set the REMOTE_SPOTLIGHT_HOST="0.0.0.0".  The line/option doesn't even  appear in the list within the web UI.  Additionally, is SSH access required to run the following commands per the instructions in the "?" pop out for the REMOTE_SPOTLIGHT line on the web UI, or is there another way to run those commands?

socat TCP4-LISTEN:9000,reuseaddr,fork /dev/ttyUSB0,raw,echo=0
Change /dev/ttyUSB0 to whatever path your spotlight enumerated to if necessary. The camera will now be able to control the spotlight.

tmsteinhardt, 2024-09-20:
I pulled the SD card on the cam and added the SSH public key and added the REMOTE_SPOTLIGHT_HOST="0.0.0.0" line to the wz_mini.conf file.  The REMOTE_SPOTLIGHT_HOST="0.0.0.0" line now shows up in WZ Mini Config page in the web UI.  When I try to SSH into the camera both from powershell and PUTTY to run the "socat TCP4-LISTEN:9000,reuseaddr,fork /dev/ttyUSB0,raw,echo=0" command I get "NO SUPPORTED AUTHENTICATION METHODS AVAILABLE (SERVER SENT: PUBLICKEY).  Any help with possible errors that I made when pasting the public key to the camera or error when attempting to SSH into the camera?

tmsteinhardt, 2024-09-20:
Turns out I didn't fully understand the SSH Keys.  I had created multiple keys with unique names for each camera so the files weren't named is_rsa.  After renaming the key files I can now access the camera through SSH.  After running the socat TCP4-LISTEN:9000,reuseaddr,fork /dev/ttyUSB0,raw,echo=0 command I get the following output:

socat[21558] E bind(5, {AF=2 0.0.0.0:9000}, 16): Address already in use

I rebooted the camera but nothing seems to be different and I still don't see any ability to change the flood light settings.  Do I need to do anything with what came from the output of the command?

ryang3d, 2024-09-25:
Replace your kernel file with the one from this post (#172): https://github.com/gtxaspec/wz_mini_hacks/issues/172#issuecomment-1179881571

tmsteinhardt, 2024-09-26:
> Replace your kernel file with the one from this post (#172): [#172 (comment)](https://github.com/gtxaspec/wz_mini_hacks/issues/172#issuecomment-1179881571)
> 
> I have 3 floodlights with this installed and light control works as expected through the Wyze app and Home Assistant.

That's the kernal that I already used.  I even tried reloading it just to make sure but still don't have access to the settings in the web UI as they are all blocked from making changes to them.  In the wyze app you can't access anything because as soon as you click on the camera wyze tries to force a firmware update with no way of ignoring it other than closing the app.

ryang3d, 2024-09-26:
Did you extract the "kernel-t31.bin" file from the "kernel-t31.bin.gz" file and rename it to "factory_t31_ZMC6tiIDQN" (no .bin extension) before replacing the existing one in the root of your SD Card?

Also, is your camera on firmware 4.36.9.139?

Also just noticed you are using the "REMOTE SPOTLIGHT" setting... this has nothing to do with floodlights that are attached to v3 Cams, so make sure that is set to "False". (For reference, _this_ is the Wyze _Spotlight_: https://www.wyze.com/products/wyze-cam-v3-spotlight but I assume you are trying to use the Wyze _Floodlight_ v1: https://www.wyze.com/products/wyze-cam-floodlight-v1 which I believe is the only floodlight that is supported by wz_mini_hacks)

I do believe when I first got this working on my floodlights, they did require a reboot (fully disconnect power and reconnect power to be safe; i.e. flip the circuit the floodlight is on to off and back), and it may have taken a moment for things to work in the Wyze app again. But it definitely does work.

tmsteinhardt, 2024-09-26:
> Did you extract the "kernel-t31.bin" file from the "kernel-t31.bin.gz" archive from that post and rename it to "factory_t31_ZMC6tiIDQN" (no .bin extension) before replacing the existing one in the root of your SD Card?
> 
> Also, is your camera on firmware 4.36.9.139?
> 
> Also just noticed you are using the "REMOTE SPOTLIGHT" setting... this has nothing to do with floodlights that are attached to v3 Cams, so make sure that is set to "False". (For reference, _this_ is the Wyze _Spotlight_: https://www.wyze.com/products/wyze-cam-v3-spotlight but I assume you are trying to use the Wyze _Floodlight_ v1: https://www.wyze.com/products/wyze-cam-floodlight-v1 which I believe is the only floodlight that is supported by wz_mini_hacks)
> 
> I do believe when I first got this working on my floodlights, they did require a reboot (fully disconnect power and reconnect power to be safe; i.e. flip the circuit the floodlight is on to off and back), and it may have taken a moment for things to work in the Wyze app again. But it definitely does work.

Yes, I extracted the bin file from the .gz file and removed the .bin suffix after renaming it to factory_t31_ZMC6tiIDQN and my camera is on 4.36.9.139 firmware.  I just set remote spotlight to false and when I rebooted couldn't get the camera to come back online so I pulled the sd card and put a fresh copy of all the files onto it.  That option is very confusing, thanks for explaining that it's for the spotlight not flood light.  I just did a full reboot from the breaker.  Maybe I'm not understanding what controls I'm supposed to have but in the wz mini web tools interface I don't see any settings that I can change to affect the sensitivity of the flood light PIR sensor.  As I stated previously if I try to go into the camera from the wyze app wyze just tries to force a firmware update which obviously can't happen so there's no way to access the floodlight menu in the wyze app either.  I appreciate the help, I just don't know what I'm missing.

ryang3d, 2024-09-26:
None of those controls are available through wz_mini_hacks web UI. They are solely controlled through the Wyze app as far as I understand it. 

I have not tried this myself yet since I had everything set before Wyze started forcing the latest firmware upgrades (and I only use Home Assistant to turn the floodlight on/off), but you might consider trying this fix that tricks the app into thinking you have the latest firmware, which should then allow you to access those controls without actually updating the firmware: 
https://github.com/gtxaspec/wz_mini_hacks/issues/786#issuecomment-2342605394

ryang3d, 2024-09-26:
It sounds like gtxaspec already integrated that fix into the official files a couple weeks ago, so maybe just try downloading fresh copies of the SD card files from the GitHub repo, and still replace the kennel file... Might still be worth confirming that fix does exist in your files before you get ahead of yourself.

endertable, 2024-09-28:
As I stated previously if I try to go into the camera from the wyze app wyze just tries to force a firmware update which obviously can't happen so there's no way to access the floodlight menu in the wyze app either. 

If you have an iPhone, this hack works for me for the Wyze app to get past the FW forced update barker screen:

Once the camera connects and the barker screen comes up, switch to another app for 1 second, then switch back to the Wyze app.  It will reconnect to the cam but no more barker screen (for this session).   Even just minimizing the app, and bringing it back up works for me.

Not sure if this work with other OS's, but worth a try.

ryang3d, 2024-09-30:
> None of those controls are available through wz_mini_hacks web UI. They are solely controlled through the Wyze app as far as I understand it.
> 
> I have not tried this myself yet since I had everything set before Wyze started forcing the latest firmware upgrades (and I only use Home Assistant to turn the floodlight on/off), but you might consider trying this fix that tricks the app into thinking you have the latest firmware, which should then allow you to access those controls without actually updating the firmware: [#786 (comment)](https://github.com/gtxaspec/wz_mini_hacks/issues/786#issuecomment-2342605394)

@tmsteinhardt I just tried this fix and it definitely works (for now); I can see and control my floodlight in the Wyze App again without being prompted for the forced firmware upgrade. As I previously mentioned, it might just be easiest for you to re-download the SD_ROOT files from github to get the latest versions (which include this fix).