gtxaspec/wz_mini_hacks #800: SSH...

BigBobDalton, 2024-10-12:
Okay, I am not new to using ssh, I have raspberry's working with my pub file but this cam just won't authorize.  Not sure if dropbear has anything to do with it but I keep getting refusal to login stating Permission denied (publickey).
the authorization file has my correct generated key.  I even tried recreating a new pub key and adding that to the file, still it fails.

On my Raspberry's I can edit the ssh.hd file but this wz_mini doesn't appear to have that option.

Any help would be great.

Thanks.

bjs-pdx, 2024-10-21:
Any chance your key file has line feeds/carriage returns in it?  I had a similar problem until I realized mine did.

my-umd, 2024-10-25:
What type of key are you using?  I can only get type ed25519 key working.

BigBobDalton, 2024-10-25:
No, my key does not have carriage returns, I even tried an empty key.  Odd that my key works on every other device I have, albeit windows, linux and macOS.

Encoding is ed25519

virmaior, 2024-10-27:
One question would be whether your camera is properly running wz_mini.
can you access the web interface for wz_mini on that camera?
what version of the wyze firmware are you running?
is it a v2 or v3 camera?

BigBobDalton, 2024-10-27:
yeah, I can access it via wz_mini camera is version 3.  

At this point I will just keep taking out the sd-card since ssh is not working between my iMac and the camera.

virmaior, 2024-10-28:
I had quite a bit of difficulty getting osx to use keys. 
There's some settings I've had to do on the security chain _every time I reboot_ and then have to use `sudo ssh <camera ip>`.

donsandrop, 2024-12-08:
to get SSH working while using macOS

1. Open a terminal window
2. type "ssh-keygen" and follow the prompts, example below

```
% ssh-keygen
Generating public/private ed25519 key pair.
Enter file in which to save the key (/Users/UserID/.ssh/id_ed25519): wz_mini
Enter passphrase (empty for no passphrase): 
Enter same passphrase again: 
Your identification has been saved in wz_mini
Your public key has been saved in wz_mini.pub
```

3. find the .pub file that was created. This was in /Users/UserID for me.
4. Open the file in a text editor
5. Copy the key and place it in the /wz_mini_hacks-master/SD_ROOT/wz_mini/etc/ssh/authorized_keys file and save the setting. It should look like the below

```
#PLACE AUTHORIZED KEYS BELOW

ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIHayxPmJSWfujHUmPgIrB2lOf6bE1jsQwiUR77InsbTh UserID@computername.local

```

7.  to ssh into the camera type a similar command in a terminal window:
`ssh root@wyzecameraip -i /Users/UserID/ssh/wz_mini`

untotren, 2025-06-02:
> to get SSH working while using macOS
>
> 1. Open a terminal window
> 2. type "ssh-keygen" and follow the prompts, example below
>
> ```
> % ssh-keygen
> Generating public/private ed25519 key pair.
> Enter file in [...]

Honestly thank you for this, this needs to be on the front page it is so much more helpful than any of the existing documentation.