gtxaspec/wz_mini_hacks #805: Wyze V2 audio distorted

my-umd, 2024-10-25:
I'm following [this instruction](https://github.com/gtxaspec/wz_mini_hacks/wiki/Audio-Playback) to make wav files for my V2.  But when I played it on the camera, it plays slower, and a female voice would change to male voice.  I verified that the converted wav file plays fine on my computer.  Does anyone know what isgoing on?

gtxaspec, 2024-10-25:
you need to use the right sample rate, check the samplerate of the files included, i think for the v2 it's 8000hz

hua-bing, 2024-10-25:
Thanks @gtxaspec.  Using 8000 works.
BTW, the sample audio gives errors when trying to check the info.:
```
$ ffmpeg -i init_v2.wav
......
init_v2.wav: Invalid data found when processing input
```
Pls close the issue.  For some reason, I do not see an option for me to close it.

closed 2024-10-27 by my-umd