gtxaspec/wz_mini_hacks #807: Crashing v3 cameras ("firmware upgrade blocking" regression)

dudududodododedede, 2024-11-02:
Bringing this from discussion into the issue tracker:
https://github.com/gtxaspec/wz_mini_hacks/discussions/803

I think there is a regression with the "firmware upgrade blocking" feature which causes v3 cameras to reboot very often (in my case every 1-2 minutes)

As soon as I set DISABLE_FW_UPGRADE to FALSE the reboots stop and the cameras are rock solid. Sadly, this workaround isn't great since there is then an unskippable nag screen in the Wyze app

My exact camera hardware is Wyze v3
cpu: t31x
wifi chipset: rtl

gtxaspec, 2024-11-02:
need logs from a device before it reboots.  
`logread`

gponick, 2024-11-08:
Attached are the last bit from `logread -f` from the a WyzeCam V3 that reboots pretty regularly (hasn't been up more than 5m today before rebooting). I'll leave it in this state in case there's anything worth testing or we need more confirmation.
[logread-wcv3-reboot-1.txt](https://github.com/user-attachments/files/17684922/logread-wcv3-reboot-1.txt)
[logread-wcv3-reboot.txt](https://github.com/user-attachments/files/17684923/logread-wcv3-reboot.txt)

gtxaspec, 2024-11-08:
`Nov  8 11:28:51 syslog: [   71.407643] do_page_fault() #2: sending SIGSEGV to alarm-record for invalid read access from`

need to think about this one...hmm

edsantiago, 2025-01-18:
Another confirmation. A few months ago I set `DISABLE_FW_UPGRADE="true"` among other changes... and both my cameras became unusable. Crashing on every motion event, saving zero-length jpgs (when saving anything at all) and not saving event videos. I tried reverting everything else, but not this one, because, nah, this couldn't possibly cause the camera to crash...

With `DISABLE_FW_UPGRADE="false"` I now have 36 hours of uptime, and motion events are actually saved. THANK YOU!

HeyItsDaddy, 2025-02-18:
Hey @gtxaspec , love your work!  Any chance we might be able to figure out what's causing the reboots when doing the firmware spoofing?  I've been living with the reboots (as RTSP + Wyze App usage are more important to me right now), but some cams reboot way more than others (like every few minutes), and would be spectacular to know if there might be a resolution in sight.  (I know you're rocking out Thingino...but until I can find an app that works as well as the Wyze app, on Android and Apple, with SD card replay, and with access to the cams stuck behind a darn CGNAT...I have to stick with what I have for now.)  :-(

gtxaspec, 2025-02-20:
just got a spare v3 back on stock firmware, i'll try to take a look this weekend!

edmonddc, 2025-05-14:
This also happened on a new installation. Performance is terrible when the disable fw upgrade is set to true.