gtxaspec/wz_mini_hacks #810: Motion stops recording after 00.mp4

jaajee, 2024-11-29:
I have a unique issue. I’ve been using wz_mini_hacks on my V3 for a long time, but for the past 1.5 years, I’ve noticed something strange. Every time the clock hits zero minutes (e.g., 10:00), my camera stops recording and writing to the tmp folder.

I’ve observed this happening live. When the clock reaches 00 (e.g., 10:00), the 00.mp4 file gets stuck and stops recording. That file remains in the tmp folder, and no further recordings occur afterward. To fix this issue, I have to restart my camera. However, I’m unsure what’s causing it to stop precisely at zero minutes every time.

virmaior, 2024-12-30:
this sounds like an issue related to the cleanup function.

are you using network storage?
how big is your SD card?

jaajee, 2025-01-04:
> this sounds like an issue related to the cleanup function.
> 
> are you using network storage? how big is your SD card?

Thank you for your reply.
Yes, I am using network storage to save my recordings and have configured a _cifs_ mount in _/etc/fstab_. Additionally, I have a custom script (_/opt/wz_mini/etc/rc.local_) that runs _mount -a_ during reboot to ensure the following directories are mounted:

-     /media/mmc/record
-     /media/mmc/alarm
-     /media/mmc/time_lapse

My SD card has a capacity of 128 GB.

virmaior, 2025-01-05:
I think this is a known issue dating back to wyze_hacks where the iCamera script designed to sweep out old files craps out when the size of the drive exceeds its comprehension.

virmaior, 2025-02-11:
See https://github.com/HclX/WyzeHacks : "SD Card Size emulation":

> People reports the hack doesn't work with NFS shares bigger than a certain size. So I added a SD card size emulation so that no matter how big your NFS share is, the hack will always report an acceptable size with used space set to zero unless your free space is less than 16GB, in which case, it will report a 16GB SD card with the correct free space information.

I don't know what "doesn't work" means here, but my sense is that the sweep utility inside of iCamera that deletes things off a SD card is getting confused by your NFS.

jaajee, 2025-02-12:
> See https://github.com/HclX/WyzeHacks : "SD Card Size emulation":
>
>
> I don't know what "doesn't work" means here, but my sense is that the sweep utility inside of iCamera that deletes things off a SD [...]

I can run another test by setting the NFS size to a significantly lower value (currently, it's 6TB) to determine if it has any impact on the issue.

jaajee, 2025-02-21:
I think I found the solution, if I disable 'MicroCard recording' on Wyze app, it will not cause this problem. After doing this, so far I have not encountered this issue.