gtxaspec/wz_mini_hacks #813: mac address

ju5t3nc4s3, 2025-01-20:
is there anyway around having 2 wyze v3 that have the same mac address.  had to you a partition backup from one to fix another,, tryed on diferent AP they have diferent ip addreses . both show on each router but no rtsp from one. power off the other and the one works again..

robughblah, 2025-01-21:
Well, MAC addresses have to be unique on the same subnet. To ensure that, manufacturers make them unique per product.So in theory you could have each one on a different subnet. But i doubt that having one on your guest network and one on your main network would do that. My decos keep guest and main network all in the same subnet, but just vlan them.  Could try it.Rob

virmaior, 2025-01-29:
all of my cameras are using USB_DIRECT where you can directly wet it.
same came done if using an ethernet adapter

I don't see an option for doing that with the wifi chip but ... you could try the bonding interface, which I think has to generate a different mac.